merbanan/rtl_433 #1223: Adding Florabest BBQ Thermometer

ai-xyz, 2019-12-03:
Hi i have [this](https://www.lidl.de/de/florabest-funk-grillthermometer-mit-8-garprogrammen-5-garstufen-waehlbar-mit-edelstahl/p271491) barbecue thermometer and would like to add it to rtl_433. It consists of two parts, one sending the temperature of the sensor and the other, the receiver showing the temperature in 1°C/F steps
I already did some analysis and got it working, but not the cleanest way because i am not programming regulary with C.

It seems like the device is sending always 30bit repeated about 16 times.

Here is the output of some signal i recorded: 

>  Reading samples from file: g047_433.92M_250k.cu8
> *** signal_start = 36507, signal_end = 323025, signal_len = 286518, pulses_found = 296
> Iteration 1. t: 133    min: 131 (289)    max: 135 (7)    delta 5
> Iteration 2. t: 133    min: 131 (289)    max: 135 (7)    delta 0
> Distance coding: Pulse length 133
> 
> Short distance: 502, long distance: 1003, packet distance: 2258
> 
> p_limit: 133
> bitbuffer:: Number of rows: 18 
> [00] {30} 49 09 32 e0 : 01001001 00001001 00110010 111000
> [01] { 1} 00          : 0
> [02] {30} 49 09 32 e0 : 01001001 00001001 00110010 111000
> [03] { 1} 00          : 0
> [04] {30} 49 09 32 e0 : 01001001 00001001 00110010 111000
> [05] { 1} 00          : 0
> [06] {30} 49 09 32 e0 : 01001001 00001001 00110010 111000
> [07] { 1} 00          : 0
> [08] {30} 49 09 32 e0 : 01001001 00001001 00110010 111000
> [09] { 1} 00          : 0
> [10] {30} 49 09 32 e0 : 01001001 00001001 00110010 111000
> [11] { 1} 00          : 0
> [12] {30} 49 09 32 e0 : 01001001 00001001 00110010 111000
> [13] { 1} 00          : 0
> [14] {30} 49 09 32 e0 : 01001001 00001001 00110010 111000
> [15] { 1} 00          : 0
> [16] {30} 49 09 32 e0 : 01001001 00001001 00110010 111000
> [17] { 0}             : 
> Iteration 1. t: 0    min: 0 (0)    max: 0 (0)    delta -727379968
> Iteration 2. t: 0    min: 0 (0)    max: 0 (0)    delta 0
> Distance coding: Pulse length 0
> 
> Short distance: 1000000, long distance: 0, packet distance: 0
> 
> p_limit: 0
> bitbuffer:: Number of rows: 0 

I recorded even more and put it on this [bitbench](https://triq.net/bitbench?c=%7B30%7D%2049093204%20%20%5B21%C2%B0C%5D&c=%7B30%7D%20490932bc%20%20%5B23%C2%B0C%5D&c=%7B30%7D%20490932b0%20%20%5B24%C2%B0C%5D&c=%7B30%7D%20490933ac%20%20%5B25%C2%B0C%5D&c=%7B30%7D%2049093494%20%20%5B26%C2%B0C%5D&c=%7B30%7D%204909359c%20%20%5B27%C2%B0C%5D&c=%7B30%7D%204909360c%20%20%5B28%C2%B0C%5D&c=%7B30%7D%2049093660%20%20%5B29%C2%B0C%5D&c=%7B30%7D%2049093720%20%20%5B30%C2%B0C%5D&c=%7B30%7D%2049093754%20%20%5B31%C2%B0C%5D&c=%7B30%7D%204909377c%20%20%5B31%C2%B0C%5D&c=%7B30%7D%2049093798%20%20%5B31%C2%B0C%5D&c=%7B30%7D%20490937a4%20%20%5B31%C2%B0C%5D&c=%7B30%7D%20490937c4%20%20%5B31%C2%B0C%5D&f=8h%0Ab%0A16x14d) with the temperature that i saw on the thermometer (maybe some mistakes because the temperature is something moving very quickly). It seems like the first 16 bits are fixed (at least on my device) and the temperature is encoded inside the last 14 bits. I am totally not an expert on doing this encoding stuff. But it seems like that

```
    temp1_raw = (bytes[2] << 2) | (bytes[3] & 0xc0) >> 6;
    temp1_c = temp1_raw*0.444-67.5;
```

is fitting the temperature if you round it to the nearest integer. I got this formula by transferring the last 14bit to a digit and solving least squares on this.
For sure this is not the exact decoding but i think the sensor is sending a floating point value and the receiver is somehow truncating this. Maybe someone has a clue about the encoding of this temperature with 14bits? Than it would be great if we could add this to rtl_433.

[florabest.zip](https://github.com/merbanan/rtl_433/files/3916781/florabest.zip)

zuckschwerdt, 2019-12-03:
Looks good so far. Can you try to capture more extreme temperatures? E.g. put it in a freezer and in a pot of boiling water.
Also can you reset the battery and try to get different first 16 bits? Likely only 8 bits will change and the rest is a fixed sync.

zuckschwerdt, 2019-12-03:
And a captured sample would be great to check if it's really 30 bits or if there is some sync bits in there.

ai-xyz, 2019-12-03:
Hi, 

i forgot to add some temperatures, here is a [bitbench](https://triq.net/bitbench?c=%7B30%7D%2049093204%20%20%5B21%C2%B0C%5D&c=%7B30%7D%20490932bc%20%20%5B23%C2%B0C%5D&c=%7B30%7D%20490932b0%20%20%5B24%C2%B0C%5D&c=%7B30%7D%20490933ac%20%20%5B25%C2%B0C%5D&c=%7B30%7D%2049093494%20%20%5B26%C2%B0C%5D&c=%7B30%7D%204909359c%20%20%5B27%C2%B0C%5D&c=%7B30%7D%204909360c%20%20%5B28%C2%B0C%5D&c=%7B30%7D%2049093660%20%20%5B29%C2%B0C%5D&c=%7B30%7D%2049093720%20%20%5B30%C2%B0C%5D&c=%7B30%7D%2049093754%20%20%5B31%C2%B0C%5D&c=%7B30%7D%204909377c%20%20%5B31%C2%B0C%5D&c=%7B30%7D%2049093798%20%20%5B31%C2%B0C%5D&c=%7B30%7D%20490937a4%20%20%5B31%C2%B0C%5D&c=%7B30%7D%20490937c4%20%20%5B31%C2%B0C%5D&c=%7B30%7D%2049095194%20%20%5B77%C2%B0C%5D&c=%7B30%7D%2049095300%20%20%5B79%C2%B0C%5D&c=%7B30%7D%20490962f8%20%20%5B108%C2%B0C%5D&c=%7B30%7D%20490978fc%20%20%5B147%C2%B0C%5D&c=%7B30%7D%204909a6fc%20%20%5B229%C2%B0C%5D&f=8h%0Ab%0A16x14d) with more hot temperatures. Later on today i will reset the battery and do some extreme temperatures like putting in the freezer (even if supposed to the manual the BBQ thermometer is only supposed to work from 0 to 200°C)

ai-xyz, 2019-12-03:
Here two samples i still have
[samples.zip](https://github.com/merbanan/rtl_433/files/3916947/samples.zip)

zuckschwerdt, 2019-12-03:
Pulseview shows 30 PPM bits (2ms/4ms) and a sync bit with 9/2/9ms.
- .modulation    = OOK_PULSE_PPM,
- .short_width   = 2000,
- .long_width    = 4000,
- .gap_limit     = 6000, // more than 4000 but less than 9000
- .reset_limit   = 11000 // more than 9000

zuckschwerdt, 2019-12-03:
Btw. the last bit is parity (Calculation: Offset=0 Length=30 Width=1).

zuckschwerdt, 2019-12-03:
And I guess scale 10, offset 90 would match in Fahrenheit (i.e. value x 0.1 - 90 = Temp_F), so actual conversion to C would be (x * 0.0555 - 67.778)

ai-xyz, 2019-12-03:
Ah the parity bit, i think that's the game changer, was looking all the time for some 4bit crc somewhere hidden. I will implement your advices later on and do some tests. But this calculation looks already great on the before captured data! Thank you very much!

ai-xyz, 2019-12-03:
So i plugged multiple times the battery in and out and it is still sending as first to bytes 0x4909,
I also did some tests with colder temperatures and in °F (because this is more accurate in 1 unit steps as °C), you can find them [here](https://triq.net/bitbench?c=%7B30%7D%2049%2009%202a%2004%20%20%5B44%C2%B0F%5D&c=%7B30%7D%2049%2009%202a%2010%20%20%5B44%C2%B0F%5D&c=%7B30%7D%2049%2009%202a%201c%20%20%5B44%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20f8%20%20%5B44%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20a8%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20c4%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20c8%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20b0%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20bc%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20c8%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20c4%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20bc%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20b0%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20a8%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20a4%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%2098%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%20dc%20%20%5B43%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%208c%20%20%5B42%C2%B0F%5D&c=%7B30%7D%2049%2009%2029%2080%20%20%5B42%C2%B0F%5D&c=%7B30%7D%2049%2009%202b%2018%20%20%5B47%C2%B0F%5D&c=%7B30%7D%2049%2009%2024%200c%20%20%5B25%C2%B0F%5D&c=%7B30%7D%2049%2009%2022%2074%20%20%5B20%C2%B0F%5D&c=%7B30%7D%2049%2009%2020%20f4%20%20%5B15%C2%B0F%5D&c=%7B30%7D%2049%2009%201f%2010%20%20%5B9%C2%B0F%5D&c=%7B30%7D%2049%2009%2019%20a8%20%20%5Blow%20temp%20(freezer)%5D&c=%7B30%7D%2049%2009%2016%20a4%20%20%5Blow%20temp%20(freezer)%5D&c=%7B30%7D%2049%2009%202b%2000%20%5B8%C2%B0C%5D&f=16x14d).
Seems like (if we talk about the 13 bits as digit) you are correct with the *0.1 factor

gdt, 2023-09-30:
@ai-xyz Are you still making progress?  Is this something you could start to prepare a PR for?

gdt, 2024-08-04:
@ai-xyz Are you still out there?  I am marking this inactive due to lack of response.