merbanan/rtl_433 #1226: Southern Hemisphere flag for Weather Stations

Zodiac-69, 2019-12-04:
Is it possable to add a "flag" to indicate that the Weather Station is in the Southern Hemisphere?
I use this for my WH24 here in Cape Town South Africa.
// Correction for Southern Hemisphere
if(windDIR >= 180)
{
    windDIR = (windDIR -180);
}
else
{
    windDIR = (windDIR +180);
}

zuckschwerdt, 2019-12-05:
rtl_433 only outputs the raw data from a sensor transmission. There are no extra conversion steps apart from the bare minimum unit conversion.
Perhaps you can do this mapping when consuming the data, i.e. on database insert or screen display?

Zodiac-69, 2019-12-06:
Thank you for the response.
I am still new with domoticz plug-in, so i shall see if i can do a plug-in to decode the rtl_433 instead of using the included rtl_433 where you have to edit the source code and compile domoticz.

closed 2019-12-06 by Zodiac-69