merbanan/rtl_433 #1227: Adding Luxorparts smoke detector  21130

Net-time, 2019-12-04:
[image]
Hi!
Decoded with _
"n=deckare,m=OOK_PPM,s=1050,l=2600,r=30000,g=6000,repeats>=3,tolerance=400,bits=25,
invert"
Actual signal, "sync" pulse 300, gap 6700, "Data: Trig pulses 700-1000, gap "1" 800-1300, gap "0" 2500-2700", gap to next sync pulse 17000.
First trig pulse a bit short (700), the rest are 800-1000.
Have also observerved the initial sync pulse missing at train start.
Tests done with the "test" button, have not tested with fire yet but I presume no diffrence.
According to the manual no option for "sync" with OOK_PPM?
As far as I can make out they just send their fixed id or the cloned id of the master.
1100011111111010110011011 unit one
1111011010110110110000001 unit two
Only got two so not much to extract a "match" on, first two bits perhaps?
Hopefully there is a crc in there somewhere?
At least they do not give a false positive.
Doable?

zuckschwerdt, 2019-12-05:
No Sync-Option on PPM, sorry.

Your description looks nice. Add the text and the flex decoder as [conf](https://github.com/merbanan/rtl_433/tree/master/conf) file if you like. (then we know there is at least one device that needs sync on PPM ;) )

If you have Pulseview (from Sigrok) installed try [examples/sigrok-open.sh](https://github.com/merbanan/rtl_433/blob/master/examples/sigrok-open.sh) to see what rtl_433 "sees".

merbanan, 2020-01-30:
No activity for a while and the device should be added as a flex config.

closed 2020-01-30 by merbanan