merbanan/rtl_433 #1231: Adding Rubicson 48686(sensor) Meat Thermometer

Net-time, 2019-12-10:
First of all this first gives a false positive as a Thermopro TP11 Thermometer (nonsense temperature)
Then it falls in line as Rubicson-48659.
The pulse lengths are a bit shorter though. so perhaps the tolerance is a bit wide for 48659.
rtl_433 -X "n=48686,m=OOK_PPM,s=600,l=1600,r=5000,g=3400,repeats>=3"

"-R 133"  does not work with Domoticz  , I need to use "-R -84 -R 133"
off topic but probably because of the "false positive" (works with rtl_433)

Due to it's function it's not suitable for fast changing measurments as it will send almost continuously.
Which will shorten battery life and block other sensors. (Not really a good sensor but it was on sale :0 )

zuckschwerdt, 2019-12-10:
So the Rubicson-48686 works with the Rubicson-48659 decoder and shows correct values?
Can you provide a captured sample (maybe also one that false-positives as TP11).
Use `-S all`, check the sample with `-r` then upload here as zip. Thanks!

Net-time, 2019-12-11:
As I said it "falls in line" so I can only get false positive and the correct one as one file.
As I've had time to sleep on it, I reckoned that the false positive was because of a truncated message at start, as I have on one of my OPUS sensors, but was wrong there too. It is because of a preamble with 3 short and one gap.
[48686.cu8.zip](https://github.com/merbanan/rtl_433/files/3951544/48686.cu8.zip)
[image]

merbanan, 2020-02-09:
Ok, added integrity checks for Rubicson and Thermopro sensors. The duplicates are gone. Thanks for the bug report.

closed 2020-02-09 by merbanan