merbanan/rtl_433 #1232: [Q] Are signals decoded and interpreted differently for different devices?

wsw70, 2019-12-16:
*This is one of my first attempts to work on RF 433 signal so I apologize in advance if the question is dumb, I hope it makes sense at least. Please let me know if this is not the right forum to ask the question. I also added an introduction for the context but the essence of my question is more or less "when I see decoded and interpreted signals, is the interpretation the same for different devices"?*

As part of my home automation project, I have simple 433 MHz wall switches ([example](https://fr.aliexpress.com/item/32877220475.html?spm=a2g0o.productlist.0.0.4fce6c886ZrnuT&algo_pvid=a4694498-e888-4807-a39e-0dce944a1813&algo_expid=a4694498-e888-4807-a39e-0dce944a1813-0&btsid=05ccf57c-6c47-4337-a01b-ed1a807bd4f0&ws_ab_test=searchweb0_0,searchweb201602_7,searchweb201603_53)). The general idea is to receive a signal from them, match a known "physical switch <--> signal" entry and decide on  some actions based on that (I use Home Assistant and AppDaemon to automate this process).

Until today I used a [Sonoff RF Bridge](https://sonoff.tech/product/accessories/433-rf-bridge) flashed with [ESPurna](https://github.com/xoseperez/espurna). It was receiving the signal and pushing it raw to MQTT. Via some experimentation I found out the piece of the signal which is immutable for a given switch (its ID of sorts).

Today I plugged in a [E4000 USB DVB-T + RTL-SDR Realtek RTL2832U + R820T](https://fr.aliexpress.com/item/33010970179.html?spm=a2g0s.9042311.0.0.3e5f6c37dfpRMI), paired it with [RTL433-to-mqtt](https://github.com/mverleun/RTL433-to-mqtt) and started observing  the messages.

The first thing I see is that I receive more messages than with the Sonoff. This is probably because the RTL-SDR has better reception and/or decodes on more frequencies.

The second and more worrisome part is that the messages I see are interpreted in a surprising way. For instance this is what I see when I press the button of a wall switch:

```
{
  "time": "2019-12-16 18:16:07",
  "model": "Smoke detector GS 558",
  "id": 12473,
  "unit": 31,
  "learn": 0,
  "code": "16173f"
}
```

The message on the Sonoff for the same switch is `21980118037AFCE868`. My experience with the Sonoff messages led me to believe that the last 6 characters of the message are unique to the switch, `FCE868` in the case above.

I do not see this sequence in the RTL-SDR output.

My question: when decoding a signal, are the elements interpreted based on a specific device, or do they follow a standard?

In other words and for the case above: are `id`, `unit`, `learn` and `code` specific to the device `Smoke detector GS 558` (which my switch is not), or do all device carry these information?

Can I make sense of the information (probably `code`) to discriminate between switches? Or would I need to somehow access the raw data (similarly to the Sonoff message) , and make sense of it myself?(for instance by pressing a switch several times and discover the part which does not change and which should be the "id" of the switch)?

osborne82, 2019-12-16:
Just set your home assistant sensor to use value_template and extract
either the unique code or id or both of your switch's then u carry on with
what ever you want ha to do

wsw70, 2019-12-16:
> extract either the unique code or id or both of your switch

My question was exactly about the 

- uniqueness of such codes (= can they be used as the ID of the device?)
- standardization of the fields (= are the same fields available for all devices, and do they have the same meaning)

How to actually use them in AppDaemaon/HA is not a problem - I do this already today with the data I get from the Sonoff RF Bridge.

zuckschwerdt, 2019-12-16:
There is some info here: https://triq.org/rtl_433/DATA_FORMAT.html
Generally we aim to have a common set of fields per device type (e.g. weather station).
`id` is almost always available. The smoke detector output is a false positive, the on-the-air protocol is the same (x1527 type) but the decoding isn't helpful here.
You likely need to analyze your devices (`-A`)  and write a config file, see the [conf folder](https://github.com/merbanan/rtl_433/tree/master/conf) for examples.

wsw70, 2019-12-17:
@zuckschwerdt thank you for the comment. Is there a guide about how to move from 

the data in the analysis output, in my case

```
Detected OOK package    2019-12-17 15:52:09
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-12-17 15:52:09
model     : Smoke detector GS 558                  id        : 12473
unit      : 31           learn     : 0             Raw Code  : 16173f
Analyzing pulses...
Total count:  187,  width: 5720.79 ms           (1430198 S)
Pulse width distribution:
 [ 0] count:    2,  width: 34608 us [32112;37104]       (8652 S)
 [ 1] count:    1,  width: 586536 us [586536;586536]    (146634 S)
 [ 2] count:    1,  width: 1165736 us [1165736;1165736] (291434 S)
 [ 3] count:    3,  width: 199652 us [182756;226800]    (49913 S)
 [ 4] count:    1,  width: 150336 us [150336;150336]    (37584 S)
 [ 5] count:    2,  width: 66360 us [62392;70332]       (16590 S)
 [ 6] count:    1,  width:  600 us [600;600]    ( 150 S)
 [ 7] count:    1,  width: 2754280 us [2754280;2754280] (688570 S)
 [ 8] count:   91,  width:  924 us [884;948]    ( 231 S)
 [ 9] count:   84,  width:  320 us [312;336]    (  80 S)
Gap width distribution:
 [ 0] count:    7,  width:   48 us [44;56]      (  12 S)
 [ 1] count:    4,  width:   64 us [56;76]      (  16 S)
 [ 2] count:    7,  width: 8700 us [8404;8840]  (2175 S)
 [ 3] count:   91,  width:  268 us [256;280]    (  67 S)
 [ 4] count:   77,  width:  840 us [816;860]    ( 210 S)
Pulse period distribution:
 [ 0] count:    2,  width: 34656 us [32156;37156]       (8664 S)
 [ 1] count:    1,  width: 586584 us [586584;586584]    (146646 S)
 [ 2] count:    1,  width: 1165792 us [1165792;1165792] (291448 S)
 [ 3] count:    3,  width: 174220 us [150408;189456]    (43555 S)
 [ 4] count:    2,  width: 66416 us [62456;70380]       (16604 S)
 [ 5] count:    1,  width:  648 us [648;648]    ( 162 S)
 [ 6] count:    1,  width: 2754356 us [2754356;2754356] (688589 S)
 [ 7] count:    1,  width: 235204 us [235204;235204]    (58801 S)
 [ 8] count:  168,  width: 1180 us [1136;1220]  ( 295 S)
 [ 9] count:    6,  width: 9068 us [8940;9164]  (2267 S)
Level estimates [high, low]:  15912,      6
RSSI: -0.1 dB SNR: 34.2 dB Noise: -34.4 dB
Frequency offsets [F1, F2]:    4722,      0     (+18.0 kHz, +0.0 kHz)
Guessing modulation: No clue...
```

to the configuration file, such as the one below which seems to be simple (my device is very simple too, it just sends the same identification code each time)

```
decoder {
    name        = Steffen-Switch,
    modulation  = OOK_PPM,
    short       = 370,
    long        = 750,
    gap         = 1080,
    reset       = 6000,
    bits        = 25,
    get         = @0:{5}:code,
    get         = @20:{4}:button:[14:A 13:B 11:C 7:D 15:ALL],
    get         = @16:{4}:state:[15:OFF 0:ON],
}
```

I guess that the `@` part is the offset of the bytes in the raw message (which I do not see in the first output) and attaches a meaning to it. For the others I am clueless.

zuckschwerdt, 2019-12-17:
We have some advanced guides in the wiki, but we still need a simple guide for switches and doorbells.

Note:

    Pulse width distribution:
     [ 8] count:   91,  width:  924 us [884;948]    ( 231 S)
     [ 9] count:   84,  width:  320 us [312;336]    (  80 S)
    Gap width distribution:
     [ 3] count:   91,  width:  268 us [256;280]    (  67 S)
     [ 4] count:   77,  width:  840 us [816;860]    ( 210 S)
    Pulse period distribution:
     [ 8] count:  168,  width: 1180 us [1136;1220]  ( 295 S)
     [ 9] count:    6,  width: 9068 us [8940;9164]  (2267 S)

It's PWM with 924 pulse + 268 gap, 320 pulse + 840 gap (for about 1180 total in both cases) symbols and an inter-packet gap of 9068.

This should do

    decoder {
        name        = YOUR-SWITCH,
        modulation  = OOK_PWM,
        short       = 320,
        long        = 924,
        gap         = 990,
        reset       = 9900,
        bits        = 25,
    }

merbanan, 2020-01-30:
I consider the raised issue answered. Closing.

closed 2020-01-30 by merbanan