merbanan/rtl_433 #1233: rtl_433 no longer receives messages

grilli303, 2019-12-17:
Hi there, hope someone can help me....
you can answer to me in English or German...

I used rtl_433 (version 18.12-262-g74b9a82) a few weeks an a Raspberry Pi3.

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

since two days I get any messages, when I start the program.

pi@raspi3:~ $ rtl_433 -G
rtl_433 version 18.12-262-g74b9a82 branch master at 201908151427 inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

Use -G for testing only. Enable protocols with -R if you really need them.

Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.
Registered 128 out of 134 device decoding protocols [ 1-4 6-8 10-17 19-26 29-64 67-134 ]
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers

after a minute or two I get this message:

pulse_FSK_detect(): Maximum number of pulses reached!

What does this message means ???

I know this dongle is not the best, but it worked a few weeks 24/7
the only problem with this device was it hangs up sometimes....
I get this message when it hungs up.

Async read stalled, exiting!

Question: what USB donges do you use with good Experience ?

I checked the USB DVB-T stick on a Win 7 computer using rtl_433, with the same result.
My first idea was that the reciever is broken, but with a SDR software it seems to work.
I also have a RTL.SDR.COM V.3 Device and tried this with rtl_433
(the RTL.SDR.COM V.3 works with a SDR software too)

I also tried with the RTL-SDR.com V3
hrere the result:

pi@raspi3:~ $ rtl_433 -G
rtl_433 version 18.12-262-g74b9a82 branch master at 201908151427 inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

Use -G for testing only. Enable protocols with -R if you really need them.

Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.
Registered 128 out of 134 device decoding protocols [ 1-4 6-8 10-17 19-26 29-64 67-134 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers

after 2-3 Minutes I get the same message:

pulse_FSK_detect(): Maximum number of pulses reached!

So I tried also with different 433 MHz antennas, so it isn't an antenna issue.
Does anyone have a good idea, what the problem could be ?
I think two broken devices are unlikely....
What options Do I have, to analize whats wrong ?

Kind regards, Daniel

grilli303, 2019-12-18:
I found the problem with help from **Benjamin Larsson** via google groups. 
Thanks for helping me, Benjamin !

I analized the frequency of one of my sensors with SDR Console V3

[image]

As you can see the exact frequency of my sensor is 433.915 MHz.
the standart frequency of rtl_433 is 433.920 MHz.
So I tried to run rtl_433 with 433.915 MHz and Bingo :)

[image]

So i find it's still strange, that the Dongle runs for weeks with 433.920 MHz...

Kind regards, Daniel

merbanan, 2019-12-18:
Hitting the center frequency directly might be bad for reception. That is one of the reasons.

grilli303, 2019-12-18:
I'm not sure, if I understood your comment correctly.
So you mean I should NOT tune to the center frequency of my device I want to receive ?
In the past I used 433.920 MHz, but I had to change because I didn't receive suddenly nothing...

I want to say thank you, for this great Software Tool you create, I love it.

merbanan, 2019-12-18:
Yes, add or subtract an offset to this. If you hit the center spike exactly you will most likely not be able to receive those signals. Signals drift based on temperature, maybe that is the reason.

closed 2020-01-11 by zuckschwerdt