merbanan/rtl_433 #1234: Add option to MQTT settings to send payload as single JSON per device

borpin, 2019-12-17:
This is a fantastic package - just wonderful!!!!

I could not see that this was possible - if it is accept my apologies.

Could I ask for an option to send the data to a single MQTT topic as a JSON payload please?

It is actually a more efficient method of posting the data.  As the time stamp applies to all the data is also makes logical sense.

merbanan, 2019-12-17:
https://github.com/merbanan/rtl_433/wiki/Useful-commands

That should answer your question.

closed 2019-12-17 by merbanan

borpin, 2019-12-17:
Sorry 😢 .

merbanan, 2019-12-17:
Did it work?

zuckschwerdt, 2019-12-17:
The `events` option of the MQTT output may be what you are looking for.

Just run the MQTT output without any specific options and look at `rtl_433/<hostname>/events`.

See https://triq.org/rtl_433/OPERATION.html#mqtt-output if you want a custom topic.

borpin, 2019-12-18:
> Did it work?

Yes it did. I also took examples from the OpenHAB doc as well.

borpin, 2019-12-18:
> See https://triq.org/rtl_433/OPERATION.html#mqtt-output if you want a custom topic.

That is good documentation, I didn't find a link to it anywhere.

Cheers

zuckschwerdt, 2019-12-18:
Yes, docs could be more prominent. Where did you look / where did you expect to find something? We should put some text or a link there. Thanks.

borpin, 2019-12-18:
> where did you expect to find something

Near the top of the ReadMe, just under *The official source code is in...* line.
I'd take out the output from the `rtl_433 -h` command but leave in the devices supported section.

The example commands need to match the docs (didn't check if they do) and the Wiki should also direct you to those docs (and possibly be depreciated as having 2 'docs' is a pain). You could use the Wiki for solution recipies (such as the OpenHAB info).

Just my suggestions 0.02c worth :grinning:.

Great project.