merbanan/rtl_433 #1235: Debian build fails (warnings and two breaking errors)

wsw70, 2019-12-18:
I tried to build `rtl_433` in a docker image but the build issues warnings and breaks twice. I am not sure if I should open separate issues for each problem - please let me know and I will break this down into unrelated issues if needed.

The image is based on Debian (buster). The `Dockerfile` is

```
FROM debian
RUN apt update && apt install -y libtool libusb-1.0-0-dev librtlsdr-dev rtl-sdr build-essential autoconf cmake pkg-config git
RUN git clone --depth 1 https://github.com/merbanan/rtl_433.git
WORKDIR rtl_433
RUN autoreconf --install
RUN ./configure
RUN make && make install
```

The output issues warnings and finally breaks:

```
Step 1/7 : FROM debian
 ---> 67e34c1c9477
Step 2/7 : RUN apt update && apt install -y libtool libusb-1.0-0-dev librtlsdr-dev rtl-sdr build-essential autoconf cmake pkg-config git
 ---> Using cache
 ---> d34956aea2b3
Step 3/7 : RUN git clone --depth 1 https://github.com/merbanan/rtl_433.git
 ---> Using cache
 ---> 7c218fe78868
Step 4/7 : WORKDIR rtl_433
 ---> Using cache
 ---> 067ffc51a822
Step 5/7 : RUN autoreconf --install
 ---> Using cache
 ---> 148ef978c62a
Step 6/7 : RUN ./configure
 ---> Using cache
 ---> eebdcc4a3aee
Step 7/7 : RUN make && make install
 ---> Running in b419dac3da57
echo UNKNOWN > .version-t && mv .version-t .version
make  all-recursive
make[1]: Entering directory '/rtl_433'
Making all in include
make[2]: Entering directory '/rtl_433/include'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/rtl_433/include'
Making all in src
make[2]: Entering directory '/rtl_433/src'
  CC       abuf.o
  CC       am_analyze.o
  CC       baseband.o
  CC       bitbuffer.o
  CC       compat_paths.o
  CC       compat_time.o
  CC       confparse.o
  CC       data.o
  CC       decoder_util.o
  CC       fileformat.o
  CC       list.o
  CC       mongoose.o
mongoose.c: In function 'mg_send_ssi_file.part.42':
mongoose.c:9525:40: warning: 'snprintf' output may be truncated before the last format character [-Wformat-truncation=]
     snprintf(path, sizeof(path), "%s/%s", opts->document_root, file_name);
                                        ^
mongoose.c:9525:5: note: 'snprintf' output 2 or more bytes (assuming 4097) into a destination of size 4096
     snprintf(path, sizeof(path), "%s/%s", opts->document_root, file_name);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC       optparse.o
  CC       output_influx.o
  CC       output_mqtt.o
  CC       pulse_demod.o
  CC       pulse_detect.o
  CC       pulse_detect_fsk.o
[... 203 lines cut ...]
  CC       devices/tpms_elantra2012.o
  CC       devices/tpms_ford.o
  CC       devices/tpms_jansite.o
  CC       devices/tpms_pmv107j.o
  CC       devices/tpms_renault.o
  CC       devices/tpms_toyota.o
  CC       devices/ts_ft002.o
  CC       devices/ttx201.o
  CC       devices/vaillant_vrt340f.o
  CC       devices/waveman.o
  CC       devices/wg_pb12v1.o
  CC       devices/wssensor.o
  CC       devices/wt0124.o
  CC       devices/wt450.o
  CC       devices/x10_rf.o
  CC       devices/x10_sec.o
  CCLD     rtl_433
libtool:   error: require no space between '-L' and '-lrtlsdr'
make[2]: *** [Makefile:976: rtl_433] Error 1
make[2]: Leaving directory '/rtl_433/src'
make[1]: *** [Makefile:497: all-recursive] Error 1
make[1]: Leaving directory '/rtl_433'
make: *** [Makefile:383: all] Error 2
Error response from daemon: The command '/bin/sh -c make && make install' returned a non-zero code: 2
```

zuckschwerdt, 2019-12-18:
We recommend building with CMake. Autotools might be broken and we don't really support it. PR with fixes much welcome.
What you are seeing might be a problem with pkg-config.

wsw70, 2019-12-18:
@zuckschwerdt Thank you Christian for the hint. 

With CMake the build is seamless (it is even surprising that the warnings are gone). If there is interest I will make a PR with instructions for docker (and probably a Docker Hub version based on a nightly build)

closed 2019-12-18 by wsw70