merbanan/rtl_433 #1236: Async read stalled, exiting!

xKronyx, 2019-12-19:
On ubuntu 19.04
I have this error since 2 weeks after running for about 20 minutes

Async read stalled, exiting!

It was working since about 6 months without any issues

Thanks !

zuckschwerdt, 2019-12-19:
Your rtlsdr receiver is bad or having intermittent problems. I encounter this on higher sample rates (more than 1M) due to heat problems. I "roasted" at least one cheap dongle with permanent damage. Add heatsinks or use something like the Nooelec nesdr smart with a metal body.

If the device still mostly works use a supervisor to restart rtl_433. http://triq.net/articles/rtl-433-networking

closed 2019-12-19 by zuckschwerdt

xKronyx, 2019-12-19:
It's a NooElec NESDR SMArt - Premium RTL-SDR w/ Aluminum Enclosure, running for 433 mhz dvices. I'll take a look at that, thanks for your help !

zuckschwerdt, 2019-12-19:
Never had a problem with the Nooelec, only with the cheapo blue or black plastic ones.
Could be a power supply issue then. The error message is definitly the dongle stopping to work.