merbanan/rtl_433 #1237: use of -Y classic flag and -H time flag

bronze51, 2019-12-23:
With “rtl_433 -f 915M -M 76 -Y classic” and “rtl_433 -f 433M -Y classic -M 12”, I pick up signals from my weather sensors.

But with “rtl_433 -f 433M -f 915M -M 12,76 -Y classic -H 240”, I do not

zuckschwerdt, 2019-12-23:
Your option flags are not compatible with rtl_433. Why do you use `-M` instead of `-R`?
Also `-M 12,76` won't work. It should be `-R 12 -R 76`.

closed 2019-12-23 by zuckschwerdt

bronze51, 2019-12-26:
Thanks for the detailed response. Sorry for the incorrect description,  I did the tests again. With "rtl_433 -f 433M -Y classic" (I don't need the "classic" flag on 433M) and with "rtl_433 -f 915M -Y classic" signals are decoded. But with "rtl_433 -f 433M -f 915M -Y classic -H 360" signals on 433M are not decoded.