merbanan/rtl_433 #1243: use of -Y classic flag and -H time flag

bronze51, 2019-12-28:
With "rtl_433 -f 433M -Y classic" (I don't need the "classic" flag on 433M) and with "rtl_433 -f 915M -Y classic" signals are decoded. But with "rtl_433 -f 433M -f 915M -Y classic -H 360" signals on 433M are not decoded.

_Originally posted by @bronze51 in https://github.com/merbanan/rtl_433/issues/1237#issuecomment-568964596_

zuckschwerdt, 2019-12-28:
Thanks! Yes, you can use either the classic demod (default for 433) or the minmax demod (default for 915). Could be that the classic demod is not selected despite the `-Y classic` option when there are multiple frequencies given? Needs some investigation ;)

bronze51, 2019-12-28:
Just to make sure my report is clear. 

- On 433M I receive information from "Oregon Scientific Weather Sensor"/device protocol 12. 

- On 915M I receive information from "LaCrosse TX29IT Temperature sensor"/device protocol 76 with "classic" flag (see https://github.com/merbanan/rtl_433/issues/1204#issue-527495240).

My problem:
> But with "rtl_433 -f 433M -f 915M -Y classic -H 360" signals on 433M are not decoded.

So, the classic flag overrides the default "demod" on 915M correctly as signals are decoded. However, no signals are decoded on 433M (where classic flag should be ignored/have no effect?).

peteakalad, 2020-01-27:
Also after updating rtl_433 it wasn't decoding my doorbell using this;
`rtl_433 -d 1 -F mqtt://localhost:1883 -M utc -f 868221000 -R 116`

After some diagnosing seems I need to change the FSK detector to classic.
`rtl_433 -d 1 -F mqtt://localhost:1883 -M utc -f 868221000 -R 116 -Y classic`

May be worth mentioning in docs / --help that the new FSK detector will be used over FSK_PULSE_DETECTOR_LIMIT (800Mhz currently).

Not sure about the purpose of the new detector, but its broken for my device/uses.

Sample, Honeywell ActivLink, Wireless Doorbell (FSK) 
[g005_868.221M_1000k.cu8.zip](https://github.com/merbanan/rtl_433/files/4118953/g005_868.221M_1000k.cu8.zip)

merbanan, 2020-02-16:
This is a current limitation with regards to fsk-bases signal handling. It will be fixed eventually but in the mean time you could work around this by periodically running with different options.

So the issue is not really an issue but the current design. Not the best one but it is what we have.

closed 2020-02-16 by merbanan