merbanan/rtl_433 #1247: incorrect support for OWL CM180

dgrostoto, 2020-01-02:
I recently plugged a CM180 with RTL_433 and i figured out that support is incorrect for this sensor.
-The sequence number (start from 4 and decrease to 0) is presently interpreted as the ID, so rtl_433 shows 5 differents devices instead of one.
-the actual ID is not reported at all
-the battery level is not reported (i have to check that battery level is correctly sended and is not a fixed value).
-there is some flags that are not reported (but i don't know what it is about).

I'm working on a fix for the three first problems and will publish, as soon as it works OK at home. I hope that you would kindly integrate it

zuckschwerdt, 2020-01-02:
Great! Not sure if the CM180 support was ever thoroughly tested -- use [Git blame](https://github.com/merbanan/rtl_433/blame/master/src/devices/oregon_scientific.c) if you want to know the history of the decoder.

dgrostoto, 2020-01-03:
I also found that secondary frames (sended by the hardware device when power consumption does vary a lot and relatively quickly) are ignored/dropped whereas they were correctly reported in older versions (regression). I will correct that too.
=> Seems to be a problem is oregon_v3_decoder concerning specific line of code for CM180 device

dgrostoto, 2020-01-03:
Here it is the fix for CM180 support :
- added  battery level report OK/LOW (method is specific for this device)
- corrected watt correction for 230V (same power wattage than the display when the display is set with 230V factor is 1.00625)
[oregon_scientific_03012020_release.c.txt](https://github.com/merbanan/rtl_433/files/4020941/oregon_scientific_03012020_release.c.txt)

- now report correct ID for this sensor
- put ID to sequence number (used same value as ikea sparsnas which the same function).
- Now report secondary frames (instant power usage) again which was broken recently.

Note : CM160 support is probably in the same situation but since i don't own any, i cannot be sure so i left CM160 support untouched.

dgrostoto, 2020-01-03:
Thanks in advance for your integration

dgrostoto, 2020-01-08:
Hello, maybe i did not it the right way ? may i have to fork the repository or something like that to facilitate integration ?

zuckschwerdt, 2020-01-09:
Wait a few more days.

closed 2020-01-09 by zuckschwerdt

zuckschwerdt, 2020-01-10:
Well, thanks for finding, fixing, and reporting the issue :)