merbanan/rtl_433 #1248: Add Blineline PowerCost Monitor

gcormier, 2020-01-02:
Documentation : https://static.hydroottawa.com/documents/residential/conservation/pdf/peaksaver_user_guides/8-Energy-Display-Installation-Guide-EN.pdf

I'm a bit confused as there is reference to rtl_433 ( https://github.com/CapnBry/Powermon433/blob/master/arduino/Powermon433/Powermon433.ino ) but it's not a supported device. Perhaps the author meant he used rtl_433 to do the decoding.

I'm willing to capture the samples and submit to test repo. Based on the code above, I am using this

`-X "name=blueline,modulation=OOK_MC_ZEROBIT,short=500,long=1000,r=58000"`

Does that seem accurate?

gcormier, 2020-01-02:
FCC : Data obtained by an IR-reader/sensor is transmitted in form of short bursts every 28.5 to 31.5 seconds on 433.92 MHz (single frequency). The carrier is On/Off pulse modulated (logic ‘1’: 0.5ms TX-on followed by 2ms TX-off, logic ‘0’: 0.5ms TX- on followed by 4ms TX-off).

merbanan, 2020-01-03:
https://github.com/radredgreen/rtl_433/commits/master

merbanan, 2020-01-03:
http://scruss.com/blog/2013/12/03/blueline-black-decker-power-monitor-rf-packets/

So you can just clean up the old code from the above branch.

gcormier, 2020-01-03:
I think I have a valid packet captured. The ON period is 2ms, not 0.5ms as described in the FCC filing, but it has matching OFF periods of 2ms or 4ms.

Can I ask why these produce the same results? Swapping short and long values in the decoder string?

````
greg@biglenny:~/rtl_433_tests/tests/blueline/lime-outside$ rtl_433 -r g010_433.92M_1000k.cs16 -X "name=blueline,modulation=OOK_PPM,short=2000,long=4000,gap=7000,reset=350000"
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.134372s
model     : blueline     count     : 2             num_rows  : 2             rows      : 
len       : 32           data      : 010c99a8, 
len       : 0            data      : 
codes     : {32}010c99a8, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.549692s
model     : blueline     count     : 2             num_rows  : 2             rows      : 
len       : 32           data      : 01f2bcd5, 
len       : 0            data      : 
codes     : {32}01f2bcd5, {0}
````

````
greg@biglenny:~/rtl_433_tests/tests/blueline/lime-outside$ rtl_433 -r g010_433.92M_1000k.cs16 -X "name=blueline,modulation=OOK_PPM,short=2000,long=4000,gap=7000,reset=350000"
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.134372s
model     : blueline     count     : 2             num_rows  : 2             rows      : 
len       : 32           data      : 010c99a8, 
len       : 0            data      : 
codes     : {32}010c99a8, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.549692s
model     : blueline     count     : 2             num_rows  : 2             rows      : 
len       : 32           data      : 01f2bcd5, 
len       : 0            data      : 
codes     : {32}01f2bcd5, {0}
````

zuckschwerdt, 2020-01-03:
That example might get the wrong results, you do need `-s 1M`. Both examples look the same? Copy/paste error?

gcormier, 2020-01-03:
Sorry, I made a copy/paste error.

One was `short=2000,long=4000`, the other was `short=4000,long=2000`, but the output is the same. But it should be interpreting the bits differently (inversely), no?

zuckschwerdt, 2020-01-03:
For PPM short and long will only result in a threshold (6000 here) if no tolerance value is used.

jbrzozoski, 2020-12-24:
Blueline PowerCost monitor added by #1580

gcormier, 2021-01-18:
I'm unable to enable this filter for rtl_433:

`~/rtl_433/build/src/rtl_433 -R 176`

Then the output interestingly:

`Registered 1 out of 176 device decoding protocols [ ]`

Adding a second random protocol with `-R 1`:

`Registered 2 out of 176 device decoding protocols [ 1 ]`

merbanan, 2021-01-18:
rtl_433 -R 176:auto should give you the key. And then rtl_433 -R 176:"key" should work.

gcormier, 2021-01-18:
I've read the PR to try and get a better understanding of what's happening.

I've run it with `-R 176:auto`, it picked up the first few - then I ran outside and pressed the button on the device. None of these seem to be the key?

```
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:07:40
model     : Blueline PowerCost Monitor             id        : 12080
gap       : 5281         mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:08:12
model     : Blueline PowerCost Monitor             id        : 12080
impulses  : 531          mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:08:12
model     : Blueline PowerCost Monitor             id        : 12080
impulses  : 531          mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:08:12
model     : Blueline PowerCost Monitor             id        : 12080
gap       : 5489         mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:08:22
model     : Blueline PowerCost Monitor             id        : 12080
mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:08:23
model     : Blueline PowerCost Monitor             id        : 12080
mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:08:23
model     : Blueline PowerCost Monitor             id        : 12080
mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:08:23
model     : Blueline PowerCost Monitor             id        : 12080
mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:08:27
model     : Blueline PowerCost Monitor             id        : 12080
mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:08:27
model     : Blueline PowerCost Monitor             id        : 12080
mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:08:27
model     : Blueline PowerCost Monitor             id        : 12080
mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:08:27
model     : Blueline PowerCost Monitor             id        : 12080
mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:09:13
model     : Blueline PowerCost Monitor             id        : 12080
gap       : 5617         mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:09:14
model     : Blueline PowerCost Monitor             id        : 12080
gap       : 5617         mic       : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2021-01-18 10:09:14
model     : Blueline PowerCost Monitor             id        : 12080
gap       : 5617         mic       : CRC
```

gcormier, 2021-01-18:
Aha, note to future people following the breadcrumps - read the comments in the source file for an exhaustive description of steps to follow and what is going on, as well as how to calculate kWh - https://github.com/merbanan/rtl_433/blob/master/src/devices/blueline.c

gcormier, 2021-01-18:
And, here it is pulled in to Home Assistant, using rtl_433 with the MQTT function.

```
  - platform: mqtt
    name: "Power Consumption"
    state_topic: "rtl_433/Blueline_PowerCost_Monitor/12080/gap"
    unit_of_measurement: 'kWh'
    value_template: "{{ '%.2f'%(3600 / (value | float)) | float }}"
```

[image]

merbanan, 2021-01-18:
s/power/energy/

zuckschwerdt, 2021-01-18:
[image]

bkenobi, 2021-02-22:
Has anyone looked at the battery info?  Mine is returning a value of 1 with new batteries.  I'm trying to determine if this is a flag that returns 1 for "good" and 0 (or some other value) for "bad" or if it returns a remaining percentage.  I'm trying to set up a notification and right now all I have to trigger on is when battery_ok doesn't equal 1.

jbrzozoski, 2021-02-22:
I never hooked my monitor up to a variable power supply to see how the flags or battery status changed as the voltage decreased.  My implementation was purely trusting the prior work.  The old implementations only looked at one single bit for the battery, so that's all I did in mine.

bkenobi, 2021-02-22:
It looks like the code simply outputs "1" in my case.  I can review what the previous work found if you have a link.  I've looked at lots of code for this device over the last few years but never considered the battery because I never had anything working reliably (until now).

jbrzozoski, 2021-02-22:
> It looks like the code simply outputs "1" in my case. I can review what the previous work found if you have a link. I've looked at lots of code for this device over the last few years but never considered the battery because I never had anything working reliably (until now).

Some references used include:
- https://github.com/merbanan/rtl_433/pull/38 - an abandoned pull request on rtl_433 by radredgreen
- https://github.com/CapnBry/Powermon433 - a standalone Arduino-based Blueline monitor
- http://scruss.com/blog/2013/12/03/blueline-black-decker-power-monitor-rf-packets/ - the blog post where the other authors were trading notes in the comments

merbanan, 2021-03-03:
I am closing this issue now. If some one figures out the battery flag send a PR or reopen this issue.

closed 2021-03-03 by merbanan

gcormier, 2021-08-05:
Home Assistant now has Energy tracking. Power monitoring devices not yet supported, but posting this here for reference as it may help other if/when power devices become supported.

```
  - platform: mqtt
    name: "Power Consumption"
    state_topic: "rtl_433/Blueline_PowerCost_Monitor/12080/gap"
    unit_of_measurement: 'kW'
    value_template: "{{ '%.2f'%(3600 / (value | float)) | float }}"
    device_class: power
    state_class: measurement
```