merbanan/rtl_433 #1251: Help interpreting output

mahescho, 2020-01-03:
Sorry for this, but I' completely new to this stuff :-) I want to emulate a 433 remote for my fans with a raspi and a transmitter. As rtl_433 can not decode the signal by one of it's decoders I've used `rtl_433 -A -R 0` and got:

    Detected OOK package	2020-01-03 19:55:01
    Analyzing pulses...
    Total count:  192,  width: 274.20 ms		(68549 S)
    Pulse width distribution:
     [ 0] count:   54,  width:  396 us [392;416]	(  99 S)
     [ 1] count:  138,  width:  796 us [792;804]	( 199 S)
    Gap width distribution:
     [ 0] count:  138,  width:  400 us [396;408]	( 100 S)
     [ 1] count:   48,  width:  800 us [800;812]	( 200 S)
     [ 2] count:    5,  width: 9620 us [9620;9624]	(2405 S)
    Pulse period distribution:
     [ 0] count:   42,  width:  800 us [796;820]	( 200 S)
     [ 1] count:  102,  width: 1200 us [1196;1212]	( 300 S)
     [ 2] count:   42,  width: 1600 us [1596;1608]	( 400 S)
     [ 3] count:    5,  width: 10020 us [10020;10024]	(2505 S)
    Level estimates [high, low]:  15979,     36
    RSSI: -0.1 dB SNR: 26.5 dB Noise: -26.6 dB
    Frequency offsets [F1, F2]:   18478,      0	(+70.5 kHz, +0.0 kHz)
    Guessing modulation: Pulse Width Modulation with multiple packets
    Attempting demodulation... short_width: 396, long_width: 796, reset_limit: 9628, sync_width: 0
    Use a flex decoder with -X 'n=name,m=OOK_PWM,s=396,l=796,r=9628,g=816,t=160,y=0'
    pulse_demod_pwm(): Analyzer Device 
    bitbuffer:: Number of rows: 6 
    [00] {32} 90 a0 c0 29 : 10010000 10100000 11000000 00101001 
    [01] {32} 90 a0 c0 29 : 10010000 10100000 11000000 00101001 
    [02] {32} 90 a0 c0 29 : 10010000 10100000 11000000 00101001 
    [03] {32} 90 a0 c0 29 : 10010000 10100000 11000000 00101001 
    [04] {32} 90 a0 c0 29 : 10010000 10100000 11000000 00101001 
    [05] {32} 90 a0 c0 29 : 10010000 10100000 11000000 00101001 

No surprise, the values found change as I press different keys.
How to interpret this output and how to send these codes with a 433 MHz transmitter with a raspi?

TIA

zuckschwerdt, 2020-01-03:
I guess your transmitter is of the cheap and common toggle-a-pin OOK type? Then you can directly use the OOK output (`rtl_433 -w OOK:-`) to derive the timings.

mahescho, 2020-01-03:
Ok, now I get two columns of numbers as output. What to do with them? As said, I'm a complete novice to this :-)

zuckschwerdt, 2020-01-03:
Those columns are pulse (on) and gap (off) times in µs. So toggle your trasmitter on for period of each first value and off for every period of the second value.

mahescho, 2020-01-04:
Like described here ?
https://www.instructables.com/id/Super-Simple-Raspberry-Pi-433MHz-Home-Automation/
This is an representative sniplet:

    800 804
    400 404
    796 804
    400 400
    800 404
    800 400
    804 396
    804 404
    800 804
    400 804
    396 400
    804 404
    800 400
    800 400
    804 400
    800 804
    400 404
    800 400
    804 400
    800 400
    804 804
    396 400
    804 404
    800 804
    400 9616
    400 404
    800 404
    800 800

Can you please explain the values in short for better understanding.

zuckschwerdt, 2020-01-04:
Different gap lengths: likely because the (-A) decoding is wrong and this is Manchester encoding.
The last long gap is a "forever" gap, there is no pulse after it.

mahescho, 2020-01-04:
I think I found the pattern. Every shot "on" is preceded by a long gap. Correct?

zuckschwerdt, 2020-01-04:
Hard to say. It's [not Manchester](http://triq.net/bitbench?c=y101101100101101101101100101100101101101101101100100101101101101101100101101101101100101101100100000&f=v) though.

zuckschwerdt, 2020-01-11:
Let us know how you solved this. Would be an interesting project for other people.

closed 2020-01-11 by zuckschwerdt