merbanan/rtl_433 #1252: RSSI SNR and Noise could be added to standard csv output

dgrostoto, 2020-01-03:
Hello again,
I'm very interested in adding RSSI SNR and Noise information to standard csv output in addition to other data. This information can be added to the "well known" information (which seems to be  "time","msg" and "codes").
The snr/noise information : tell user about quality reception on order to help to seat sensor correctly or can be used to alarm if noise get suddenly high.
Can you tell me your feeling about this proposal ? I think i can do the modification for this.

merbanan, 2020-01-06:
rtl_433 -M level -F csv

time,msg,codes,mod,freq,freq1,freq2,rssi,snr,noise,model,button, ...

2020-01-06 16:01:57,,,FSK,,433.925,433.858,-9.647,9.112,-18.760,Toyota,,

Looks to me like it is already supported.

closed 2020-01-06 by merbanan

dgrostoto, 2020-01-07:
You're right. my bad. it Fit perfect for my use.