merbanan/rtl_433 #1253: Possible to use MQTTGateway output as input for rtl_433 ?

dgrostoto, 2020-01-03:
rtl_433 has great support for decoding various protocols and hardware support for few receivers.
MQTTGateway has great connectivity for various transceivers/receivers hardware solutions (and almost no support for decoding frames).
May the "output" of MQTTGateway could be used as an input to rtl_433 trough a MQTT broker sucj as mosquitto for example ?
In particular, is the data information produced by MQTTGateway usable or easily convertible to make it usable by rtl_433 decoders ?
If so, it could be a great end to end solution for networking RF data.

zuckschwerdt, 2020-01-03:
Can you add a link and more info to the MQTT Gateway in question?

dgrostoto, 2020-01-03:
Of course : i missed the "open" keyword of this project
https://github.com/1technophile/OpenMQTTGateway

dgrostoto, 2020-01-03:
the part to be considered in openMQTTGateway is the RF gateway part (see https://docs.openmqttgateway.com/setitup/rf.html#compatible-parts ) where no solution for decoding frame.

zuckschwerdt, 2020-01-04:
Looks like a neat project. Those 1-bit OOK receivers are a pain to work with for real SDR applications. You mostly use them for single known protocols. I guess EV1527 / PT2260 and alike.

We can't support those OOK receivers, please reopen if you hack an adapter script or find one.

closed 2020-01-04 by zuckschwerdt

dgrostoto, 2020-01-04:
OK, I understand the signal from OpenMQTTGateway is globally not suitable as is for rtl_433 analysis. I think the work in order that signal could be processed right by rtl_433 should be on the producer side (and thus not by rtl_433). Thanks for your response.

zuckschwerdt, 2020-01-04:
Yes, it's only OOK (no FSK), it does not have signal levels (only on/off). Thus you'll need heavy filtering and quite some assumptions on the protocol -- not suitable for finding arbitrary transmissions.