merbanan/rtl_433 #1254: Support for TFA 30.3222.02

mboehm21, 2020-01-04:
Hi everyone,

I have the following sensor for temperature / humidity / wind:

https://www.tfa-dostmann.de/en/produkt/temperature-humidity-wind-transmitter/

Product description says that it is sending on 433 MHz.

Can you give me some hints how I can contribute to the project to get this sensor supported by rtl_433? Which parameters should I use to read the raw-data? Is there any tool to help to reverse-engineer the protocol?

I'm using this stick as the sdr-receiver:

Thanks a lot.

mboehm21, 2020-01-04:
I created four samples using the "-S unknown" switch. Maybe this is a good starting point.

[samples.zip](https://github.com/merbanan/rtl_433/files/4022005/samples.zip)

merbanan, 2020-01-06:
Issue #1077 also reference TFA 30.3222. Anyway only the first 2 samples seems to be valid.

`-X 'n=name,m=OOK_PWM,s=256,l=500,r=1932,g=0,t=0,y=748'` will most likely give you proper bits.

2 different rows are observed though.

```
[07] {65} f3 ab 8b fe d3 2f d8 5f 80 
[08] {65} f3 ab 8b fd fb 4f ff ac 80 

```
and

```
[07] {65} f3 ab 8b fe d3 1f d5 a1 80 
[08] {65} f3 ab 8b fd fc 8f ff e5 80 
```

The payload is assumed to be 64 bits. Now that we can get the bits we need to try to figure out the pattern by varying the parameters and record the payload. Best is to get samples where only one parameters is varied. So start recording and note down the receiver values. Issue #1077 might have some hints.

mboehm21, 2020-01-06:
Thanks for your answer. So you say it would be a good idea to run some tests?

I could simulate the following situations:

- room temperature / humidity with no wind
- room temperature / humidity with simulated wind through spinning the sensor
- temperature / humidity in the fridge under 0 degree Celsius
- outside temperature / humidity with no wind

Would it help to create those traces?

merbanan, 2020-01-07:
@mboehm21 Yes, that sounds like a good strategy.

mboehm21, 2020-01-10:
Hi,

attached you find 6 traces of the signal.

Later on I'll provide more under different conditions.

[sensor.zip](https://github.com/merbanan/rtl_433/files/4046554/sensor.zip)

merbanan, 2020-01-10:
This looks good but you only need to post the payload.

mboehm21, 2020-01-10:
Okay, you mean output of `rtl_433 -X 'n=name,m=OOK_PWM,s=256,l=500,r=1932,g=0,t=0,y=748'` like this?

`_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-01-10 17:06:25
model     : name         count     : 5             num_rows  : 5             rows      : 
len       : 65           data      : f06d0afdf7efff1b8, 
len       : 65           data      : f06d0afdf7efff1b8, 
len       : 65           data      : f06d0afdf7efff1b0, 
len       : 65           data      : f06d0afdf7efff1b8, 
len       : 65           data      : f06d0afdf7efff1b8
codes     : {65}f06d0afdf7efff1b8, {65}f06d0afdf7efff1b8, {65}f06d0afdf7efff1b0, {65}f06d0afdf7efff1b8, {65}f06d0afdf7efff1b8
*** Saving signal to file g002_433.92M_250k.cu8 (195871 samples, 393216 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
`

merbanan, 2020-01-10:
Yes, but only the raw payload:

f06d0afdf7efff1b8, 14C, wind W (or something)

merbanan, 2020-01-10:
You can use http://triq.net/bitbench.

mboehm21, 2020-01-10:
- 001_inside_25.1DegCels_no_wind.cu8`

`[07] {65} f0 6d 0a fe d1 8f da 7d 80 
[08] {65} f0 6d 0a fd ff ff ff 27 80`

- 002_inside_25.2DegCels_no_wind.cu8

`[07] {65} f0 6d 0a fe d1 6f da 0b 80 
[08] {65} f0 6d 0a fd ff ff ff 27 80`

- 003_inside_25.3DegCels_slow_wind.cu8

`[07] {65} f0 6d 0a fe d1 1f d7 7c 80 
[08] {65} f0 6d 0a fd fe 8f ff 5a 80`

- 004_inside_25.4DegCels_slow_wind.cu8

`[07] {65} f0 6d 0a fe d1 0f da 5e 80 
[08] {65} f0 6d 0a fd fe ef ff 0f 80`

- 005_inside_25.4DegCels_fast_wind.cu8

`[07] {65} f0 6d 0a fe d0 ef d8 0c 80 
[08] {65} f0 6d 0a fd f4 af ff 58 80`

- 006_inside_25.4DegCels_fast_wind.cu8

`[07] {65} f0 6d 0a fe d0 bf d8 eb 80 
[08] {65} f0 6d 0a fd f7 ef ff 1b 80`

I don't have an eye for the pattern in bitbench, yet.

mboehm21, 2020-01-10:
After 1,5 h in the freezer which shows -18 °C with no wind:

- 008

`[07] {65} f0 6d 0a fe eb 9f d7 5c 80` 
`[08] {65} f0 6d 0a fd ff ff ff 27 80`

- 009

`[07] {65} f0 6d 0a fe eb 9f d7 5c 80` 
`[08] {65} f0 6d 0a fd ff ff ff 27 80`

merbanan, 2020-01-10:
Track the 2 output separately. I assume the first is temperature and the second is wind.

[TFA bitbench](http://triq.net/bitbench?c=%7B65%7D%20f0%206d%200a%20fe%20eb%209f%20d7%205c%20%20%20%23-18C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%208f%20da%207d%20%20%20%2325.1C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%206f%20da%200b%20%20%20%2325.2C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%201f%20d7%207c%20%20%20%2325.3C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%200f%20da%205e%20%20%20%2325.4C&c=%7B65%7D%20f0%206d%200a%20fe%20d0%20ef%20d8%200c%20%20%20%2325.4C&c=%7B65%7D%20f0%206d%200a%20fe%20d0%20bf%20d8%20eb%20%20%20%2325.4C&f=hh%20hh%20hh%20hh%20~h~h~h%20hhh%20CRC%3Ahh%20%0A) with all the current values.

I strongly suspect that I mapped out the temperature correct. Bytes 5+half 6 inverted. So more temperature readings are needed to verify this.

merbanan, 2020-01-10:
How to convert the values needs to be figured out also.

merbanan, 2020-01-10:
The crc is assumed. Might not be true but the temperature fields behave logical.

mboehm21, 2020-01-10:
The sensor is currently outdoors. In an hour or so I can provide values for about 8 °C.

Out of curiosity: What do vendor gain from making it so obfuscated?

merbanan, 2020-01-10:
There is no obfuscation here. This is slightly above normal level of complexity. We just have to figure out piece by piece, which means loads of payloads and corresponding readings.

mboehm21, 2020-01-10:
I see, but I am kind of new to radio and would like to understand why something like "Bytes 5+half 6 inverted" makes sense instead of sending it without inversion. Sorry, I know that a github issue is not the right place to ask such things.

merbanan, 2020-01-10:
Ahh sorry. FSK uses 2 frequencies, hi and low. You are free to map 0 -> hi and 1 ->low or vice versa. In our case we need to invert the bits to get the proper representation.

merbanan, 2020-01-10:
But in reality you should investigate this bit per bit, there are decoders where some parameters are inverted and others are not.

merbanan, 2020-01-10:
So look at bits until the pattern emerge is the recipe.

mboehm21, 2020-01-10:
Makes sense, thank you. And this can change inside of one transmission or does it generally stay the same?

merbanan, 2020-01-10:
The value encoding is fixed for this kind of simple devices.

mboehm21, 2020-01-10:
Okay and why don't we invert the whole 64 bits then?

merbanan, 2020-01-10:
One bit at the time. You are most likely correct but focus on the temperature. Inverting the bits in bitbench is easy. Figuring out the temperature and humidity formula is the hard part.

merbanan, 2020-01-10:
Or collecting all the the payloads is the hard part.

mboehm21, 2020-01-10:
Don't worry, I can provide as many payloads as we need to figure it out.

Outside for 1,5 h, ~6,5 °C, slight rain, very little wind:

- 010

`[07] {65} f0 6d 0a fe dd 0f 9e dc 80`
`[08] {65} f0 6d 0a fd ff ff ff 27 80`

- 011

`[07] {65} f0 6d 0a fe dd 1f 9e b2 80`
`[08] {65} f0 6d 0a fd ff ff ff 27 80`

[bitbench](http://triq.net/bitbench?c=%7B65%7D%20f0%206d%200a%20fe%20eb%209f%20d7%205c%20%20%20%23-18C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%208f%20da%207d%20%20%20%2325.1C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%206f%20da%200b%20%20%20%2325.2C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%201f%20d7%207c%20%20%20%2325.3C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%200f%20da%205e%20%20%20%2325.4C&c=%7B65%7D%20f0%206d%200a%20fe%20d0%20ef%20d8%200c%20%20%20%2325.4C&c=%7B65%7D%20f0%206d%200a%20fe%20d0%20bf%20d8%20eb%20%20%20%2325.4C&c=%7B65%7D%20f0%206d%200a%20fe%20dd%200f%209e%20dc%20%20%20%236.5C&c=%7B65%7D%20f0%206d%200a%20fe%20dd%201f%209e%20b2%20%20%20%236.5C&f=hh%20hh%20hh%20hh%20~h~h~h%20hhh%20CRC%3Ahh%20%0A)

mboehm21, 2020-01-10:
And a [http://triq.net/bitbench?c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%23%20no%20wind&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%23%20no%20wind&c=f0%206d%200a%20fd%20fe%208f%20ff%205a%20%23%20weak%20wind&c=f0%206d%200a%20fd%20fe%20ef%20ff%200f%20%23%20weak%20wind&c=f0%206d%200a%20fd%20f4%20af%20ff%2058%20%23%20strong%20wind&c=f0%206d%200a%20fd%20f7%20ef%20ff%201b%20%23%20strong%20wind&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%23%20no%20wind&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%23%20no%20wind&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%23%20no%20wind&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%23%20no%20wind&f=hhhhhhhhh%20WIND%3A%20~h~h%20STUFF%3A%20hhhhhhhh](bitbench) for line 2 with some wild wind guessing.

merbanan, 2020-01-10:
[wind bitbench](http://triq.net/bitbench?c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&c=f0%206d%200a%20fd%20fe%208f%20ff%205a%20%5B%23%20weak%20wind%5D&c=f0%206d%200a%20fd%20fe%20ef%20ff%200f%20%5B%23%20weak%20wind%5D&c=f0%206d%200a%20fd%20f4%20af%20ff%2058%20%5B%23%20strong%20wind%5D&c=f0%206d%200a%20fd%20f7%20ef%20ff%201b%20%5B%23%20strong%20wind%5D&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%20%5B%23%20no%20wind%5D&f=hh%20hh%20hh%20hhh%20WIND%3A%20~8d%20STUFF%3A%20h%20hh%20CRC%3A%20)

I changed it a bit. And you are probably correct. Now we just need to figure out the correct formula.

merbanan, 2020-01-10:
[temp/hum bitbench](http://triq.net/bitbench?c=%7B65%7D%20f0%206d%200a%20fe%20eb%209f%20d7%205c%20%20%20%23-18C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%208f%20da%207d%20%20%20%2325.1C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%206f%20da%200b%20%20%20%2325.2C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%201f%20d7%207c%20%20%20%2325.3C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%200f%20da%205e%20%20%20%2325.4C&c=%7B65%7D%20f0%206d%200a%20fe%20d0%20ef%20d8%200c%20%20%20%2325.4C&c=%7B65%7D%20f0%206d%200a%20fe%20d0%20bf%20d8%20eb%20%20%20%2325.4C&f=hh%20hh%20hh%20hh%20Tem%3A%20~12d%20h%20Hum%3A%20hh%20CRC%3Ahh%20%0A)

I changed the payload description. I think we have the values for all the sensor data now. Just need to figure out the encoding. Then we have channel, battery and probably CRC state also.

mboehm21, 2020-01-10:
Nice, making progress...

Interesting to know that my sensor is starting with `f0 6d 0a`, in #1077 it is `f5 78 0a`. So first 4 bits seem to be filler. Then 12 bits that vary and might be channel, id, battery or something and then a `0a` that could be static or include the battery as well.

mboehm21, 2020-01-10:
Is that too easy or is the humidity just a percentage when inverting the bits and converting to decimal? The first bit would always be zero in that case.

[temp /hum bitbench](http://triq.net/bitbench?c=%7B65%7D%20f0%206d%200a%20fe%20eb%209f%20d7%205c%20%20%20%23-18C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%208f%20da%207d%20%20%20%2325.1C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%206f%20da%200b%20%20%20%2325.2C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%201f%20d7%207c%20%20%20%2325.3C&c=%7B65%7D%20f0%206d%200a%20fe%20d1%200f%20da%205e%20%20%20%2325.4C&c=%7B65%7D%20f0%206d%200a%20fe%20d0%20ef%20d8%200c%20%20%20%2325.4C&c=%7B65%7D%20f0%206d%200a%20fe%20d0%20bf%20d8%20eb%20%20%20%2325.4C&c=%7B65%7D%20f0%206d%200a%20fe%20dd%200f%209e%20dc%20%20%20%236.5C&c=%7B65%7D%20f0%206d%200a%20fe%20dd%201f%209e%20b2%20%20%20%236.5C&f=hh%20hh%20hh%20hh%20Tem%3A%20~12d%20h%20Hum%3A%20~8d%20CRC%3Ahhl)

mboehm21, 2020-01-11:
Three samples indoor with re-inserting the batteries after each try under equal conditions:

- 12

`[07] {65} f5 8e 23 fe d3 1f d5 c1 80`
`[08] {65} f5 8e 23 fd ff ff ff 74 80`

- 13

`[07] {65} f4 45 c4 fe d2 ff d3 d6 80`
`[08] {65} f4 45 c4 fd ff ff ff f5 80`

- 14

`[07] {65} f6 db 5b fe d2 cf d3 5d 80`
`[08] {65} f6 db 5b fd ff ff ff cc 80`

mboehm21, 2020-01-11:
[bitbench that makes me almost certain with the battery](http://triq.net/bitbench?c=f0%206d%200a%20fe%20%5Bnew%20battery%5D&c=f0%206d%200a%20fd%20%5Bnew%20battery%5D&c=f5%208e%2023%20fe%20%5Bnew%20battery%5D&c=f5%208e%2023%20fd%20%5Bnew%20battery%5D&c=f4%2045%20c4%20fe%20%5Bnew%20battery%5D&c=f4%2045%20c4%20fd%20%5Bnew%20battery%5D&c=f6%20db%205b%20fe%20%5Bnew%20battery%5D&c=f6%20db%205b%20fd%20%5Bnew%20battery%5D&c=f5%20d7%2080%207e%20%5Bold%20battery%5D&c=f5%20d7%2080%207d%20%5Bold%20battery%5D&c=f2%2026%2074%20fe%20%5Bnew%20battery%5D&c=f2%2026%2074%20fd%20%5Bnew%20battery%5D&c=f1%209d%207e%207e%20%5Bold%20battery%5D&c=f1%209d%207e%207d%20%5Bold%20battery%5D&c=f5%202c%20f0%20fe%20%5Bnew%20battery%5D&c=f5%202c%20f0%20fd%20%5Bnew%20battery%5D&f=FILLER%3A%20~h%20VARIABLE%20ID%20%2F%20RID%20%2F%20CHANNEL%3A%20~h~h~h~h~h%20BATTERY%3A%20~h%20TEMP%20%2F%20WIND%3A%20~hl)

zuckschwerdt, 2020-01-11:
Does the decoded humidity match what you expect (40 seems rather dry, I don't usually observe below 45)?
But, yes, the humidity is almost always a plain and simple 8 (or 7) bit number. It's usually the first thing you can find and decode :)

mboehm21, 2020-01-11:
The best I can come up with for the moment:

[temp / hum](http://triq.net/bitbench?c=f1%209d%207e%207e%20d2%203f%20d1%20aa%2080&c=f4%2045%20c4%20fe%20d2%20ff%20d3%20d6%2080&c=f6%20db%205b%20fe%20d2%20cf%20d3%205d%2080&c=f0%206d%200a%20fe%20dd%201f%209e%20b2%2080&c=f0%206d%200a%20fe%20d0%20ef%20d8%200c%2080&c=f0%206d%200a%20fe%20d1%208f%20da%207d%2080&c=f0%206d%200a%20fe%20d0%20bf%20d8%20eb%2080&f=FILLER%3A%20~h%20VARIABLE%20ID%20%2F%20RID%20%2F%20CHANNEL%3A%20~h~h~h~h~h%20BATTERY%3A%20~h%20TEMP%20%2F%20WIND%3A%20~h%20TEMP%3A%20~12d%20FILLER%3A%20~h%20HUM%3A%20~d%20CRC%3A%20hh%20FILLER%3A%20hh)

@zuckschwerdt Unfortunately I don't have another sensor to confirm the humidity-values. Though, I know that it is way too dry in the livingroom. There is tile stove heating up. Kachelofen :). Humidity showed 97 while raining. Can that be a thing?

mboehm21, 2020-01-11:
And that would be my approach for the wind-part for the moment:

[wind](http://triq.net/bitbench?c=f0%206d%200a%20fd%20ff%20ff%20ff%2027%2080&c=f0%206d%200a%20fd%20fe%208f%20ff%205a%2080&c=f0%206d%200a%20fd%20f7%20ef%20ff%201b%2080&c=f5%208e%2023%20fd%20ff%20ff%20ff%2074%2080&f=FILLER%3A%20~h%20VARIABLE%20ID%20%2F%20RID%20%2F%20CHANNEL%3A%20~h~h~h~h~h%20BATTERY%3A%20~h%20TEMP%20%2F%20WIND%3A%20~h%20FILLER%3A%20h%20WIND%3A%20~h~h%20%20FILLER%3A%20hhh%20CRC%3A%20hh%20FILLER%3A%20hh)

Any other ideas / errors I could have made?

mboehm21, 2020-01-15:
Hi, does someone have some time to look into my results and give some hints regarding the encoding?

It would be awesome if this sensor would be supported. Thanks a lot.

mboehm21, 2020-01-16:
Datasheet with some specifications in English on page 13:

https://clientmedia.trade-server.net/1768_tfadost/media/2/60/16260.pdf

Nothing about the encoding, of course.

mboehm21, 2020-01-16:
My approach to decode the temperature:

[bitbench](http://triq.net/bitbench?c=f1%209d%207e%207e%20d2%203f%20d1%20aa%2080&c=f4%2045%20c4%20fe%20d2%20ff%20d3%20d6%2080&c=f6%20db%205b%20fe%20d2%20cf%20d3%205d%2080&c=f0%206d%200a%20fe%20dd%201f%209e%20b2%2080&c=f0%206d%200a%20fe%20d0%20ef%20d8%200c%2080&c=f0%206d%200a%20fe%20d1%208f%20da%207d%2080&c=f0%206d%200a%20fe%20d0%20bf%20d8%20eb%2080&f=FILLER%3A%20~h%20VARIABLE%20ID%20%2F%20RID%20%2F%20CHANNEL%3A%20~h~h~h~h~h%20BATTERY%3A%20~h%20TEMP%20%2F%20WIND%3A%20~h%20TEMP%3A%20~12d%20FILLER%3A%20~h%20HUM%3A%20~d%20CRC%3A%20hh%20FILLER%3A%20hh)

First 4 bit of the field might indicate if the temperature is positive or negative:

- 0010 -> -
- 0001 -> +

Next byte could be the temperature value if divided by 10. Does that make sense?

merbanan, 2020-01-16:
While the datasheet doesn't document the encoding it actually specifies the range. For example the temperature is encoded to at least cover -40.0 - 60.0 degrees Celsius. The total range covers 1000 positions. To encode that you need 11 bits minimum. So your guess doesn't have enough bits. But I don't think you are far off. Collect more readings and we should be able to figure out the encoding.

mboehm21, 2020-01-17:
Nice catch! Sensor is currently lying on the stove heating to give us some high temperature reading.

mboehm21, 2020-01-17:
Now I heated up the sensor to at least 43 °C and rtl_433 recognizes the signal somehow messed up now:
```

time      : 2020-01-17 19:52:02
model     : name         count     : 25            num_rows  : 25            rows      : 
len       : 1            data      : 8, 
len       : 38           data      : f1b067fec4, 
len       : 12           data      : efb, 
len       : 5            data      : 98, 
len       : 5            data      : f8, 
len       : 6            data      : b0, 
len       : 11           data      : 906, 
len       : 14           data      : df48, 
len       : 1            data      : 8, 
len       : 5            data      : d8, 
len       : 8            data      : d9, 
len       : 2            data      : c, 
len       : 4            data      : f, 
len       : 10           data      : f08, 
len       : 1            data      : 0, 
len       : 5            data      : 10, 
len       : 14           data      : fba0, 
len       : 4            data      : 3, 
len       : 4            data      : f, 
len       : 15           data      : c9be, 
len       : 1            data      : 8, 
len       : 5            data      : c0, 
len       : 57           data      : b067fec4fff26f8, 
len       : 65           data      : f1b067fec4fff26f8, 
len       : 65           data      : f1b067fec4fff26f8
codes     : {1}8, {38}f1b067fec4, {12}efb, {5}98, {5}f8, {6}b0, {11}906, {14}df48, {1}8, {5}d8, {8}d9, {2}c, {4}f, {10}f08, {1}0, {5}10, {14}fba0, {4}3, {4}f, {15}c9be, {1}8, {5}c0, {57}b067fec4fff26f8, {65}f1b067fec4fff26f8, {65}f1b067fec4fff26f8
time      : 2020-01-17 19:52:02
model     : name         count     : 5             num_rows  : 5             rows      : 
len       : 65           data      : f1b067fdfffffffa8, 
len       : 65           data      : f1b067fdfffffffa8, 
len       : 65           data      : f1b067fdfffffffa0, 
len       : 65           data      : f1b067fdfffffffa8, 
len       : 65           data      : f1b067fdfffffffa8
codes     : {65}f1b067fdfffffffa8, {65}f1b067fdfffffffa8, {65}f1b067fdfffffffa0, {65}f1b067fdfffffffa8, {65}f1b067fdfffffffa8
^CSignal caught, exiting!
```
Could it be that the harsh heat influences the sensor's internal clock / electronics?

I'll let the sensor cool down a bit and check again.

mboehm21, 2020-01-17:
After cooldown:

`_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-01-17 20:10:56
model     : name         count     : 8             num_rows  : 8             rows      : 
len       : 1            data      : 8, 
len       : 65           data      : f1b067fece2fe9dc8, 
len       : 65           data      : f1b067fece2fe9dc0, 
len       : 65           data      : f1b067fece2fe9dc8, 
len       : 65           data      : f1b067fece2fe9dc8, 
len       : 65           data      : f1b067fece2fe9dc8, 
len       : 65           data      : f1b067fece2fe9dc8, 
len       : 65           data      : f1b067fece2fe9dc8
codes     : {1}8, {65}f1b067fece2fe9dc8, {65}f1b067fece2fe9dc0, {65}f1b067fece2fe9dc8, {65}f1b067fece2fe9dc8, {65}f1b067fece2fe9dc8, {65}f1b067fece2fe9dc8, {65}f1b067fece2fe9dc8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-01-17 20:10:56
model     : name         count     : 5             num_rows  : 5             rows      : 
len       : 65           data      : f1b067fdfffffffa8, 
len       : 65           data      : f1b067fdfffffffa8, 
len       : 65           data      : f1b067fdfffffffa0, 
len       : 65           data      : f1b067fdfffffffa8, 
len       : 65           data      : f1b067fdfffffffa8
codes     : {65}f1b067fdfffffffa8, {65}f1b067fdfffffffa8, {65}f1b067fdfffffffa0, {65}f1b067fdfffffffa8, {65}f1b067fdfffffffa8`

It should still be above 35 °C, I guess.

mboehm21, 2020-01-17:
Some more cooldown:

`time      : 2020-01-17 20:23:01
model     : name         count     : 8             num_rows  : 8             rows      : 
len       : 1            data      : 8, 
len       : 65           data      : f1b067fed0bfe48e8, 
len       : 65           data      : f1b067fed0bfe48e0, 
len       : 65           data      : f1b067fed0bfe48e8, 
len       : 65           data      : f1b067fed0bfe48e8, 
len       : 65           data      : f1b067fed0bfe48e8, 
len       : 65           data      : f1b067fed0bfe48e8, 
len       : 65           data      : f1b067fed0bfe48e8
codes     : {1}8, {65}f1b067fed0bfe48e8, {65}f1b067fed0bfe48e0, {65}f1b067fed0bfe48e8, {65}f1b067fed0bfe48e8, {65}f1b067fed0bfe48e8, {65}f1b067fed0bfe48e8, {65}f1b067fed0bfe48e8

time      : 2020-01-17 20:23:01
model     : name         count     : 5             num_rows  : 5             rows      : 
len       : 65           data      : f1b067fdfffffffa8, 
len       : 65           data      : f1b067fdfffffffa8, 
len       : 65           data      : f1b067fdfffffffa0, 
len       : 65           data      : f1b067fdfffffffa8, 
len       : 65           data      : f1b067fdfffffffa8
codes     : {65}f1b067fdfffffffa8, {65}f1b067fdfffffffa8, {65}f1b067fdfffffffa0, {65}f1b067fdfffffffa8, {65}f1b067fdfffffffa8`

Somehow the signal looks a bit different...

merbanan, 2020-02-13:
Do you have an official receiver for this that you can correlate with the readings? If not I'm not able to help any more. I don't see any patterns there are to many unknowns.

merbanan, 2020-02-13:
Closing issue. Reopen if you can get readings from an official receiver.

closed 2020-02-13 by merbanan

mboehm21, 2020-03-05:
Sorry, I don't have the official receiver. I will start a new series of tests when I have more time. Really want to get this going...