merbanan/rtl_433 #1256: Lacrosse WS6010 and SMA SUNNY BEAM

rserroyen, 2020-01-08:
Hello
I have a "lacrosse solar temperature station WS6010
868 MHz [http://www.lacrossetechnology.fr/donnees/documents/produits/WS6010/PDF-fr-WS6010.pdf].

I also have an SMA and SUNNY BEAM solar inverter with Radiotransmission with 868 MHz inverters according to ETSI EN 300 220 [https://www.viessmann.com/web/france/fr_tdis.nsf/4563588e6a92515ac1256b1200303f6e/61a43b2eb7de4579c1275 SUNNY_BEAM.pdf](url)

With SDRSharp I try in the 868 Hhz game I get a strong 868.250 Mhz signal.

With the command rtl_433 -f 868250000 -a -A -W Test868.cu8
I get this but I cannot decode the frames.

With the frames starting
[00] {125} 21 55 55 55 55 55 55 54 5b a9 31 0a 44 b3 39 58
measurements
```

{00} 2 1 5 5 5 5 5 5 5 5 5 5 5 5 5 4 5 b a 9 3 1 0 9 0 2 a a e b 5 8
00001001000000101010101011101011
8 ° 46.4 101110.
85 101 010 1 humidity, in%
[00] {125} 2 1 5 5 5 5 5 5 5 5 5 5 5 5 5 4 5 b a 9 3 1 0 8 e c a b 8 7 5 8
00001000111011001010101110000111
7 ° 44.6 101 100.
85 101 010 1
[00] {125} 2 1 5 5 5 5 5 5 5 5 5 5 5 5 5 4 5 b a 9 3 1 0 a 2 8 b 9 0 3 5 0
00001010001010001011100100000011010
11 ° 51.8 1010010
52 1010010
92 10010010
[00] {125} 2 1 5 5 5 5 5 5 5 5 5 5 5 5 5 4 5 b a 9 3 1 0 a 3 2 b 9 8 3 5 8
                                                                                                                         0000000000110010101110011000001101011000
12 ° 53.6 1010011
                                                                                                                92 10010010

```
but nothing coincides

Thanks for your help

`Test mode active. Reading samples from file: Test868.cu8
Input format: uint8
Detected FSK package
Analyzing pulses...
Total count:   49,  width:  3254                (13.0 ms)
Pulse width distribution:
 [ 0] count:    1,  width:     0 [ 0; 0]        (   0 us)
 [ 1] count:   41,  width:    26 [25;27]        ( 104 us)
 [ 2] count:    5,  width:    50 [42;53]        ( 200 us)
 [ 3] count:    2,  width:    78 [78;78]        ( 312 us)
Gap width distribution:
 [ 0] count:    5,  width:    52 [52;53]        ( 208 us)
 [ 1] count:    3,  width:   104 [104;105]      ( 416 us)
 [ 2] count:   37,  width:    26 [25;27]        ( 104 us)
 [ 3] count:    3,  width:    78 [78;79]        ( 312 us)
Pulse period distribution:
 [ 0] count:   34,  width:    52 [52;53]        ( 208 us)
 [ 1] count:    7,  width:   115 [104;131]      ( 460 us)
 [ 2] count:    6,  width:    78 [78;79]        ( 312 us)
 [ 3] count:    1,  width:   156 [156;156]      ( 624 us)
Level estimates [high, low]:  15955,     22
Frequency offsets [F1, F2]:    4387,  -7520     (+16.7 kHz, -28.7 kHz)
Guessing modulation: Pulse Code Modulation (Not Return to Zero)
Attempting demodulation... short_limit: 26, long_limit: 26, reset_limit: 26624, demod_arg: 0
pulse_demod_pcm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {125} 21 55 55 55 55 55 55 54 5b a9 31 0a 30 b5 27 58

Detected FSK package
Analyzing pulses...
Total count:   49,  width:  3255                (13.0 ms)
[... 147 lines cut ...]
Pulse width distribution:
 [ 0] count:    1,  width:     0 [ 0; 0]        (   0 us)
 [ 1] count:   41,  width:    26 [26;27]        ( 104 us)
 [ 2] count:    5,  width:    50 [42;52]        ( 200 us)
 [ 3] count:    2,  width:    78 [78;79]        ( 312 us)
Gap width distribution:
 [ 0] count:    5,  width:    52 [52;54]        ( 208 us)
 [ 1] count:    3,  width:   104 [104;105]      ( 416 us)
 [ 2] count:   37,  width:    26 [26;27]        ( 104 us)
 [ 3] count:    3,  width:    78 [78;78]        ( 312 us)
Pulse period distribution:
 [ 0] count:   34,  width:    52 [52;54]        ( 208 us)
 [ 1] count:    7,  width:   115 [104;131]      ( 460 us)
 [ 2] count:    6,  width:    78 [78;79]        ( 312 us)
 [ 3] count:    1,  width:   157 [157;157]      ( 628 us)
Level estimates [high, low]:  15901,      7
Frequency offsets [F1, F2]:    3056,  -8473     (+11.7 kHz, -32.3 kHz)
Guessing modulation: Pulse Code Modulation (Not Return to Zero)
Attempting demodulation... short_limit: 26, long_limit: 26, reset_limit: 26624, demod_arg: 0
pulse_demod_pcm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {125} 21 55 55 55 55 55 55 54 5b a9 31 0a 30 b5 27 58

Test mode file issued 316 packets
`

zuckschwerdt, 2020-01-08:
Shift by one bit and you'll see a familar pattern: `aaaaaaaaaa 2dd4…`. This will be the correct alignment. Try [this BitBench](http://triq.net/bitbench?c=%7B125%7D21555555555555545ba9310a44b33958&f=8h%20%0A8b%20&a=Preamble&m=aaaa2dd4&s=1) to explore the data.

gdt, 2023-09-30:
@rserroyen Are you still working on this?   This issue has only a small clue and I'm not sure we want to leave it open after 3 years with no progress. It woudl be good to figure out how to capture the knowledge so far, though.

gdt, 2023-09-30:
Also, I can't tell what device this is about.  Can you update to the latest rtl_433 and re-run, and turn off one or the other, and figure out what's really going on?