merbanan/rtl_433 #1257: Help receiving and analyzing new simplisafe system signals

niemilkm, 2020-01-08:
I just bought a new SimpliSafe system and I am trying to read the signals to ultimately determine if the alarm is in AWAY, HOME, or OFF mode and then also which sensors are open/close as they report. I am starting with reading data from an entry sensor since that is on the 433 MHz frequency. I am very new to all of this. Steps so far.

1. Bought this [antenna](https://www.amazon.com/gp/product/B011HVUEME/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1).
2. Installed raspberry pi [image](https://www.youtube.com/watch?v=L0fSEbGEY-Q&t=4s) which comes with rtl_433 on it (see comments of video for link).
3. Running rtl_433 I have pulled in at least two devices (Schrader TPMS sensor and Oregon Scientific SL109H - not my devices but at least verified everything is working).
4. Open and Close door entry sensors and nothing shows up in my terminal. I then ran it with verbose on and saw signals coming in as I opened/closed door entry sensors. This output is shown below.

Pulse Data: 15 Pulses was from the entry sensor opening and then after those pulses was from the entry sensor closing.
```
pi@sdr:~ $ rtl_433 -R 102 -v -vv -vvv -vvvv -M newmodel
rtl_433 version 19.08-3-g04e818a branch master at 201909021329 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 136 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
Pulse data: 2 pulses
[  0] Pulse:   37, Gap:   12, Period:   49
[  1] Pulse:   63, Gap: 2501, Period: 2564
Pulse data: 1 pulses
[  0] Pulse:   44, Gap: 2501, Period: 2545
Pulse data: 1 pulses
[  0] Pulse:   43, Gap: 2501, Period: 2544
Pulse data: 1 pulses
[  0] Pulse:   42, Gap: 2501, Period: 2543
Pulse data: 1 pulses
[  0] Pulse: 18468, Gap: 25001, Period: 43469
Pulse data: 15 pulses
[  0] Pulse:  181, Gap:   11, Period:  192
[  1] Pulse:  304, Gap:   13, Period:  317
[  2] Pulse:  398, Gap:   14, Period:  412
[  3] Pulse:  301, Gap:   17, Period:  318
[  4] Pulse:  298, Gap:   13, Period:  311
[  5] Pulse: 1034, Gap:   11, Period: 1045
[  6] Pulse:  198, Gap:   11, Period:  209
[  7] Pulse:  812, Gap:   18, Period:  830
[  8] Pulse: 2076, Gap:   11, Period: 2087
[  9] Pulse: 3000, Gap:   12, Period: 3012
[ 10] Pulse:  507, Gap:   13, Period:  520
[ 11] Pulse:  507, Gap:   14, Period:  521
[ 12] Pulse: 2382, Gap:   13, Period: 2395
[ 13] Pulse:  925, Gap:   12, Period:  937
[ 14] Pulse: 5367, Gap: 25001, Period: 30368
Pulse data: 1 pulses
[  0] Pulse: 18468, Gap: 25001, Period: 43469
Pulse data: 1 pulses
[  0] Pulse: 18470, Gap: 25001, Period: 43471
Pulse data: 1 pulses
[  0] Pulse: 18403, Gap: 25001, Period: 43404
Pulse data: 1 pulses
[  0] Pulse: 18475, Gap: 25001, Period: 43476
```

If I do not use `-M newmodel`. Pulses 18506, 18583, and 18574 from opening same sensor used above and then the last 3 pulses were from closing the entry sensor used above.

```
pi@sdr:~ $ rtl_433 -R 102 -v -vv -vvv -vvvv            
rtl_433 version 19.08-3-g04e818a branch master at 201909021329 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 1 out of 136 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
Pulse data: 2 pulses
[  0] Pulse:   42, Gap:   55, Period:   97
[  1] Pulse:   40, Gap: 2501, Period: 2541
Pulse data: 1 pulses
[  0] Pulse:   43, Gap: 2501, Period: 2544
Pulse data: 1 pulses
[  0] Pulse:   43, Gap: 2501, Period: 2544
Pulse data: 1 pulses
[  0] Pulse: 18506, Gap: 25001, Period: 43507
Pulse data: 1 pulses
[  0] Pulse: 18583, Gap: 25001, Period: 43584
Pulse data: 1 pulses
[  0] Pulse: 18574, Gap: 25001, Period: 43575
Pulse data: 1 pulses
[  0] Pulse: 18622, Gap: 25001, Period: 43623
Pulse data: 1 pulses
[  0] Pulse: 18527, Gap: 25001, Period: 43528
Pulse data: 1 pulses
[  0] Pulse: 18621, Gap: 25001, Period: 43622
```

Any help with this is greatly appreciated. Great work on rtl_433!
Kevin

merbanan, 2020-01-08:
Run with rtl_433 -A remove the antenna, place a sensor nearby and then trigger the sensor. Then post the output.

niemilkm, 2020-01-08:
Steps:

1. Ran rtl_433 -A
2. Here is the output from that
```
pi@sdr:~ $ rtl_433 -A    
rtl_433 version 19.08-3-g04e818a branch master at 201909021329 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 106 out of 136 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 131-136 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
Detected OOK package	2020-01-08 19:38:44
Analyzing pulses...
Total count:    1,  width: 0.17 ms		(   43 S)
Pulse width distribution:
 [ 0] count:    1,  width:  172 us [172;172]	(  43 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:  12395,    262
RSSI: -1.2 dB SNR: 16.7 dB Noise: -17.9 dB
Frequency offsets [F1, F2]:   12369,      0	(+47.2 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

Detected OOK package	2020-01-08 19:38:44
Analyzing pulses...
Total count:    1,  width: 0.18 ms		(   44 S)
Pulse width distribution:
 [ 0] count:    1,  width:  176 us [176;176]	(  44 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:  11990,    190
RSSI: -1.4 dB SNR: 18.0 dB Noise: -19.3 dB
Frequency offsets [F1, F2]:   10454,      0	(+39.9 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

Detected OOK package	2020-01-08 19:38:44
Analyzing pulses...
Total count:    1,  width: 0.18 ms		(   44 S)
Pulse width distribution:
 [ 0] count:    1,  width:  176 us [176;176]	(  44 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:  11990,    191
RSSI: -1.4 dB SNR: 18.0 dB Noise: -19.3 dB
Frequency offsets [F1, F2]:   13002,      0	(+49.6 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...

Detected OOK package	2020-01-08 19:39:25
Analyzing pulses...
Total count:    1,  width: 0.07 ms		(   17 S)
Pulse width distribution:
 [ 0] count:    1,  width:   68 us [68;68]	(  17 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   3127,    387
RSSI: -7.2 dB SNR: 9.1 dB Noise: -16.3 dB
Frequency offsets [F1, F2]:     881,      0	(+3.4 kHz, +0.0 kHz)
[... 242 lines cut ...]
 [ 0] count:    1,  width:  232 us [232;232]	(  58 S)
 [ 1] count:   32,  width:  412 us [400;420]	( 103 S)
 [ 2] count:   19,  width:  620 us [608;628]	( 155 S)
 [ 3] count:    5,  width: 1332 us [1248;1460]	( 333 S)
 [ 4] count:   11,  width:  872 us [828;1044]	( 218 S)
 [ 5] count:    5,  width: 1916 us [1668;2088]	( 479 S)
 [ 6] count:    1,  width: 2500 us [2500;2500]	( 625 S)
 [ 7] count:    1,  width: 20664 us [20664;20664]	(5166 S)
Level estimates [high, low]:  15938,    121
RSSI: -0.1 dB SNR: 21.2 dB Noise: -21.3 dB
Frequency offsets [F1, F2]:   12734,  -5773	(+48.6 kHz, -22.0 kHz)
Guessing modulation: No clue...

Detected OOK package	2020-01-08 19:46:54
Analyzing pulses...
Total count:    1,  width: 0.04 ms		(   10 S)
Pulse width distribution:
 [ 0] count:    1,  width:   40 us [40;40]	(  10 S)
Gap width distribution:
Pulse period distribution:
Level estimates [high, low]:   1000,    123
RSSI: -12.1 dB SNR: 9.1 dB Noise: -21.2 dB
Frequency offsets [F1, F2]:    -572,      0	(-2.2 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
```

merbanan, 2020-01-08:
Ok, run rtl_433 -S unknown to save a few sample signals. And then post them here.

niemilkm, 2020-01-08:
g001 and g002 were immediate captures. g003-g006 were from after I opened and then closed the entry sensor.

[433_rtl.zip](https://github.com/merbanan/rtl_433/files/4036831/433_rtl.zip)

merbanan, 2020-01-08:
rtl_433 -X "n=doorbell,m=FSK_PCM,s=204,l=204,r=7000,g=1000,preamble=0x5526" might give the proper decoded bits. The preamble sync might need to be adjusted.

```
codes     : {257}16a3bc2c04013ea26bd01e0030cae222ffd842634d89ef7574da7286000000000
codes     : {257}16a3bc2c04013ea26bd01e0030cae222ffd842634d89ef7574da7286000000000
codes     : {256}16a3bc2c04013ea26bd21e0103b1a07f861673b5d1c531fa0bcd269c00000000
codes     : {256}16a3bc2c04013ea26bd21e0103b1a07f861673b5d1c531fa0bcd269c00000000
codes     : {256}16a3bc2c04013ea26bd21e0103b1a07f861673b5d1c531fa0bcd269c00000000
```
After that you need to figure out the meaning of the different bits by varying states and comparing different sensors with each other.

Worth not is also that the recorded signals are distorted. Reduce the gain some.

niemilkm, 2020-01-08:
I am pulling back data now. I will compare this data to itself after opening/closing multiple times and to other sensors and see if I can find a pattern where I can determine which sensor is which and the status (OPEN/CLOSE) of each sensor.

Is there a way to tell what gain it selected automatically so that I can reduce it manually as you mentioned (using the -g flag)?

Thank you very much for all your help - this is super helpful!

zuckschwerdt, 2020-01-09:
Be sure to closely look at the signal, if it's Simplisafe the modulation should be PIWM. The captures look more like PCM though.

zuckschwerdt, 2020-01-09:
> Is there a way to tell what gain it selected automatically so that I can reduce it manually as you mentioned (using the -g flag)?

No.

niemilkm, 2020-01-09:
Thank you all for the help. Here is an update. I am getting data and can tell which device is which. Figuring out the status (OPEN/CLOSED) is more difficult. For now, I am going to go under the impression that if it is currently closed and I get a signal for that sensor, then I can assume open - same vice versa. Once I get this data into my application, I am going to move on to the keypad (capture alarm status: home, away, or off) and then maybe come back to trying to decode the status of the entry sensors. Here is the data.

[image]

Here is the data from the screenshot above.
```
16a3bc2c04013ea26bd41e019a2ef5f05739b7be716dfe78713e8bda000000000
16a3bc2c04013ea26bde1e014d2860ef9cca9c2d6de3f671480d2922000000000
16a3bc2c04013ea26be21e00f36b1f5cb5dc42d878c000bdcdd07e9a000000000
16a3bc2c04013ea26be61e002cd4a5227c542a0282acd91c494dc0d6000000000
16a3bc2c04013ea26bd61e00afe48d1229cd4cd87b14ac2d49c0223400000000
16a3bc2c04013ea26be01e01d415cefbd5a08d28b77dbe91793de876000000000
16a3bc2c04013ea26be41e0123ce6a3e41d20bfc4c36099ca164059000000000
16a3bc2c04013ea26be81e01c9f496e42c2bafdb99f914c1ede0471a000000000
16a3bc2c0400d1abac645601933165c5a8ebb3f85faf8b4971ed556000000000
16a3bc2c0400d1abac6a56003027364a9f207580f87e7e228814671000000000
16a3bc2c0400d1abac6e56010e668fce7e5e0ba8a3d1c71a569aa5da000000000
16a3bc2c0400d1abac7256004e677a2cdaacd30d94fdcc312f1f0c8400000000
16a3bc2c0400d1abac6656017f5bb6466380920c2c36c62f1cf04ac00000000
16a3bc2c0400d1abac6c5601e96734b74760a2f8a0cd07f4e7785b2000000000
16a3bc2c0400d1abac7056007f7bcb0bcf8a5342829f5f191163f49a000000000
16a3bc2c0400d1abac74560043cd769789294ef66aac61c279091400000000
16a3bc2c04013ea2a0922001733c1d4faba50b9ae1f0c5b01fd04abe000000000
16a3bc2c04013ea2a0962001d941c18d0fa20ff704db27e54caef1fc00000000
16a3bc2c04013ea2a09a2000d0bd8db71c7cb34324d1b59efcbfe25a000000000
16a3bc2c04013ea2a09e20010c4fa918feb245938521be220886357400000000
16a3bc2c04013ea2a0942001e71b517b2549dc5495efe2d7620cc6ae000000000
16a3bc2c04013ea2a0982000e39586ddd8d75da262a30a405d2c6c26000000000
16a3bc2c04013ea2a09c2000c0f08f947df1af5afbfbfafde7287926000000000
16a3bc2c04013ea2a0a0200193050becf1b42d52c463e40c7a285dbe000000000
```

Thanks again!

zuckschwerdt, 2020-01-09:
Maybe [a BitBench](http://triq.net/bitbench?c=16a3bc2c04013ea26bd41e019a2ef5f05739b7be716dfe78713e8bda000000000%20%5BS1%20Open%5D&c=16a3bc2c04013ea26bde1e014d2860ef9cca9c2d6de3f671480d2922000000000%20%5BS1%20Open%5D&c=16a3bc2c04013ea26be21e00f36b1f5cb5dc42d878c000bdcdd07e9a000000000%20%5BS1%20Open%5D&c=16a3bc2c04013ea26be61e002cd4a5227c542a0282acd91c494dc0d6000000000%20%5BS1%20Open%5D&c=16a3bc2c04013ea26bd61e00afe48d1229cd4cd87b14ac2d49c0223400000000%20%5BS1%20Close%5D&c=16a3bc2c04013ea26be01e01d415cefbd5a08d28b77dbe91793de876000000000%20%5BS1%20Close%5D&c=16a3bc2c04013ea26be41e0123ce6a3e41d20bfc4c36099ca164059000000000%20%5BS1%20Close%5D&c=16a3bc2c04013ea26be81e01c9f496e42c2bafdb99f914c1ede0471a000000000%20%5BS1%20Close%5D&c=&c=16a3bc2c0400d1abac645601933165c5a8ebb3f85faf8b4971ed556000000000%20%5BS2%20Open%5D&c=16a3bc2c0400d1abac6a56003027364a9f207580f87e7e228814671000000000%20%5BS2%20Open%5D&c=16a3bc2c0400d1abac6e56010e668fce7e5e0ba8a3d1c71a569aa5da000000000%20%5BS2%20Open%5D&c=16a3bc2c0400d1abac7256004e677a2cdaacd30d94fdcc312f1f0c8400000000%20%5BS2%20Open%5D&c=16a3bc2c0400d1abac6656017f5bb6466380920c2c36c62f1cf04ac00000000%20%5BS2%20Close%5D&c=16a3bc2c0400d1abac6c5601e96734b74760a2f8a0cd07f4e7785b2000000000%20%5BS2%20Close%5D&c=16a3bc2c0400d1abac7056007f7bcb0bcf8a5342829f5f191163f49a000000000%20%5BS2%20Close%5D&c=16a3bc2c0400d1abac74560043cd769789294ef66aac61c279091400000000%20%5BS2%20Close%5D&c=&c=16a3bc2c04013ea2a0922001733c1d4faba50b9ae1f0c5b01fd04abe000000000%20%5BS3%20Open%5D&c=16a3bc2c04013ea2a0962001d941c18d0fa20ff704db27e54caef1fc00000000%20%5BS3%20Open%5D&c=16a3bc2c04013ea2a09a2000d0bd8db71c7cb34324d1b59efcbfe25a000000000%20%5BS3%20Open%5D&c=16a3bc2c04013ea2a09e20010c4fa918feb245938521be220886357400000000%20%5BS3%20Open%5D&c=16a3bc2c04013ea2a0942001e71b517b2549dc5495efe2d7620cc6ae000000000%20%5BS3%20Close%5D&c=16a3bc2c04013ea2a0982000e39586ddd8d75da262a30a405d2c6c26000000000%20%5BS3%20Close%5D&c=16a3bc2c04013ea2a09c2000c0f08f947df1af5afbfbfafde7287926000000000%20%5BS3%20Close%5D&c=16a3bc2c04013ea2a0a0200193050becf1b42d52c463e40c7a285dbe000000000%20%5BS3%20Close%5D&f=40h%2032h%20%7C%208h%208h%2016h%2032h%2032h%2032h%2032h%2032h) can help to structure the data.
There could be plain data and a MAC, or some of the data could be encrypted.

merbanan, 2020-01-09:
Nice find with the counter. I would say the id is the 4 bytes before the counter. But the random data after the second id guess is the real problem. This looks encrypted. I assume that the counter + id is used to encrypt the data.

A significant effort will be needed to figure out this. Some concrete things to test are:

1. loop the counter to see if it creates the same data
2. collide the loop counter from 2 sensors
3. provide more data from the system, maybe they mention something about encryption in the manual
4. open up the sensors and document the chips

Lots and lots of readings will be needed. Read up on the configuration files (/cfg/*) and start creating a decoding template.

merbanan, 2020-01-09:
Colliding the loop counters will tell if the encryption uses the id or only the counters.

mores, 2021-12-23:
@niemilkm - Which version of the sensor are you using ?
[image]

niemilkm, 2021-12-23:
> @niemilkm - Which version of the sensor are you using ? [image]

GEN 3

mores, 2021-12-23:
> 4\. open up the sensors and document the chips

Here is what the inside of a GEN 3 door sensor looks like:

[image]

giantorth, 2021-12-28:
I too am interested in support for the v3 sensors.  
A couple people have had some success that may help some with these efforts:
 - [RadioDoor](https://github.com/joelewiss/radiodoor ) uses rtl_433
 - [Teardown article](https://medium.com/tenable-techblog/inside-simplisafe-alarm-system-291a8c3e4d89) and a [PoC using a Yardstick one](https://github.com/tenable/poc/tree/master/SimpliSafe)

zuckschwerdt, 2021-12-28:
That teardown is a great read!
Additionally we did spot a counter, but there won't be any chance to decode the AES payload.

giantorth, 2021-12-28:
It seems the radiodoor project managed to get open/closed state without any decryption.  I've only tested my motion sensors with the code and they seem to oscillate between two different values every time they're triggered.

mores, 2021-12-28:
> Additionally we did spot a counter, but there won't be any chance to decode the AES payload.

It looks like someone has: 
https://github.com/tenable/poc/blob/983dcf94577b1a041f304c8e0537b670c0c18655/SimpliSafe/packet_decoder.py#L110

zuckschwerdt, 2021-12-28:
> It looks like someone has

But isn't that a per-device key?

zuckschwerdt, 2021-12-28:
> my motion sensors with the code and they seem to oscillate between two different values every time they're triggered.

That's not visible in the list of codes we have above. Can you capture codes and show an example?

giantorth, 2021-12-28:
Yeah that packet decoder came from the teardown where they wrote their own AES key to the sensor I believe.

Here is me cycling an entry sensor with that radiodoor script. I've dropped the first digit from the device ID because it seems to be garbled/random at times.  The sensors send 3 messages if all is working well.

```New data!
Length: 358
Device ID: 5555554985a8ef0b0100a1fa5f
Codes sent: 38b8039a358039e708a891dcb7d832375a2bf8000000000000000000000000
8b80

New data!
Length: 358
Device ID: 5555554985a8ef0b0100a1fa5f
Codes sent: 38b8039a358039e708a891dcb7d832375a2bf8000000000000000000000000
8b80

New data!
Length: 358
Device ID: 5555554985a8ef0b0100a1fa5f
Codes sent: 38b8039a358039e708a891dcb7d832375a2bf8000000000000000000000000
8b80

New data!
Length: 358
Device ID: 5555554985a8ef0b0100a1fa5f
Codes sent: 40b804889a97bbae35e94b8a94c5dadd42b918000000000000000000000000
0b80

New data!
Length: 358
Device ID: 5555554985a8ef0b0100a1fa5f
Codes sent: 40b804889a97bbae35e94b8a94c5dadd42b918000000000000000000000000
0b80

New data!
Length: 358
Device ID: 5555554985a8ef0b0100a1fa5f
Codes sent: 40b804889a97bbae35e94b8a94c5dadd42b918000000000000000000000000
0b80

New data!
Length: 358
Device ID: 5555554985a8ef0b0100a1fa5f
Codes sent: 48b80774191033992d11614fb45bf598e18eb8000000000000000000000000
8b80

New data!
Length: 358
Device ID: 5555554985a8ef0b0100a1fa5f
Codes sent: 48b80774191033992d11614fb45bf598e18eb8000000000000000000000000
8b80

New data!
Length: 358
Device ID: 5555554985a8ef0b0100a1fa5f
Codes sent: 48b80774191033992d11614fb45bf598e18eb8000000000000000000000000
8b80
```

mores, 2021-12-28:
I do not have a usb radio receiver yet ( my order is in process ).

From the comments in the code "# this one I got from my entry sensor firmware dump" - is it possible all the sensors use the same key ??

When you first install the sensor you push a button on the sensor....wondering if that button press sends the key so that it can be used to decode future signals. 

Anyway here is the output from that decoder script:

=== PACKET DISSECTION ===
Packet         : 1602007289cbfad2027d441c349698ab938ee973c0cd9f098d
Length         : 16 (22)
Serial         :     007289cb
Counter        :             fad202
CMAC:          :                   7d441c34
Encrypted Data :                           9698ab938ee973c0cd9f
Chksum         :                                               098d

Data Decryption:
  AES Call 1 Res : 931b85938e5073c0899fc2ba112f6b25
  Decrypted Data : 05832e0000b900004400

CMAC Verify:
  AES Call 2 Res : 34d21f79bc403b8ccbaa7fb1585be8e2
  LSFR Res1: 10e292d698e731bff897ec4458ba4fb2
  CHK Data1 : 9698ab938ee973c0cd9f
  CHK Data2 : 867a3945160e427f3508ec4458ba4fb2
  LSFR Res2: 6591a486ec54a4ee4b1c3d38c3b5cf37
  LSFR Res2_xor: 6591a486ec54a46e4b1c3d38c3b5cf67
  LSFR Res3: b2ee02cf2e485805329de21bbf2294e0
  AES Call 3 Res : cfaa1efb0c408095ed27dc40eeae81f0
  CHK Data3 (CALCULATED CMAC): 7d441c34
     CMAC Match!

Checksum Verify:
test: 0x98d
  Calculated Checksum:098d
     Checksum Verified!

giantorth, 2021-12-28:
Motion sensor dump.  Hardware seems to have 150s cooldown before it will detect again.  No clear value is ever sent.  This is two sequential triggers from a single sensor.
```
New data!
Length: 357
Device ID: 5555554985a8ef0b0100c302dc
Codes sent: c0c0051b9d9617ef1301df51ad706af0d92268000000000000000000000000
0c00

New data!
Length: 357
Device ID: 5555554985a8ef0b0100c302dc
Codes sent: c0c0051b9d9617ef1301df51ad706af0d92268000000000000000000000000
0c00

New data!
Length: 358
Device ID: 5555554985a8ef0b0100c302dc
Codes sent: c0c0051b9d9617ef1301df51ad706af0d92268000000000000000000000000
0c00

New data!
Length: 358
Device ID: 5555554985a8ef0b0100c302dc
Codes sent: c8c005413316a75301e400b1a3b2ea31aa2218000000000000000000000000
8c00

New data!
Length: 358
Device ID: 5555554985a8ef0b0100c302dc
Codes sent: c8c005413316a75301e400b1a3b2ea31aa2218000000000000000000000000
8c00

New data!
Length: 358
Device ID: 5555554985a8ef0b0100c302dc
Codes sent: c8c005413316a75301e400b1a3b2ea31aa2218000000000000000000000000
8c00
```

zuckschwerdt, 2021-12-28:
@mores drop in some of the codes from the list above and check if those decode also. (I don't have any Python2 readily available anymore)

zuckschwerdt, 2021-12-28:
@giantorth here is [your data as BitBench](https://triq.net/bitbench#c=5555554985a8ef0b0100a1fa5f%2038b8039a358039e708a891dcb7d832375a2bf8&c=5555554985a8ef0b0100a1fa5f%2038b8039a358039e708a891dcb7d832375a2bf8&c=5555554985a8ef0b0100a1fa5f%2038b8039a358039e708a891dcb7d832375a2bf8&c=5555554985a8ef0b0100a1fa5f%2040b804889a97bbae35e94b8a94c5dadd42b918&c=5555554985a8ef0b0100a1fa5f%2040b804889a97bbae35e94b8a94c5dadd42b918&c=5555554985a8ef0b0100a1fa5f%2040b804889a97bbae35e94b8a94c5dadd42b918&c=5555554985a8ef0b0100a1fa5f%2048b80774191033992d11614fb45bf598e18eb8&c=5555554985a8ef0b0100a1fa5f%2048b80774191033992d11614fb45bf598e18eb8&c=5555554985a8ef0b0100a1fa5f%2048b80774191033992d11614fb45bf598e18eb8&c=&c=5555554985a8ef0b0100c302dc%20c0c0051b9d9617ef1301df51ad706af0d92268&c=5555554985a8ef0b0100c302dc%20c0c0051b9d9617ef1301df51ad706af0d92268&c=5555554985a8ef0b0100c302dc%20c0c0051b9d9617ef1301df51ad706af0d92268&c=5555554985a8ef0b0100c302dc%20c8c005413316a75301e400b1a3b2ea31aa2218&c=5555554985a8ef0b0100c302dc%20c8c005413316a75301e400b1a3b2ea31aa2218&c=5555554985a8ef0b0100c302dc%20c8c005413316a75301e400b1a3b2ea31aa2218&f=LEN%3A8h%20TYP%3A8h%20ID%3A8h8h8h8h%208h8h%208h%202h%7C4h8h4h%7C6h%208h8h%208h8h%208h8h%208h8h%208h8h%208h8h%208h8h%208h8h%208h8h&a=Preamble&m=5526), note the raw codes (starting 5555) need to be aligned to `5526` and thus shift by two bits. The "toggle bytes" are not aligned -- play with the BitBench to find a good pattern. But it could well be that's it's just the high order bits of the counter we suspect. The visible toggle is suspicious though.

mores, 2021-12-28:
Results not very promising:

> 16a3bc2c04013ea26bd41e019a2ef5f05739b7be716dfe78713e8bda000000000

=== PACKET DISSECTION ===
Packet         : 16a3bc2c04013ea26bd41e019a2ef5f05739b7be716dfe78717833658bda
Length         : 16 (22)
Serial         :     bc2c0401
Counter        :             3ea26b
CMAC:          :                   d41e019a
Encrypted Data :                           2ef5f05739b7be716dfe
Chksum         :                                               8bda

No key found to decrypt or verify CMAC

Checksum Verify:
test: 0x13df
  Calculated Checksum:13df
     Bad Checksum!

> 16a3bc2c04013ea26bde1e014d2860ef9cca9c2d6de3f671480d2922000000000

=== PACKET DISSECTION ===
Packet         : 16a3bc2c04013ea26bde1e014d2860ef9cca9c2d6de3f671480d2922
Length         : 16 (22)
Serial         :     bc2c0401
Counter        :             3ea26b
CMAC:          :                   de1e014d
Encrypted Data :                           2860ef9cca9c2d6de3f6
Chksum         :                                               2922

No key found to decrypt or verify CMAC

Checksum Verify:
test: 0xb455
  Calculated Checksum:b455
     Bad Checksum!

zuckschwerdt, 2021-12-28:
> dropped the first digit from the device ID because it seems to be garbled/random at times.

To clear that up, the `5555` is a preamble. It ends in `5526`. This is used to wake-up train the timing and align the start of data. It does not need to be received perfectly.

zuckschwerdt, 2021-12-28:
@mores you might need to copy the key to use with that serial, i.e. `encr_keys = { '\x00\x72\x89\xcb': ...` needs to be `encr_keys = { '\xbc\x2c\x04\x01': ...`

zuckschwerdt, 2021-12-28:
Oh, I missed the "Counter, CMAC" hints there. I'll update the BitBench link.

mores, 2021-12-28:
@zuckschwerdt - I figured if the cksum did not match - why bother with the key....anyway here is one with the key set:

=== PACKET DISSECTION ===
Packet         : 16a3bc2c04013ea26bd41e019a2ef5f05739b7be716dfe78713e8bda
Length         : 16 (22)
Serial         :     bc2c0401
Counter        :             3ea26b
CMAC:          :                   d41e019a
Encrypted Data :                           2ef5f05739b7be716dfe
Chksum         :                                               8bda

Data Decryption:
  AES Call 1 Res : ef32a8b60e152c1578a83e94ab4539b5
  Decrypted Data : c1c758e137a292641556

CMAC Verify:
  AES Call 2 Res : 34d21f79bc403b8ccbaa7fb1585be8e2
  LSFR Res1: 098578fb0122381ec9500f52ead3e6cb
  CHK Data1 : 2ef5f05739b7be716dfe
  CHK Data2 : 277088ac3895866fa4ae0f52ead3e6cb
  LSFR Res2: 53958d8c1735c63569ebe2437ba785db
  LSFR Res2_xor: 53958d8c1735c6b569ebe2437ba7858b
  LSFR Res3: 75788f6753b2edad679b2511a807bbee
  AES Call 3 Res : 4dd2622af4289d404894a494f32a6057
  CHK Data3 (CALCULATED CMAC): 38aaed4d
     CMAC Mismatch!

Checksum Verify:
test: 0x13df
  Calculated Checksum:13df
     Bad Checksum!

zuckschwerdt, 2021-12-28:
Worth a try;)

zuckschwerdt, 2021-12-28:
Here is [the updated BitBench](https://triq.net/bitbench#c=5555554985a8ef0b0100a1fa5f%2038b8039a358039e708a891dcb7d832375a2bf8&c=5555554985a8ef0b0100a1fa5f%2038b8039a358039e708a891dcb7d832375a2bf8&c=5555554985a8ef0b0100a1fa5f%2038b8039a358039e708a891dcb7d832375a2bf8&c=5555554985a8ef0b0100a1fa5f%2040b804889a97bbae35e94b8a94c5dadd42b918&c=5555554985a8ef0b0100a1fa5f%2040b804889a97bbae35e94b8a94c5dadd42b918&c=5555554985a8ef0b0100a1fa5f%2040b804889a97bbae35e94b8a94c5dadd42b918&c=5555554985a8ef0b0100a1fa5f%2048b80774191033992d11614fb45bf598e18eb8&c=5555554985a8ef0b0100a1fa5f%2048b80774191033992d11614fb45bf598e18eb8&c=5555554985a8ef0b0100a1fa5f%2048b80774191033992d11614fb45bf598e18eb8&c=&c=5555554985a8ef0b0100c302dc%20c0c0051b9d9617ef1301df51ad706af0d92268&c=5555554985a8ef0b0100c302dc%20c0c0051b9d9617ef1301df51ad706af0d92268&c=5555554985a8ef0b0100c302dc%20c0c0051b9d9617ef1301df51ad706af0d92268&c=5555554985a8ef0b0100c302dc%20c8c005413316a75301e400b1a3b2ea31aa2218&c=5555554985a8ef0b0100c302dc%20c8c005413316a75301e400b1a3b2ea31aa2218&c=5555554985a8ef0b0100c302dc%20c8c005413316a75301e400b1a3b2ea31aa2218&f=LEN%3A8h%20TYP%3A8h%20ID%3A32h%20CTR%3A24h%20CMAC%3A32h%20ENCR%3A104h%20CHK%3A16h%20&a=Preamble&m=5526), with the fields.
If the CMAC is over the pain text (the rabbit hole of mac-then-encrypt vs encrypt-then-mac) it would allows us a peek into the plain text. But the counter should disturb things enough so it won't work. Seemingly not in this case. The CMAC relates to the plain text without much interference from the counter. That's where the visible "toggle" comes from: the CMAC is sensitive to a few bits (open/close...) in the plain text.

zuckschwerdt, 2021-12-28:
It would be hilarious if the encrypted portion is completly secure with AES and strong keys, just for the CMAC then to leak the data we're interested in... oO.

mores, 2021-12-28:
It looks like niemilkm might have been capturing packets with:

`rtl_433 -X "n=doorbell,m=FSK_PCM,s=204,l=204,r=7000,g=1000,preamble=0x5526"`

radiodoor is capturing with:

`rtl_443 -g 35 -f 433910000 -p 0 -X 'n=name,m=FSK_PCM,s=208,l=208,r=212992'`

Which one of these ( or perhaps something else ) should be used going forward ?

giantorth, 2021-12-28:
These parameters from the poc for rfcat were pulled from hardware 
```
d.setFreq(433899963)
d.setMdmModulation(MOD_2FSK)
d.setMdmSyncWord(0x930b)
d.setMdmSyncMode(SYNCM_16_of_16)
d.setMdmSyncMode(SYNCM_CARRIER_16_of_16)
d.setMdmDRate(4800)
d.makePktFLEN(0xff)
```

zuckschwerdt, 2021-12-28:
@mores gap (g) isn't used and the reset (r) can't be that long. The preamble can be adjusted later (e.g. in BitBench). But the one with preamble=0x5526 is preferred.

mores, 2021-12-28:
@zuckschwerdt - when the preamble is applied it seems to remove key information (8b80):

[image]

Where/how can this 8b80 be found in the data after applying the preamble ?

zuckschwerdt, 2021-12-28:
That 8b80 is the "middle" part of the CMAC in the BitBench I posted. Bytes 10-13.

giantorth, 2021-12-28:
One comment I have is I don't think any two devices will have the same CMAC values, they're only repeatable within a single device.  If you look at the two sensors coded in the radiodoor example he's got a different open/close value for each one.

giantorth, 2021-12-28:
I do think we have enough information to build a useful sensor here even without decrypting, with the ability to uniquely identify a device along with two different states for a given sensor.  This would be enough to send the values into MQTT and do useful things with the events after some value discovery.

zuckschwerdt, 2021-12-29:
Upon closer inspection: the assumed sync of 5526 produced a leading 0x16 by accident. That's still part of the sync. The full sync is `930b51de`, that shifts everything 23 bit left. And the "toggle" we saw is actually the counter. And of course that has a regular pattern.

Capture raw with: `rtl_433 -X "n=SimpliSafe=FSK_PCM,s=208,l=208,r=7000,preamble=930b51de"`
But I'll commit a decoder now too.

zuckschwerdt, 2021-12-29:
Decoder added now. Play around with all your sensors and watch the counter output, it looks like there could be a pattern based on open/close.

giantorth, 2021-12-29:
Okay got it built and testing, but the CMAC alone doesn't seem as repeatable as before somehow.
Here is the same sensor with open/close open/close events:
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:19
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 501700        CMAC      : d308ff4e      Encrypted : 6aece46f073ad73e8336
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:22
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 501700        CMAC      : d308ff4e      Encrypted : 6aece46f073ad73e8336
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:24
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 501700        CMAC      : d308ff4e      Encrypted : 6aece46f073ad73e8336
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:27
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 511700        CMAC      : 75fc0613      Encrypted : 170f39ff35eccb52573a
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:29
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 511700        CMAC      : 75fc0613      Encrypted : 170f39ff35eccb52573a
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:31
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 511700        CMAC      : 75fc0613      Encrypted : 170f39ff35eccb52573a
Integrity : CRC

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:35
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 521700        CMAC      : 1784dad7      Encrypted : d53ed9185e2db03cfe7c
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:36
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 521700        CMAC      : 1784dad7      Encrypted : d53ed9185e2db03cfe7c
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:38
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 521700        CMAC      : 1784dad7      Encrypted : d53ed9185e2db03cfe7c
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:41
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 531700        CMAC      : 636f5cf8      Encrypted : 2cf87ef4684a41323fe3
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:43
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 531700        CMAC      : 636f5cf8      Encrypted : 2cf87ef4684a41323fe3
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 07:37:45
model     : SimpliSafe-Gen3                        ID        : 0143f4bf
Type      : 02           Counter   : 531700        CMAC      : 636f5cf8      Encrypted : 2cf87ef4684a41323fe3
Integrity : CRC
```

zuckschwerdt, 2021-12-29:
> but the CMAC alone doesn't seem as repeatable as before

that field was not the CMAC but the counter.

zuckschwerdt, 2021-12-29:
Looks like the counter bytes are in the wrong order, `531700` is maybe really `001753`. And if you only look at the bottom bit **and the sensors stay in sync** then even/odd information maps to open/close. At least that's what "RadioDoor" does.

giantorth, 2021-12-29:
Ahh that makes sense now, let me collect some more data from a few sensors to see if that holds true.  I think ultimately that counter will need to be published as the sensor state if it's got the needed data.

Thanks for the decoder @zuckschwerdt!

giantorth, 2021-12-29:
Okay with a little testing, the byte order is really `001753` and the last byte does alternate between even/odd for open close states reliably within a single sensor.  

In testing with multiple entry sensors however the even/odd pattern can be flipped.  I suspect this might be the state of the reed switch when the device was first activated.  Inverting the value on a sensor-by-sensor basis would have to be at the message consumer side.

For motion sensors the counter doesn't matter as they do not appear to send a 'clear' state. Any sensor cooldown would again need to be handled in downstream systems receiving the trigger events.

giantorth, 2021-12-29:
With a little more testing, just adding a `state = (ctr %2);` the values for a single sensor seem stable over time. 
The byte ordering needs reversed, and I seem to be getting some negative device ids come out.  All said this is already quite usable for a binary sensor.

giantorth, 2021-12-29:
Even better, the type changes with the pairing button and does not disturb the counter for the normal open/close events:
```
time      : 2021-12-29 12:32:07
model     : SimpliSafe-Gen3                        ID        : 905801985
Type      : 02           Counter   : 001509        CMAC      : 775df992      Encrypted : cc0c4516cd70a4a1b35e                    Integrity : CRC
State     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 12:32:14
model     : SimpliSafe-Gen3                        ID        : 905801985
Type      : f2           Counter   : dbc105        CMAC      : 3582cfee      Encrypted : 263e2576c85229f726ba                    Integrity : CRC
State     : 1
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-12-29 12:32:29
model     : SimpliSafe-Gen3                        ID        : 905801985
Type      : 02           Counter   : 00150a        CMAC      : d1e35632      Encrypted : dc6a2bfd1a49736a7091                    Integrity : CRC
State     : 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```

zuckschwerdt, 2021-12-29:
The byte order is now corrected. Also messages for Arm/Disarm (len=21 or 24, normally 22) should perhaps work. If you test this and it does not work try removing the crc check and report what you find.
Negative ID values is a int/unsigned issue. Perhaps ID should be a string not int, not sure.

Type f2 is interesting, what was that?

giantorth, 2021-12-29:
The original simplisafe support does use a whole function to stuff the ID into a string while reversing it.

The type f2 was pressing the pairing/binding button on an entry sensor, in between an open and close event that shows as type 02.

giantorth, 2021-12-30:
Some tidbits for anybody trying to turn this code into a usable sensor with Home Assistant.

Running RTL_433 with events published to /model/id topics.  Autodiscovery will not work with a sensor this primitive. 

```rtl_433 -F "mqtt://hostname.com:1883,user=XXX,pass=XXX,retain=1,events=rtl_433[/model][/id]"```

Template sensor in Home Assistant:

```
template:
  - trigger:
      - platform: mqtt
        topic: "rtl_433/SimpliSafe-Gen3/00000000"
    binary_sensor:
      - name: Dining Room Motion
        state: "true"
        device_class: motion
        auto_off: 150
```

giantorth, 2021-12-30:
Adding info for entry sensor config, using the binary_sensor MQTT platform.  Invert the payload_on/off values as needed.  

Sensor will show random state if paring button is used but will self-correct on next event.  

One word of caution, if you cycle the sensor rapidly enough it is possible to get the counter out of sync.
```
binary_sensor:
  - platform: mqtt
    name: "Deck Sliding Door"
    state_topic: "rtl_433/SimpliSafe-Gen3/0000000"
    device_class: "door"
    value_template: >
      {{ (value_json.ctr % 2) }}
    payload_off: "1"
    payload_on: "0"
```

giantorth, 2021-12-31:
So, there is a wrinkle in these sensors that makes watching the counter potentially unreliable.  These sensors appear to re-transmit state every 6-8hrs or so.  For motion sensors this means periodic false triggers and for entry sensors it changes the even/odd state with a re-transmit.

loca5790, 2022-04-19:
I just received newer sensors and I do not see anything on the 433 MHz band when analyzing that frequency.  Not 908 or 345 either.  Any way this changed?

DaveCo1701, 2022-05-29:
> I just received newer sensors and I do not see anything on the 433 MHz band when analyzing that frequency. Not 908 or 345 either. Any way this changed?

I recently upgraded from a v2 system to a v3 system and am able to see the sensors transmitting on the default 433MHz frequency.

DaveCo1701, 2022-05-29:
Has there been any movement on this? I found this project ([LINK](https://github.com/joelewiss/radiodoor)) which claims to read v3 entry sensor states using rtl_433, but was unable to reproduce their results. I’m no coder, just a tinkerer, maybe I’m missing something?

zuckschwerdt, 2022-05-30:
@DaveCo1701 the RadioDoor project is wrong. They are just looking at random bits (the encryption counter).

giantorth, 2022-06-02:
The encryption counter is the only semi-useful data you get from these devices, but they randomly increment the counter without an event as well.  In my testing these sensors are appallingly slow compared to any other sensor I have.  For home security a couple seconds doesn't matter but for motion-activated light triggers it's not really usable.  

My zigbee and z-wave motion sensors are way more functional with nearly instant events.

[2 more replies]

closed 2023-02-09 by merbanan