merbanan/rtl_433 #1260: No longer working for Holman-WS5029 (134)

mic159, 2020-01-10:
Hi there,

I was using the latest release 19.08 and it was working, picking up the weather data reliably.
I just switched to the latest commit on master, and it no longer works.

After some binary-searching of commits, I found this is the one that "broke" it for me:
https://github.com/merbanan/rtl_433/commit/9a3de5fcf8c17d6919b23ad59ff9e1dc24185632

My command line:
`./rtl_433 -f 917M -R 134`

I would have no idea that I would need to tweak the sample rate if it didnt work with the default sample rate before.

Why was the sample rate changed for 917MHz?

mic159, 2020-01-10:
My workaround is to `git revert 442dc4bf58f16e8dbcfa48b1a4b6c76a8329956c`, then add `-s 250000` on the CLI.

It then works on the current master, phew!

merbanan, 2020-01-10:
@mic159 this device seems to use FSK, a new FSK demodulator was introduced to handle more sensors. This demodulator performed poorly on some signals (generic demodulation/decode is hard). Around the same time the default sample rate for >800MHz was changed to 1MS/s.

If I can get a sample I might be able to tune the new FSK demodulator so it can handle this sensor. (-S known). Post the signal in this issue.

To restore previous behavior use -s 250000 -Y classic with the current code. The plan is to enable both demodulators at the same time but has not been done due to time constraints.

merbanan, 2020-01-12:
Closing issue. This should be fixed in the future, right now it is possible to select the correct parameters to get back in working order.

closed 2020-01-12 by merbanan