merbanan/rtl_433 #1262: Dealing with frequency drift

jacopo-j, 2020-01-11:
Hello. I am using rtl_433 for monitoring OOK signals from a few dozen old sensors. Component aging has caused a significant frequency drift on some of them (up to about ±7 kHz from the center frequency) and this prevents the me from correctly decoding their signals (especially those further away from the antenna).

I have looked into frequency hopping as a possible solution, but it has turned out quite impractical. The drift varies from sensor to sensor among a continuous range, and hopping between some "centroid" frequencies does not yield very good results. Moreover, rtl_433 only supports monitoring each frequency for at least 1 second (as far as I can tell), which is a quite long period and losing messages becomes very common.

Is there any particular reason why hopping cannot be made faster? If not, I may get by forking the repo and making the changes I need.

Anyway, any suggestion is very welcome - you will probably come up with much better ideas for dealing with this problem.

zuckschwerdt, 2020-01-12:
Hoping is not what you are looking for. Make a table of all the sensor frequencies you see, perhaps use Gqrx or CubicSDR for this. Then calculate a window that safely covers all frequencies (leave at least 10k room at both ends). Pick the center frequency, cover the window with at least the same sample rate. E.g. sensors from 433.89M to 434.05 : Center 433.97M, window 160kHz, fits with default 250k sample rate.

closed 2020-01-12 by zuckschwerdt

merbanan, 2020-01-12:
Increasing the sample rate might work also. -s 1M.