merbanan/rtl_433 #1264: Acurite 896 not working

faspina, 2020-01-14:
I have rtl433 running in a docker picking up a few acurite temperature sensor and a radio head ASK home made sensor. Up until last week it was picking up a Accurite 896 rain gauge,worked well with RTL433 to monitor rain fall. it stopped working I am pretty sure the inside just got corroded over time. So I replaced it, the new one came today, plugged it up and ran RTL433 manual to find device -R 10 which is Accurite 896, but its not picking up the signal which I know is being transmitted to the little display that came with the unit as I can make the rain go up by manually moving the tilt sensor. Both the rain unit and the display have accurite 896 on them , although the rain gauge says 896tx1 and the old one say 896tx . 

Any thoughts as to why I can't pick up the signal. I pick up everything else when I add my -R devices

zuckschwerdt, 2020-01-14:
Is the Acurite 896 sensor on 868 MHz? The maybe the new demod is not well suited, try `-Y classic`. Other wise try to capture a sample (capture many and remove the ones that decode to something) and post it here.

faspina, 2020-01-14:
I ran rtl_433 -S unknown.  It recognize my existing devices ok.  But saved a a .cu8 file every 30 seconds for something that is out there. Here is the first ones name, 
g001_433.92M_250k.cu8
So I brought the rain sensor in and tilted it a couple of times , a new file was written every time it changed on the display unit. I am fairly certain that its picking it up its just not registering. 

RTL_433 -R 10 did not pick up the changes. 

How do I ready this file and figure out what is going on.

merbanan, 2020-01-14:
Run rtl_433 -A -r rainsensor_file.cu8.

merbanan, 2020-01-14:
Attach a few files here also.

faspina, 2020-01-14:
Here you go thank you.  

Test mode active. Reading samples from file: rainsensor_file.cu8
Detected OOK package    @0.133936s
Analyzing pulses...
Total count:   36,  width: 52.67 ms             (13167 S)
Pulse width distribution:
 [ 0] count:    6,  width:  116 us [108;136]    (  29 S)
 [ 1] count:    6,  width:  268 us [228;320]    (  67 S)
 [ 2] count:    5,  width:  180 us [168;188]    (  45 S)
 [ 3] count:   15,  width:  392 us [340;456]    (  98 S)
 [ 4] count:    2,  width:   48 us [44;52]      (  12 S)
 [ 5] count:    2,  width:   72 us [68;76]      (  18 S)
Gap width distribution:
 [ 0] count:    4,  width:   56 us [52;64]      (  14 S)
 [ 1] count:   14,  width: 1136 us [1028;1380]  ( 284 S)
 [ 2] count:   13,  width: 2056 us [1984;2108]  ( 514 S)
 [ 3] count:    2,  width:  124 us [116;136]    (  31 S)
 [ 4] count:    1,  width:   88 us [88;88]      (  22 S)
 [ 5] count:    1,  width:   44 us [44;44]      (  11 S)
Pulse period distribution:
 [ 0] count:    4,  width:  156 us [140;172]    (  39 S)
 [ 1] count:   14,  width: 1424 us [1180;1680]  ( 356 S)
 [ 2] count:    3,  width:  244 us [220;264]    (  61 S)
 [ 3] count:   13,  width: 2360 us [2072;2520]  ( 590 S)
 [ 4] count:    1,  width:  316 us [316;316]    (  79 S)
Level estimates [high, low]:   1000,     48
RSSI: -12.1 dB SNR: 13.2 dB Noise: -25.3 dB
Frequency offsets [F1, F2]:   16128,      0     (+61.5 kHz, +0.0 kHz)
Guessing modulation: No clue...

[rainsensor_file.zip](https://github.com/merbanan/rtl_433/files/4062155/rainsensor_file.zip)

faspina, 2020-01-14:
[output.txt](https://github.com/merbanan/rtl_433/files/4062279/output.txt)

faspina, 2020-01-14:
I ran this . Both files attached. 

rtl_433 -r g001_433.92M_250k.cu8 -a 4  -A &> output.txt

[rtl_433.zip](https://github.com/merbanan/rtl_433/files/4062283/rtl_433.zip)

faspina, 2020-01-16:
Is there anything you need on this. I can't figure out why its not picking it up. I really hate to order a 2nd one just to see if it works.

zuckschwerdt, 2020-01-16:
The sample does not look like the expected PPM. There are some strange gaps. The device is broken, the battery emtpy, or it's some strange new encoding (unlikely).
[image]

closed 2020-01-16 by zuckschwerdt