merbanan/rtl_433 #1266: doxygen not listed as requirement for building

nivex, 2020-01-18:
doxygen is not listed in the documentation as a requirement for building:
```
kjotte@daedalus:~/git/rtl_433/build$ grep -i doxygen ../README.md 
kjotte@daedalus:~/git/rtl_433/build$ grep -i doxygen ../docs/BUILDING.md 
```
but `cmake` refuses to complete without it:
```
kjotte@daedalus:~/git/rtl_433/build$ cmake ..
-- Build type not specified: defaulting to release.
-- Found Git version: refs/heads/master commit c496a5fd536c7485de786737af50310102fef8d0 from 2020-01-18T11:20:46+0100
-- Using Git version tag: 19.08-146-gc496a5f on master at 202001181120
-- RTL-SDR device input will be compiled.
-- SoapySDR device input disabled.
CMake Warning at /usr/share/cmake-3.10/Modules/FindDoxygen.cmake:396 (message):
  Unable to determine doxygen version: No such file or directory
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindDoxygen.cmake:551 (_Doxygen_find_doxygen)
  CMakeLists.txt:177 (find_package)

CMake Error at /usr/share/cmake-3.10/Modules/FindDoxygen.cmake:630 (message):
  Unable to generate Doxyfile template: No such file or directory
Call Stack (most recent call first):
  CMakeLists.txt:177 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/kjotte/git/rtl_433/build/CMakeFiles/CMakeOutput.log".
```

Please either remove doxygen as a requirement for building or amend the build docs to list it as a requirement.

nivex, 2020-01-18:
Kindly ignore this. Blowing away my `build` directory and recreating it allowed proper detection to occur.

closed 2020-01-18 by nivex

zuckschwerdt, 2020-01-18:
No problem. doxygen is only a requirement if building with `BUILD_DOCUMENTATION=YES` and that's targeted at maintainers and CI.