merbanan/rtl_433 #1267: -M option frequencies

JSMSolns, 2020-01-20:
Would it be possible to change the frequency reporting for the -M option from 1 decimal place to 3decimal places please? 
This would help with identifying the centre and deviation frequencies more accurately (assuming this is appropriate).

I believe this should simply be a matter of changing the following lines in r_api.c from %.1f MHz to %.3f MHz:
664
665
674

zuckschwerdt, 2020-01-20:
Sorry, the frequency report is just a very rough estimation. We are currently modularizing the demodulator step and we might add a true dominate frequency (likely FFT) someday.

closed 2020-01-20 by zuckschwerdt

merbanan, 2020-01-20:
Currently it should only be used to identify what band the signal is in. We would need new dsp code for this. And it is unlikely that something like that will ever be written. At least by me.

JSMSolns, 2020-01-20:
Understood - thanks for the quick response