merbanan/rtl_433 #1268: Add support for TX141TH-BV3

tvass, 2020-01-23:
Hi (again),
I am now looking to add support for the sensor La Crosse TX141**TH**-BV3.
It should not be very from `[73]  LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W sensor`.

```
pi@pi3sdr:~/work/rtl433/build/src $ ./rtl_433 -X 'n=lacrosse-tx141th-bv3,m=OOK_PWM,s=256,l=500,r=1888,g=0,t=0,y=748' g001_433.92M_250k.cu8
rtl_433 version 19.08-144-g8d45456 branch master at 202001141626 inputs file rtl_tcp RTL-SDR
[...]
time      : @0.270072s
model     : lacrosse-tx141th-bv3                   count     : 12            num_rows  : 12            rows      : 
len       : 41           data      : 0c7d2ae2798, 
len       : 41           data      : 0c7d2ae2798, 
len       : 41           data      : 0c7d2ae2798, 
len       : 41           data      : 0c7d2ae2790, 
len       : 41           data      : 0c7d2ae2798, 
len       : 41           data      : 0c7d2ae2798, 
len       : 41           data      : 0c7d2ae2798, 
len       : 41           data      : 0c7d2ae2798, 
len       : 41           data      : 0c7d2ae2790, 
len       : 41           data      : 0c7d2ae2798, 
len       : 41           data      : 0c7d2ae2790, 
len       : 41           data      : 0c7d2ae2798
```
**Values**
[image]

**Samples**
[tx141v3.zip](https://github.com/merbanan/rtl_433/files/4100874/tx141v3.zip)

tvass, 2020-01-23:
More sample.
```
pi@pi3sdr:~/work/rtl433/build/src $ ./rtl_433 -X 'n=lacrosse-tx141th-bv3,m=OOK_PWM,s=256,l=500,r=1888,g=0,t=0,y=748' g002_433.92M_250k.cu8 
[...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.270108s
model     : lacrosse-tx141th-bv3                   count     : 12            num_rows  : 12            rows      : 
len       : 41           data      : 0c7e30da1a8, 
len       : 41           data      : 0c7e30da1a8, 
len       : 41           data      : 0c7e30da1a8, 
len       : 41           data      : 0c7e30da1a0, 
len       : 41           data      : 0c7e30da1a8, 
len       : 41           data      : 0c7e30da1a8, 
len       : 41           data      : 0c7e30da1a8, 
len       : 41           data      : 0c7e30da1a8, 
len       : 41           data      : 0c7e30da1a0, 
len       : 41           data      : 0c7e30da1a8, 
len       : 41           data      : 0c7e30da1a0, 
len       : 41           data      : 0c7e30da1a8
codes     : {41}0c7e30da1a8, {41}0c7e30da1a8, {41}0c7e30da1a8, {41}0c7e30da1a0, {41}0c7e30da1a8, {41}0c7e30da1a8, {41}0c7e30da1a8, {41}0c7e30da1a8, {41}0c7e30da1a0, {41}0c7e30da1a8, {41}0c7e30da1a0, {41}0c7e30da1a8
```
[tx141thbv3.zip](https://github.com/merbanan/rtl_433/files/4101030/tx141thbv3.zip)

[image]

zuckschwerdt, 2020-01-30:
Thanks for the data and samples. I'll have a look soon.

zuckschwerdt, 2020-01-30:
I've added preliminary support with 8ecd5bc. Please test, the battery and channel look wrong. Perhaps you play with the settings and use fresh and known flat batteries.

thermseekr, 2020-02-10:
Good morning!

Tested, temperature and humidity working perfectly. Channel shows 00, which I think is right as -BV3 model doesn't have a channel select switch. Test button is ok. Haven't yet tested with flat batteries.

Thank you!
Tales

ErebusBat, 2021-11-30:
I have one of these sensors... is there anything I can do to help move this forward?

merbanan, 2021-11-30:
Well it should just work. If not then it is probably a new id that needs to be added. So can you confirm or deny if it is working?

ErebusBat, 2021-11-30:
@merbanan It is NOT working for the TX141TH-BV3 (notice the TH / Temperature Humidity variant).

I just tested today running from a build that I made a bit earlier in the month (`21.05-143-g4ca3cb0d branch master at 202111171123`)

merbanan, 2021-11-30:
Ok, then we need a signal sample. Use rtl_433 -S unknown to record and then attach it to the issue.

ErebusBat, 2021-11-30:
@merbanan Interesting... I attempted to do this.  While I was getting captures, they did not seem to be associated with my triggering the BV3 sensor (via test button).  Shouldn't it have created a capture as soon as it got the test packet?

It was picking up my other BV2 sensors.

zuckschwerdt, 2021-12-01:
Yes, it should. Maybe this helps https://triq.org/rtl_433/ANALYZE.html

richardkuw, 2021-12-09:
Greetings!
Just to confuse the issue further, the model I just received is the tx141th-bchv3. It appears to be the same as the bv3 with the addition of a channel selector switch. I'm not familiar with this workflow but I did a capture (attached) and when I run the command previously referenced on the capture file I get this:
[capture.zip](https://github.com/merbanan/rtl_433/files/7687689/capture.zip)

#rtl_433 -X 'n=lacrosse-tx141th-bchv3,m=OOK_PWM,s=256,l=500,r=1888,g=0,t=0,y=748' file_433.92M_250k.cu

Registered 146 out of 175 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Test mode active. Reading samples from file: file_433.92M_250k.cu
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @1.304400s
model     : lacrosse-tx141th-bchv3                 count     : 4             num_rows  : 4             rows      : 
len       : 41           data      : e6bd37d88e8, 
len       : 41           data      : e6bd37d88e0, 
len       : 41           data      : e6bd37d88e8, 
len       : 41           data      : e6bd37d88e8
codes     : {41}e6bd37d88e8, {41}e6bd37d88e0, {41}e6bd37d88e8, {41}e6bd37d88e8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @1.598724s
model     : lacrosse-tx141th-bchv3                 count     : 4             num_rows  : 4             rows      : 
len       : 41           data      : e6bd37d88e8, 
len       : 41           data      : e6bd37d88e8, 
len       : 41           data      : e6bd37d88e8, 
len       : 41           data      : e6bd37d88e0
codes     : {41}e6bd37d88e8, {41}e6bd37d88e8, {41}e6bd37d88e8, {41}e6bd37d88e0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @8.485404s
model     : lacrosse-tx141th-bchv3                 count     : 4             num_rows  : 4             rows      : 
len       : 41           data      : e6fd39d8998, 
len       : 41           data      : e6fd39d8998, 
len       : 41           data      : e6fd39d8998, 
len       : 41           data      : e6fd39d8990
codes     : {41}e6fd39d8998, {41}e6fd39d8998, {41}e6fd39d8998, {41}e6fd39d8990
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @8.779733s
model     : lacrosse-tx141th-bchv3                 count     : 4             num_rows  : 4             rows      : 
len       : 41           data      : e6fd39d8990, 
len       : 41           data      : e6fd39d8998, 
len       : 41           data      : e6fd39d8990, 
len       : 41           data      : e6fd39d8998
codes     : {41}e6fd39d8990, {41}e6fd39d8998, {41}e6fd39d8990, {41}e6fd39d8998

Hopefully that is useful information. I am running version 21.05.

zuckschwerdt, 2021-12-09:
Thanks. A new release 21.12 is planned for the next days. It should *just work* then.

flupes, 2022-09-27:
Sorry to re-open this issue. Any information if this sensor should or should not be supported would be appreciated.

I also own a La Crosse TX141TH-BV3, temperature + humidity, without a channel selector, and I *cannot* detect it with the latest build.

Exact marking on the back:
  - IC: 5049A-TX141V3
  - Model: TX141TH-BV3

I believe my setup (radio wise) is correct because I am picking a few TX141TH-BV2 (with `-R 73`, probably from my neighbors).
I have attached one sample. I can capture more under various conditions if somebody first confirms it is of usable quality (I am new to this).

One could hope that the TX141TH-BV3 would simply be a mix of the TX141TH (humidity byte) and TX141-BV3 (4 repeat rather than 12) and the [code logic already in place](https://github.com/merbanan/rtl_433/blob/master/src/devices/lacrosse_tx141x.c#L129) seems to work on this assumption. However it fails very early for me (already at `bitbuffer_find_repeated_row` that return -1).

[lacrosse_tx141th-bv3_no-test_26C_55pc.zip](https://github.com/merbanan/rtl_433/files/9651636/lacrosse_tx141th-bv3_no-test_26C_55pc.zip)

zuckschwerdt, 2022-09-27:
You stripped the sample rate and center frequency info from the recording?
Look at the spectrogram https://triq.org/pdv/ -- the signal is nearly out of band, faint and distorting.
Adjust the freq up by half of the rate.

But maybe the sender is defective, there is a very prominent spike at the pulse end. This messes with AM level detection:
[image]

closed 2022-09-27 by zuckschwerdt

flupes, 2022-09-27:
Fantastic. This is exactly the type of feedback I needed.

I naively thought that the transmitter would use the default 433.92MHz, and was searching for the issue at the wrong place.
So I looked up at the signal with CubicSdr (as the excellent documentation explains), and found that my TX141TH-BV3 is not using the same frequency than the other La Crosse sensors I was discovering!
It works perfectly fine with:
`rtl_433 -f 433.78M -R 73`
(and filters out at the same time the real LaCrosse-TX141THBv2 I was catching before)

So I can confirm: `rtl_433` is working decoding perfectly fine my `TX141TH-BV3` (displayed as TX141TH-BV2, but I can work on this).
Thanks so much!

Output with/without test:
```
time      : 2022-09-27 16:23:44
model     : LaCrosse-TX141THBv2                    Sensor ID : 94
Channel   : 0            Battery   : 1             Temperature: 22.80 C      Humidity  : 66 %
Test?     : Yes          Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2022-09-27 16:24:18
model     : LaCrosse-TX141THBv2                    Sensor ID : 94
Channel   : 0            Battery   : 1             Temperature: 23.10 C      Humidity  : 65 %
Test?     : No           Integrity : CRC
```
[image]

zuckschwerdt, 2022-09-28:
Great to hear it quickly worked out. So the signal visible at the top (+freq) was actually an alias with the original out of band towards the bottom (-freq)? No wonder the I/Q looked strange then :)

flupes, 2022-10-01:
Correct. Sorry I did not realize to select the right frequency to start with.
I am surprised however that the frequency changed between the Bv2 and Bv3 version. I am sure of the shift since at 433.92 I can capture several Bv2!

For completeness, I attached clean samples captured with my Bv3.
I also have a [fork to differentiate the Bv2 vs Bv3 models](https://github.com/merbanan/rtl_433/compare/master...flupes:rtl_433:master) in the output. If there is any interest, I can create a PR.

[lacrosse_tx141th-bv3_433.78M.zip](https://github.com/merbanan/rtl_433/files/9691108/lacrosse_tx141th-bv3_433.78M.zip)

thermseekr, 2022-10-02:
This is weird, I have both BV2 and BV3 working with the same setup. I did not check frequencies, just put in the batteries and started receiving data.

Enviado do meu iPhone

zuckschwerdt, 2022-10-02:
I'd say that inaccurate freqs are manufacturing tolerance. I often use `433.85M` here to get more devices.

I'm afraid that splitting the models on a weak attribute might do more harm than good. E.g. if a single bit is missed or added falsely by the demod your sensor would get a different name.