merbanan/rtl_433 #1269: [Feature] ouput noise only (jamming detection / noise monitoring /sensor signal under noise)

dgrostoto, 2020-01-24:
Hello,
I'd like to suggest a 'new' feature (did not see it in current docs, let me know if I'm wrong), about noise management.
Use cases :
- jamming detection : be able to detect jamming on current frequency, can also used when hoping to different frequency (make a noise measurement at each hop)
- noise monitoring : a notch above, noise in continuous monitoring (and potential alerting in third party software for variations etc...)
- be able to tell if certain sensors signals be 'under' the noise (to be differentiated from a sensor failure or battery shortage, since some sensor are not reporting battery level or not really good at reporting battery level) 
Presently, we have RSSI, SNR and noise for each signal report (or most of them i'm not sure), but if noise if above a certain level (bad wired DIY sensor, or some sensor may act as jammer with low battery) we don't have anything...
Please note that some RF transceiver have this functionnality of jam detection 
https://www.planete-domotique.com/blog/2018/06/08/jamming-detection-brouillage-rfplayer/
And i really think it would be a great feature.
I maybe can implement this feature if you agree (something like -M noise plus a polling option to repeat noise measurement)
Thanks in advance.

zuckschwerdt, 2020-01-30:
Note that noise level here is subject to the presence of a signal level (SNR). We could detect a uniformly distributed buffer and assume it's noise -- but that's already implied by not detecting a signal ;)
But we can't really put a (relative) level to the noise as we have no reference. And no absolute level as we don't know effective gain.
Maybe if you were to run with fixed gain and measure a baseline noise. Likely it would need to account for drift an variation and you would block out too weak/ too strong signals.
That's on RTL-SDR (8-bit) hardware. I guess you could do more with 12-bit or better hardware.

dgrostoto, 2020-01-31:
I'm not sure to know much about RF but if there is no reference when measuring noise level so noise reported when reporting two different sensors data, let's say noiseA for sensorA and noiseB for sensorB (sensorA and sensorB don't have the same signal level) cannot be compared, can they ?
so noise level is dependant on each sensor ? and you could only compare two noise measurement for different signals for the same sensor devices ?
A note about gain, the gain is specified when launching rtl_433 software and can be influenced by the quality of hardware and the antenna,  so i understand it may be difficult to compare noise level between two different piece of hardware (the receiver). But for the same piece of hardware (with same antenna), could noise level be compared ?

zuckschwerdt, 2020-01-31:
Basically, yes. But usually (with RTL-SDR) you would use automatic gain to get a wider range. Manual gain really limits the signals you can get. Also the same sensor could read at different (auto-) gain settings (maybe battery got weaker) or such and thus the noise would change.
Note that the automatic gain might change in the middle of a signal.

If you really want a reliable noise figure you should scan across a wider band at a number of gains, maybe with something like rtl_power.

Closing as this isn't realisticly something rtl_433 can deliver. But feel free to further discuss the idea and perhaps let us know what worked for you in the end.

closed 2020-01-31 by zuckschwerdt

dgrostoto, 2020-02-01:
Thanks for pointing another program to study (rtl_power) as i did not found any program which is specifically dedicated to noise measurements (well nothing right out of the box or so). I will also 'play' with the code that is evaluating the noise in rtl_433 to see if i can get something usable for my needs.
Anyway, i need at least to cover the jamming detection case since rtl_433 is presently not be able (i think) to detect any signal (the jammer will cover any sensor signals, so nothing will be reported).

zuckschwerdt, 2020-02-01:
Jamming is tricky. One strong carrier (could be needle thin) might swamp the gain. I wouldn't even need to be in the band, just near.
On the other hand weak wide spectrum emission could also shadow signals. Or maybe a fast hopping carrier signal.
Also you could just detect a transmission and instantly tune there and shadow everything after the preamble. It's used. And you won't detect it statically.