merbanan/rtl_433 #1270: Lacrosse TX60U-IT series support

romka1977, 2020-01-26:


zuckschwerdt, 2020-01-27:
We need at least:
- [ ] description
- [ ] sample files
- [ ] table of codes and expected values

romka1977, 2020-01-27:
Thanks for your reply @zuckschwerdt 

Unfortunately i dont have much of experience of providing the above.
I do own this sensor https://www.lacrossetechnology.com/add-on-temperature-and-humidity-sensor-with-threaded-wet-temperature-probe/
and can run rtl_433  but i would need some guidance how exactly to run it and what to capture

Thanks!

zuckschwerdt, 2020-01-27:
Run `rtl_433 -S all` to collect a few samples (e.g. 5). Then upload a .zip here. (use http://triq.net/iqs to get a picture what's in the sample files if you like.)
We can then figure out the modulation.

romka1977, 2020-01-28:
[g00x_915M_1000k.zip](https://github.com/merbanan/rtl_433/files/4120029/g00x_915M_1000k.zip)

Captured with "-f 915M" based on the info below TX60U-IT sensors use 915 Mhz
https://www.weathershack.com/static/lacrosse-sensor-compatibility-chart.html

zuckschwerdt, 2020-01-29:
Yes, the last sample in the set works. You can run:
```
rtl_433 -R 0 -X 'n=LaCrosse-TX60UIT,m=OOK_MC_ZEROBIT,s=31,r=90'
```
to get a series of outputs like
```
codes     : {96}01530470081c1688c3549769
```
Create a table (just some text to post here) of codes and expected values contained. E.g.

Code | Values
------ | ------
{96}01530470081c1688c3549769|42 C, 23 %

romka1977, 2020-01-29:
ok. This unit has 3 sensors - 2 temp and 1 humidity. I just captured the below data now with timestamps.Will add more samples later.
**TEMP=-14.4C
PROBE_TEMP=-1.9C
HUMIDITY=81%**

`08:45:16 {97}0153023700fee6c3142b2c4a8`
`08:45:20 {97}015302370118a0cbac01d2ae8`
`08:45:26 {96}015304b0069234530fff2c7b`
`08:45:31 {5}08`
`08:45:35 {96}015304f00298708812f8c658`
`08:45:36 {97}015302370277e4ca55b610b68`
`08:45:36 {96}015304300b01408173e2ccd8`
`08:45:49 {96}015304f006b8708812f8c659`
`08:45:54 {5}30`
`08:46:04 {6}04`

merbanan, 2020-01-29:
Can you collect some output where you only vary one sensor at the time?

zuckschwerdt, 2020-01-29:
Temp should be 12 bits (3 hex characters), and Hum 8 bits (2 hex chars). There is way too much data in the packet for just this (even if we remove header/id and checksum).
Also the data is all over the place, the values look mostly random. We needs lots more data to figure this out.

romka1977, 2020-01-29:
ok. i disconnected the probe which leads it to report "N/C" for probe temp.. I will post result in my original post soon trying to keep humidity constant

merbanan, 2020-01-29:
Get some samples with the probe in and out, it should be easy to figure out where it is in the bitstream.

romka1977, 2020-01-29:
btw, device id - 000134AA0B3C152A if it helps. Its on the label and on the web gui

romka1977, 2020-01-29:
added more data

romka1977, 2020-01-29:
looking at it, would it be correct to assume?
`015302370277e8ca55b65ec1 >> PROBE_TEMP N/C`
`015304f006b8a08812f81b72  >> TEMP -6.7C`

merbanan, 2020-01-29:
That looks like a scrambled payload. Just put the sensor in the fridge and record the output.

zuckschwerdt, 2020-01-29:
We are looking for simple changes in one place only, like a slight temperature change with everything else the same.
But looking at the [data in a BitBench](http://triq.net/bitbench?c=0153023700fee6c3142b2c4a%20%5B-14.4C%20-1.9C%2081%25%5D&c=015302370118a0cbac01d2ae&c=015304b0069234530fff2c7b&c=015304f00298708812f8c658&c=015302370277e4ca55b610b6&c=015304300b01408173e2ccd8&c=015304f006b8708812f8c659&c=015302370118a0cbac01d2ae%20%5B-5.7C%20N%2FC%2084%25%5D&c=015304f006b89c8812f803ba&c=015302370118a0cbac01d2ae&c=015302370277e8ca55b65ec1&c=015302370277e8ca55b65ec1%20%5B-6.7C%20N%2FC%2078%25%5D&c=015304f006b8a08812f81b72&c=015304f009175e81777ad3b2&c=015302370277e8ca55b65ec1%20%5B-6.7C%20N%2FC%2077%25%5D&c=015304f006b8a08812f81b72&c=015304f007a66c81781405d2&c=0153023600cd34cc7135623a&c=015302370277e8ca55b65ec1&c=0153043006a3aec3667cd5d9&f=20h%204d%208h%208h%208h%208h%208h%208h%208h%208h%208h%20) I see no pattern. All the bits change.

zuckschwerdt, 2020-01-29:
Just checked the MC. If I look at the raw PCM data (using
```
rtl_433 -s 1M -R 0 -X 'n=LaCrosse-TX60UIT,m=OOK_PCM,s=31,l=31,r=90'
```
I get
```
aa96665a55656a55559556a556699595a55a6665966a69960
```
In [a BitBench](http://triq.net/bitbench?c=aa96665a55656a55559556a556699595a55a6665966a6996&f=8h%20&d=MC) that does decode to the expected `(01)530470081c1688c3549769`.

romka1977, 2020-01-29:
sorry, is it good or bad?:)

romka1977, 2020-01-29:
sensor has stabilized in the fridge so i would post that data

zuckschwerdt, 2020-01-30:
My comment means we didn't get the decoding wrong. The data is really that gibberish.
Assuming the sensor is at a stable reading in the freezer all these should be the same
```
01530 2370 118a0cbac01d2ae [3.1C N/C 70%]
01530 4b00 793f4cb43cf5995
01530 4300 4e12e87d5135ccd
01530 2370 0fee6c3142b2c4a
01530 4f00 9178481777ad064
01530 2370 0fee6c3142b2c4a
```
But it's wildly all over the place. I wonder if the data is encrypted?

There might be different message types. So we pick rows that start with the same pattern:
```
01530 2370 118a0cbac01d2ae [3.1C N/C 70%]
01530 2370 0fee6c3142b2c4a
```
Now there should be some transform to make the codes the same, like "take 5th byte and xor to all remaining bytes".

zuckschwerdt, 2020-01-30:
If we look at only the `2370` prefixed messages:
```
01530 2370 0fee6c3142b2c4a [-14.4C -1.9C 81%]
01530 2370 118a0cbac01d2ae
01530 2370 277e4ca55b610b6
01530 2370 118a0cbac01d2ae [-5.7C N/C 84%]
01530 2370 118a0cbac01d2ae
01530 2370 277e8ca55b65ec1
01530 2370 277e8ca55b65ec1 [-6.7C N/C 78%]
01530 2370 277e8ca55b65ec1 [-6.7C N/C 77%]
01530 2370 277e8ca55b65ec1
01530 2370 118a0cbac01d2ae [3.1C N/C 70%]
01530 2370 0fee6c3142b2c4a
```
They have identical payloads across all readings. I don't think those message are related to measurements. Strange.

zuckschwerdt, 2020-01-30:
Ah well, small progress: You managed to capture a single bit difference message (those are key to reversing a checksum):
```
0153 0237 0277 e4ca 55b6 10b6
0153 0237 0277 e8ca 55b6 5ec1
```
Neatly shows that there is a 16 bit checksum at the end!
It works as CRC-16, poly 0x6f63, init 0x0153.
The init suggests that maybe we shouldn't check the preamble, and yes taking 0x0153 as fixed preamble and excluding that we get a very likely and known CRC:
CRC-16, poly 0x6f63,  init 0x0000  "CRC-16/LJ1200".

So, data is good, alignment is known. Now to make sense of the payload...

zuckschwerdt, 2020-01-30:
Sorry, preamble is a pattern like 0x5555, The 0x153 here would be a sync-word.
With the sync-word and the CRC-16 this very much looks like a transmission an advanced chip like the CC1101 would make. And there is a feature known as whitening on those chips. The payload will be xor'ed with a fixed white-noise pattern to reduce runs of 0's or 1's. So we might need to *just* deduce the whitening applied here.

zuckschwerdt, 2020-01-30:
An idea would be to use a known-plain-text attack. We need a days worth of payload from a controlled very stable environment (perhaps the freezer) to get a baseline of message types.
That should show only a very limited number of different payloads.
Then you would need to very slowly change parameters so we can capture a steady progression of readings. E.g. letting the sensor warm up or cool down very slowly. Ideally this would show the same messages for say 10-20 minutes, then the next set of messages...

romka1977, 2020-01-30:
thanks guys for looking into this. It sounds like it would be an endeavor to make this work. Looks like this sensor works completely differently than other LaCrosse units. I personally dont see much value spending more time on this. Feel free to close this issue unless you think otherwise.

zuckschwerdt, 2020-01-30:
Yeah, it's another of those botched protocols. There is no reason to have whitening if you use Manchester coding, both address the same problem (efficient DC-free codes). The engineer clicked the wrong chip options when putting this together.
If you happen to record more codes, dump them here, no need for timestamps or the readings. Someone might have fun with this. I know I do :)

closed 2020-01-30 by zuckschwerdt

romka1977, 2020-01-30:
sure, i can do json logging for a day or so and also attach timestamped sensor data. I resorted to web gui scraping to get that data and it is being logged.

zuckschwerdt, 2020-01-30:
Nice! That might prove effective still. Since the values then are exact, I'll just build a fat lookup table and deduce somethinb from that …maybe ;)

romka1977, 2020-01-30:
[image]

This is the bad boy. Soldered chip is 24c02a

Board itself has same "ID" e255943 as this one
https://commons.wikimedia.org/wiki/File:Technoline_TX29DTH-IT_-_sensor_and_controller-4532.jpg

merbanan, 2020-01-30:
https://www.ti.com/lit/an/swra322/swra322.pdf is the one used for the CC11xx family of chips.

zuckschwerdt, 2020-01-30:
Great! An eprom is really much effort for such cheap devices. But I take it these are uniquely identified using an IoT bridge to go cloud? And …so many test points! Looks like they brought out all-the-pins. The chip-blob near the antenna (snaking the empty board half) would be the transmitter, the other one (near the temp and hum sensor) an MCU? At least the board is single layer …no money wasted on that …other than throwing like 12 0-ohm links on it :)

romka1977, 2020-02-03:
[TX60_sensor_data.zip](https://github.com/merbanan/rtl_433/files/4149880/TX60_sensor_data.zip)
[TX60IUT_915M.log.gz](https://github.com/merbanan/rtl_433/files/4149881/TX60IUT_915M.log.gz)

as promised.
Data reported by sensor:
**utctime** - when sensor reported the data (same timezone as rtl-433 capture)
**ambient_temp** - sensor temperature, degrees C
**probe_temp** - sensor probe temperature, degrees C
**humidity** - humidity %

Let me know and have fun:)

zuckschwerdt, 2020-02-04:
Here are the CRC-16 verified clean codes. There are lots of single bit changes in the bits 47 to 55. It suggests a static encryption key (whitening) with actual readings in those bits. I'll try to map the readings to those bits now.

[TX60_sensor_data.zip](https://github.com/merbanan/rtl_433/files/4153344/TX60_sensor_data.zip)

zuckschwerdt, 2020-02-04:
Some info on the TCP stream format might be here: http://www.wxforum.net/index.php?topic=14299.0

romka1977, 2020-02-04:
yeah i read this stuff, unlikely it will help with this issue. From the gateway to the cloud they send it like 21 degrees encoded as 0x21 hex..so they just take decimal digits an put them in hex.