merbanan/rtl_433 #1271: Abarth 124 TPMS

TTigges, 2020-01-26:
### Introduction

We are receiving signals from the TPMS sensors of our (Fiat) Abarth 124 Spider TPMS sensors. It’s a bit of a unique case because the 124 Spider are actually built by Mazda on the base of the MX-5. However, only the Abarth models seem to have active TPMS sensors. Unfortunately, the car does not display the sensor infos. I am testing with a set of tires sitting inside (controled environment, temperature and pressure are known) as well as a set of tires mounted on the car with copied sensors. I also got codes from friends, also recorded with rtl_433.

### Auto suggested profile/protocol

With _-G_, rtl_433 suggests the sensors are Jansite sensors but the values are off. So from this, I assume the signals are similar/having a similar preamble? Or how is the fitting device protocol selected? (More below at #First try at a device protocol# and #What I don't understand.#)

Where Jansite’s code is
II II II IS PP TT CC

my code seems to be
?? ?? II II SS PP TT

### Received codes and first notes:

Right now, we have checked 3 cars with 4 sets of tires total = 16 sensors.

**Set A 1**
- 0f54771bb79045
- 0f5476e8b78f45
- 0f5476eab79140
- 0f547711b79043

**Set A 2**
- 0f54771bb49835
- 0f5476e8b6ac35
- 0f5476eab5ab35
- 0f547711b48e35

Notice the same IDs (byte 3 + 4). Byte 5 (expected status) behaves a bit weird, though.

**Set B**
- 0f365c01078d40
- 0f365beb079141
- 0f38cb2f078e42
- 0f365c1a079043

Here, the 5th byte is uniform (07) again. I was surprised that the 2nd byte is different for one of the sensors.

**Set C**
- 0c15590367903b
- 0c2990b667913d
- 0cbf53cb679e3c
- 0f492f8367923c

The 5th byte is uniform here, too. But the first 2 bytes are a mess here. I’m surprised by the „0c“.

TTigges, 2020-01-26:
## BitBench:
http://triq.net/bitbench?c=0f54771bb79045&c=0f5476e8b78f45&c=0f5476eab79140&c=0f547711b79043&c=&c=0f54771bb49835&c=0f5476e8b6ac35&c=0f5476eab5ab35&c=0f547711b48e35&c=&c=0f365c01078d40&c=0f365beb079141&c=0f38cb2f078e42&c=0f365c1a079043&c=&c=0c15590367903b&c=0c2990b667913d&c=0cbf53cb679e3c&c=0f492f8367923c&f=%3F%3F%3A%208h%20%3F%3F%3A%208h%20ID%3A%208d%208d%20STATUS%3A%204h%204d%20PRESS%3A%208d%20TEMP%3A%208d

## Excel list of sensors and decoded values:
https://docs.google.com/spreadsheets/d/1ZGbVTRssBKqKV_I3kSD5BuykL3l_xucRNVISEH8U45A/edit?usp=sharing

TTigges, 2020-01-26:
### Known values:

I made a lot of tests with set A1 with known temperatures and known pressures and found that the last (7th) byte in decimal – 50 is exactly the temperature.
I also noticed that the 2nd to last (6th) byte changed with the pressure. From several runs from 0.5 to 2.5 bar I found that the value as decimal * 1.4 was the pressure in kPa (+/- 1 – but that could also be my compressor or pressure tool that’s 1 kPa off).

### Unknown values:
Since that’s all that I really need, I’d be happy with that and just call the 3rd and 4th bytes the IDs, but here’s what I’m wondering:
Sets B and C were received during driving. Similar to A 1, the 2nd nibble of the 5th byte is 7. Only A 2 is off there. Set A was received during rapid pressure change.
In one case while receiving values during rapid pressure changes I received a signal from one of the sensors mounted on the car outside and the 5th bit was 74 where it was b4 before during filling.

### Thoughts about IDs:
There is also a dealer that is shipping these sensors „pre coded“, implying all cars have the same code (which I’ve never seen, actually, and can’t really believe). That would also imply that it doesn’t matter which wheel/corner the sensors are put at. That would make sense if the car just received any signals without caring about the position – as it doesn’t show any info besides „Flat Tire Warning“ anway.

TTigges, 2020-01-26:
## RTL_433
[Abarth_124_433_recorded-signals.zip](https://github.com/merbanan/rtl_433/files/4114519/Abarth_124_433_recorded-signals.zip)

### Pull request for these recordings:
https://github.com/merbanan/rtl_433_tests/pull/329

### Result comparison:
https://docs.google.com/spreadsheets/d/1ZGbVTRssBKqKV_I3kSD5BuykL3l_xucRNVISEH8U45A/edit#gid=529440585

There are two signals that were recorded, that have the same file size but were not recognized.
I also received a signal from outside. Check out the status. During de-/inflating that tire, the status byte was b4, this time it was 74 => seems to be a regular signal, not after a rapid pressure change.

### First try at a device protocol:
Since the sensors where first detected as Jansite TPMS, I used a copy of it to get quick results:
https://github.com/TTigges/rtl_433/blob/master/src/devices/tpms_abarth.c
Comments are still for  the Jansite TPMS, though. (Needs to be fixed!)

**I'm a bit at a loss here.**

TTigges, 2020-01-26:
### What I don’t understand, yet:
In audacity, the signal looks totally different to that of the Jansite TPMS. And I don’t recognize any preamble. I looked at this issue on here about the Jansite TPMS (https://github.com/merbanan/rtl_433/issues/1007) and if I understood some oft he comments correct, other signals with missing CRC were suggested as Jansite TPMS, as well.
So, ist he signal suggested to be a Janiste TPMS sensor based on the missing/unknown CRC and not because of the preamble?

### Final goal: To read the signals via CC1101 on Arduino:
Similar to https://www.hackster.io/jsmsolns/arduino-toyota-uk-tpms-tyre-pressure-display-b6e544 and https://www.hackster.io/jsmsolns/arduino-renault-tpms-tyre-pressure-display-dfb548. @JSMSolns
At the moment, I’m diffing the files and comparing them to my codes to learn the differences of the sensors and their signals.

zuckschwerdt, 2020-01-27:
Very thorough investigation so far! Thanks for the nice formatting, easier to follow this way.

Some first notes: the Jansite has an unknown checksum method, it can't be verified currently. Thus any pulse pattern with roughly matching timing will display as Jansite. It's better to use a exactly matching flex spec.
Most TPMS have the same (standard) timing though. The difference is only visible by length and checksum used.

There really should be a checksum (usually in the last byte).

If the pressure conversion looks like 1.4 ~= 1.37 it would indicate 1/5 PSI steps, maybe.

Pre-coded tires/sensors would be the right protocol with random IDs. The head-unit might be able to learn a new set of tires automatically.

TTigges, 2020-01-27:
Thank you, that clears up a little bit. And yeah, better formating and a few minutes spent structuring always helps productivity.

### No checksum visible:
I have worked with decoding BLE sensors before, the code was much longer so initially I didn't trust the received 433 codes, specially since I couldn't identify a checksum. But I'm very confident in my observations of pressure and temperature. With increasing temperature, only the last byte changed and that decimal value -50 was always the exact temperature.

### No evidence is evidence:
I'd challenge that, if I had a doubt, but **all** sensors in one room with a temp. of 18°C showing _44_ at the last byte (44h => 68d - 50 = 18°C) even though the IDs and pressure values are different, and increasing by 1 to 45 when the valve is warmed up 1°C. Or all outdoor sensors showing 34 => 2°C ... that can only be the temperature.

Similar, when the pressure changes, only the 2nd to last byte changes.

### New observation: Idle status?
Last night I had rtl_433 listening and received idle messages. Before, most of the messages came in after a rapid pressure change. My "indoor" sensors all showed a status at byte 5 _(to be challenged?)_ of _b7,_ hile the mounted tires's sensors showed _b4, b5_ and _b6._

~All 8~ 7 of the sensors, both indoor and mounted, showed idle messages with a _7_ instead of the _b_ (77, 74, 75, 76). So it's maybe only the first nibble of the 5th byte that shows the status.

Each sensor seems to send an idle message close to, but not exactly every 4 hours.

### A thought about the status:
But looking at the other sensors, they have _(uniformly for all 4 of their sensors, though)_ either _07_ or _67_.

I have an assumption: All sensors with the 7 in the 2nd nibble of the 5th byte are OEM.
My 3rd party sensors are copies of the OEM and the only difference in the code are these 4 bits in the 2nd half of the 5th byte.

- b7: rapid pressure change, OEM
- 77: idle message, OEM
- b4: rapid pressure change, 3rd party sensor with copied ID (two of them have "b4")
- b5: rapid pressure change, 3rd party sensor with copied ID
- b6: rapid pressure change, 3rd party sensor with copied ID
- 74: idle message, 3rd party sensor
- 75: idle message, 3rd party sensor
- 76: idle message, 3rd party sensor

Only guessing (to be verifieed):
- 67: while driving(?), OEM(?)
- 07: while driving(?), OEM(?)

OEM sensors are said to be **Continental/VDO Type TG1C**.
3rd party sensors are from Aligator.
Theory: 3rd party sensors have a random code pre programmed. The ID _up until the status bits_ got copied from the OEM and those 4 static bits between status and pressure remained untouched and unused?

### A thought about the ID:
So far, just from the sensors I have, I'd say it's

(?? ?? ?I II) S(X) PP TT

Everything in brackets seems static and my guess would be, the 20 bits of ID is just the result of the static part being increased during the production run. So the closer the IDs are, the closer they where in the production process.

0f547... one batch. 0f365... another batch. With one 0f38c... somehow finding it's way in between.
Looking at it that way, *the whole first 4 bytes could be seen as the ID.*

merbanan, 2020-01-27:
Usually 3 bytes (24bits) or 4 bytes are used as id. I don't think anyone would implement a 2.5 bytes id.

zuckschwerdt, 2020-01-27:
Yes, IDs so far are mostly 28 bits (7 nibbles are commonly printed on the sensors), some look like or have room for 32 bits, but the presumed first ID nibble is most likely some status code.

TTigges, 2020-01-28:
That's actually what I ment with the parts in brackets. I used _"ID"_ for the part that made the sensors identifiable in the set of four.

zuckschwerdt, 2020-01-28:
What's really surprising (and bad) here is the lack of any checksum. If you don't known which IDs you are looking for then any such transmission can be seen to fit this protocol. I.e. I expect other TPMS will false-positive as this protocol regularly :/

TTigges, 2020-01-28:
Is there something we could have missed?

merbanan, 2020-01-28:
I took one signal and manchester decode on it manually. I then got 76 bits which is 9.5 bytes. Your code only extract 56 bits -> 7 bytes. To me this indicates that there might be an 8th and a 8th byte. Can you verify this? I am almost 99% sure there is a crc at the end.

merbanan, 2020-01-28:
http://triq.net/bitbench?c=15555555956a99995a9a6a65669a9996995955559a596599c&c=15555555956a99995a9a6a65669a9999995955559a565599c&c=15555555956a99995a9a95a6a69a99aa59595555a5a6a599c&c=15555555956a99995a9a95a6a69a99aa99595555a5a66599c&f=%3F%3F%3A%208h%20%3F%3F%3A%208h%20ID%3A%208d%208d%20STATUS%3A%204h%204d%20PRESS%3A%208d%20TEMP%3A%208d%20CRC%3A%208h%20XX%3A%208h&a=Preamble&m=a9&d=MC

merbanan, 2020-01-28:
And it looks like it is a simple checksum.

merbanan, 2020-01-28:
So extract one more byte and dump alots payloads, then we can try to figure out the checksum.

zuckschwerdt, 2020-01-28:
I'd say `ID:32h STATUS: 4h 4d PRESS: 8d TEMP: 8d CHK: 8h XX: 8h` -- but yes, that looks right.
The 8th byte does not look like a proper CRC, more like a simple sum.

zuckschwerdt, 2020-01-28:
Codes from the samples are:
```
a8edd16ea68802c90a
a8edd16eaa8802c40a
a8ee376ebc8803370a
a8ee376ebe8803350a
a8edd16ea68802c80a
a8edd16ea68802c82a
a8edd16eaa8802c42a
a8edd16eaa8802c50a
a8ee376ebc8803360a
a8ee376ebc8803362a
a8ee376ebe8803340a
a8ee376ebe8803342a
a8ee236edc8803420a
a8ee236edc8803422a
a8ee236edc8803430a
a8edd56ee68002840a
a8edd56ee68002842a
a8edd56ee68002850a
a8ee22e9506805ae0a
a8ee22e9506805ae2a
a8ee22e9506805af0a
a8edd16eec8802820a
a8edd16eec8802822a
a8edd16eec8802830a
a8edd16f008a036c0a
a8edd16f008a036c2a
a8edd16f008a036d0a
a8ee376ee488036e0a
a8ee376ee488036e2a
a8ee376ee488036f0a
a8ee376ef688037c0a
a8ee376ef688037c2a
a8ee376ef688037d0a
a8ee376f0a8a02820a
a8ee376f0a8a02822a
a8ee376f0a8a02830a
a8ee376f148a029c0a
a8ee376f148a029c2a
a8ee376f148a029d0a
a8ee236f168802880a
a8ee236f168802882a
a8ee236f168802890a
a8ee236f1e8a02820a
a8ee236f1e8a02822a
a8ee236f1e8a02830a
a8edd56f108003720a
a8edd56f108003722a
a8edd56f108003730a
a8edd56f228003400a
a8edd56f228003402a
a8edd56f228003410a
a8edd56f2a8003480a
a8edd56f2a8003482a
a8edd56f2a8003490a
```
Strangely otherwise identical codes will differ in the checksum depending on the last byte.

TTigges, 2020-01-28:
I have to admit, after getting _Jansite_ suggested, I just played around with the code that was shown there. Should have checked for more bits but it didn't cross my mind that the code output couldn't be trusted.

But these new codes seem strange to me: 54 codes from my 21 samples?
And the output made more sense before.

TTigges, 2020-01-28:
How did you get _"15555555956a99995a9a6a65669a9996995955559a596599c"_?

I used _" rtl_433 -r 01_FR_1_433.92M_250k.cu8 -R 123 -A -vvvv"_ now and saw this:

d5555554d52acccd4acb2b3532cacccccd4d5554cb5334cc60

### New bitbench:
http://triq.net/bitbench?c=d5555554d52acccd4acb2b3532cacccccd4d5554cb5354cc60&c=d5555554d52acccd4acb2b3532caccd2cd4d5554cb4d34cc60&c=d5555554d52acccd4acad4b2b2caccab4d4d5554b4b2cccc60&c=d5555554d52acccd4acad4b2b2caccaacd4d5554b4b334cc60&c=d5555554d52acccd4acad4d4b2cacb2b4d4d5554b354b4cc60&c=d5555554d52acccd4acad4d4b2cacb2b4d4d5554b354b4cc40&c=d5555554d52acccd4acb2b3332cacad2cd555554cd5334cc60&c=aaaaaaa9aa55999a9595a9a9959a666aa59aaaa65995a99800&c=d5555554d52acccd4acb2b3532cacacb4d4d5554cd54b4cc60&c=d5555554d52acccd4acb2b3532cab5554d4cd554b2cb34cc40&c=d5555554d52acccd4acad4b2b2cacad34d4d5554b2cacccc60&c=d5555554d52acccd4acad4b2b2cacab2cd4d5554b2ab34cc60&c=d5555554d52acccd4acad4b2b2cab54ccd4cd554cd54b4cc60&c=d5555554d52acccd4acad4b2b2cab5334d4cd554cd2b34cc60&c=d5555554d52acccd4acad4d4b2cab532cd4d5554cd4d34cc60&c=d5555554d52acccd4acad4d4b2cab52acd4cd554cd54b4cc60&c=d5555554d52acccd4acb2b3332cab5354d555554b2b4b4cc60&c=d5555554d52acccd4acb2b3332cab4d4cd555554b35534cc60&c=d5555554d52acccd4acb2b3332cab4cccd555554b34d34cc60&f=%3F%3F%3A%208h%20%3F%3F%3A%208h%20ID%3A%2016h%20STATUS%3A%204h%204d%20PRESS%3A%208d%20TEMP%3A%208d%20%3F%3F%3A%208h%20%3F%3F%3A%208h%20%3F%3F%3A%208h&a=Preamble&m=a9&d=MCI

This gives me my previous, already decoded codes, which I know shows correct results, plus 3 bytes that were obiously cut off before.

TTigges, 2020-01-28:
Byte 8 is CheckSum8 Xor for bytes 0 to 7.
Byte 9 is?

0f 54 76 e8 b7 53 44 01 64 4a

53 => dec: 83 x 1.4 (I'll try 1.37 but at the moment 1.4 comes closer to meassured values) = 116 kPa
44 => dec: 68 - 50 = 18°C (exactly the room temperature)
01 => could be status for rapid pressure changed. The tire that was untouched sent _"02"_. To be further tested.

merbanan, 2020-01-28:
byte 9 is a stray, ignore it

TTigges, 2020-01-29:
### Updated Google Sheets with the Sample Codes
https://docs.google.com/spreadsheets/d/1ZGbVTRssBKqKV_I3kSD5BuykL3l_xucRNVISEH8U45A/edit#gid=529440585

### Pressure modulator 1.4 vs 1.37
I compared the results of multiplying the pressure value by 1.37 and 1.4. On average, the results from 1.4 were closer to the results from manual testing, but some results from 1.37 were closest. I'll try to do further tests but for now I'm staying with 1.4.
@zuckschwerdt do you have any background information on the 1.37 you suggested?

### Device protocol
I'll work on that tonight. Tried some changes yesterday but they weren't taken over by rtl_433. There is no build process after changing a device protocol, or is there?
I'm expecting a bit of a struggle with making rtl_433 auto suggest my protocol instead of the Jansite it suggests now. I'll have to look at some other protocols to learn from them. From what you said, I take it, the checksum has something to do with it?

zuckschwerdt, 2020-01-29:
Usually the pressure readings are either PSI or kPa. So I expect to see something like:
- kPa with a resolution of e.g. 2 kPa to neatly fit 0-511 kPa in a byte
- PSI with a resolution of a small fraction (0.2, 0.25, 0.5 PSI) to neatly fit e.g. 0-439 kPa in a byte (255 x 0.25 x 6.89476).

So here I suggested it might be 0.2 PSI steps, i.e. 0.2 x 6.89476 = 1.378952. The max would be 351 kPa, does that fit the sensor description?

After changing some source code from the ./build directory run `cmake .. && make`

TTigges, 2020-01-29:
Makes sense.
If it's actually the VDO sensor the car is said to have, that would go to 450 kPa. Using that formula, the modifier would be ~1.76 but the sensor would show 2,5 bar when both KS Tools manometer and my Ryobi compressor showed 2 bar (+/- 0.5).

Ah, then I forgot the make process.

merbanan, 2020-01-29:
@TTigges We have a hack that lets some protocols call the crc check of other protocols. If the crc fits then we skip the current protocol. So use that or just disable the Jansite protocol.

TTigges, 2020-01-29:
Preamble and checksum are still a bit weird to me. I'm a frontend dev, I rarely code in C and working with frequencies is absolutely new to me.

Based on the tpms_jansite.c, I worked out this device protocol:
https://github.com/TTigges/rtl_433/blob/master/src/devices/tpms_abarth.c

Am I right in understanding that the preamble is the same as the one for Jansite? When I changed line 49 slightly `preamble_pattern[2] = {0xaa, 0xa9};` it would show me no results.

Next, I think I'll have to find out how to do the checksum8 xor in C.

zuckschwerdt, 2020-01-30:
Looks good already.

The preamble is just a bit pattern, it's commonly not byte-aligned, the alignment is done later.
The idea is to see: what ends the 10.. (aaa..., or 01... = 555... doesn't really matter) pattern, then use the minimum amount of bits to define a preamble. Here `…10101001`= a9.

For CRC we already have some helper functions. E.g. use `uint8_t c_crc = crc8(b, 8, 0x.., 0x..);`

zuckschwerdt, 2020-01-30:
Btw. it's not CRC but Parity. In your BitBench use:
```
ID: 32h STATUS: 4h 4d PRESS: 8d TEMP: 8d ?:8h CHK:8h ?:8h
```
and Calculation: XOR,  Offset: 0 Length: 72 Width: 8
e.g. 0x0f^0x54^0x76^0xe8^0xb7^0x55^0x44^0x01^0x62 == 0

TTigges, 2020-01-31:
I tried working on comparing the checksum with the byte 9.

Since I'm not at home, I tried it with the online compiler and debugger (https://onlinegdb.com/Hk3bd5WML):
```
Code:   0f547711b76e4401a1
Check:  a1
Cntrl:  a1
false
```
I was expecting `true` when comparing cntrl with Check (which I'm reading from the byte).

Any idea where my error is? (And yeah, these steps might be over complicated but I'm trying to figure this out.)

merbanan, 2020-01-31:
Looks like you are comparing string pointers and not the content.

zuckschwerdt, 2020-01-31:
The decoder should have
```
// check parity on 8+1 bytes
if (xor_bytes(b, 9)) return 0;
```
perhaps you can use that function to see what's going on.

TTigges, 2020-02-08:
I updated the device protocol, changed the comments. I left flags/status out of the report, for now. The relevant information are shown.

I considered changing the title to vdo_tg1c but since I cannot confirm that it's actually the built in sensor, I'd leave it at "abarth124" for now.

https://github.com/TTigges/rtl_433/blob/master/src/devices/tpms_abarth124.c

Would you consider this ready for a pull request or do you see anything that needs to be changed?

Anyway, before the pull request, I'll check the 1.37 vs 1.4 multiplicator for the pressure again.
(As stated before, 1.4 brings results that seem closer to my two tools which I use to measure the pressure at the moment.)

zuckschwerdt, 2020-02-08:
The 1.378952 isn't guaranteed, instead of 0.2 PSI steps I think it's more common to use 1/4 or 1/8 steps.
The `/*` are actually `/**` and the second block should not be indented.
The `static const unsigned char preamble_pattern` should be inside the deocder function as `uint8_t const *preamble_pattern`
The checksum should be `xor_bytes(b, 9)`

TTigges, 2020-02-09:
Updated the code. I think there were some updates regarding the preamble_pattern since I forked?

Didn't find any time for the pressure tests, today.

TTigges, 2020-02-09:
Ok, I ran ~ 30 comparisons today. Multiplicator 1.4 had a deviation range of 0.01 – 0.04 **above** what I took as base readings (handheld analog tool from Flaig, which showed the same as the Ryobi compressor and the Draper handheld analog tool). Multiplicator 1.37 had a deviation range of 0.01 – 0.025 below the base reading. Both were _"on the dot"_ sometimes, though.

And then I rounded your 1.378952 to 1.38 and the deviation range came to 0.008 – 0,019 with an average of 0.012. The deviation would still be 

So I'll take 1.38 as the multiplicator. Thank you for bringing that up.

TTigges, 2020-02-12:
> 
> 
> http://triq.net/bitbench?c=15555555956a99995a9a6a65669a9996995955559a596599c&c=15555555956a99995a9a6a65669a9999995955559a565599c&c=15555555956a99995a9a95a6a69a99aa59595555a5a6a599c&c=15555555956a99995a9a95a6a69a99aa99595555a5a66599c&f=%3F%3F%3A%208h%20%3F%3F%3A%208h%20ID%3A%208d%208d%20STATUS%3A%204h%204d%20PRESS%3A%208d%20TEMP%3A%208d%20CRC%3A%208h%20XX%3A%208h&a=Preamble&m=a9&d=MC

Maybe a dumb question, but how do you get _Align: a9?_
This is before Manchester Encoding?

When going through this manually (without bitbench) I don't get the same hex outcome.

zuckschwerdt, 2020-02-12:
a9 would be on the inverted buffer, here it is 56, or better 5556 -- aaa9 won't match, there is just a random a9 somewhere in there.
Generally you are looking for the bits that break/end the 010101... preamble pattern. A double 0 or double 1, so we know how to align the Manchester code.

merbanan, 2020-07-04:
Protocol decoder merged.

closed 2020-07-04 by merbanan