merbanan/rtl_433 #1272: Add more LaCrosse TX145wsdth samples

weberzach, 2020-01-27:
I've added 2 pulls requests [326](https://github.com/merbanan/rtl_433_tests/pull/326) and [325](https://github.com/merbanan/rtl_433_tests/pull/325) .

Can someone help me figure out the protocol?

I thought, given my technical aptitude, this would be a great weekend learning project, to grasp some basics with a new rtl sdr.  I got it for Christmas, and I'm still trying, a month later. I've had success with other devices (picked a tpms which was supported, and worked it out on graph paper following the basic concepts in Ben Eaters videos). 

Is this lacrosse a different beast, or am I just too dim?  I notice the FCC picture had a 4 pin jst connection, which probably isn't there on the production version, bit I'm tempted to add it on, just because I'm losing so hard on this one.

zuckschwerdt, 2020-01-27:
E.g. g128 is just a carrier.
g129 shows a very clean (repeated) signal the four overlong pulses are sync.
Try http://triq.net/iqs and `examples/sigrok-open.sh`.
Looks like a PWM signal with a third, overlong pulse as sync, the guessed flex sounds right:
`-X 'n=TX145wsdth,m=OOK_PWM,s=256,l=500,y=744,r=1872'`
Now table codes with the expected values and then we can look at decoding.

weberzach, 2020-01-27:
Well, I've been using audacity.  And wow, this is helpful.  I'm dinking bitbench to figure this out.
[data on bitbench](http://triq.net/bitbench?c=f73123fed20fd6458&c=f73123fdffff48a38&c=f73123fed1ffd6978&c=f73123fdffff478d8&c=f73123fdffff30e28&c=f73123fdffff19ec8&c=f73123fdfffefe4e8&c=f73123fed1efd6f98&c=f73123fdfffec3c70&c=f73123fdfffffcd88&c=f73123fed47fd6db8&c=f73123fed5dfd7538&c=f73123fed84fd6120&c=f73123fed9ffd5960&c=f73123fedb5fd5e58&c=f73123fedcafd51e8&c=f73123feddcfd50d8&c=f73123fedecfd6948&c=f73123fedfafd5d48&c=f73123fee07fd62f8&c=f73123fdfffef4958&c=f73123fee11fd56f8&c=f73123fdfffef4958&c=f731237eeb3fd59c8&c=f731237eeb3fd59c8&c=f731237eeb3fd59c8&c=f731237eeb3fd59c8&c=f731237dfffed6808&c=f731237dfffed6808&c=f731237dfffed6800&c=f731237edf9fd7f58&c=f731237edf9fd7f58&c=f731237edf9fd7f58&c=f731237edf9fd7f58&c=f731237dfffed6808&c=f731237dfffed6808&c=f731237dfffed6800&f=hh%20ID%3Ahhhh%20hh%20hh%20hh%20hh%20bbbb%20bbbb%20bbbb)
here's what I've charted so far:
`

  | C | F | S | D |   |   |   |   |   |   |   |   |   |   | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 | 0 |   | 0 | 0 |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
  | 23.50 | 74.3 | 0 | S | f73123fed20fd6458 | f7 | 31 | 23 | fe | d2 | 0f | d6 | 45 | 8 | 11110111 | 110001 | 100011 | 11111110 | 11010010 | 1111 | 11010110 | 1000101 | 1000 |   | 214 | 69 |  
  | -17.78 |   |   |   | f73123fdffff48a38 | f7 | 31 | 23 | fd | ff | ff | 48 | a3 | 8 | 11110111 | 110001 | 100011 | 11111101 | 11111111 | 11111111 | 1001000 | 10100011 | 1000 |   | 72 | 163 |  
  | -17.78 |   |   |   | f73123fed1ffd6978 | f7 | 31 | 23 | fe | d1 | ff | d6 | 97 | 8 | 11110111 | 110001 | 100011 | 11111110 | 11010001 | 11111111 | 11010110 | 10010111 | 1000 |   | 214 | 151 |  
  | 23.61 | 74.5 | 0 | s | f73123fdffff478d8 | f7 | 31 | 23 | fd | ff | ff | 47 | 8d | 8 | 11110111 | 110001 | 100011 | 11111101 | 11111111 | 11111111 | 1000111 | 10001101 | 1000 |   | 71 | 141 |  
  | 23.72 | 74.7 | 0 | ssw | f73123fdffff30e28 | f7 | 31 | 23 | fd | ff | ff | 30 | e2 | 8 | 11110111 | 110001 | 100011 | 11111101 | 11111111 | 11111111 | 110000 | 11100010 | 1000 |   | 48 | 226 |  
  | 23.72 | 74.7 | 0 | sw | f73123fdffff19ec8 | f7 | 31 | 23 | fd | ff | ff | 19 | ec | 8 | 11110111 | 110001 | 100011 | 11111101 | 11111111 | 11111111 | 11001 | 11101100 | 1000 |   | 25 | 236 |  
  | 23.72 | 74.7 | 0 | wsw | f73123fdfffefe4e8 | f7 | 31 | 23 | fd | ff | fe | fe | 4e | 8 | 11110111 | 110001 | 100011 | 11111101 | 11111111 | 11111110 | 11111110 | 1001110 | 1000 |   | 254 | 78 |  
  | 23.72 | 74.7 | 0 | w | f73123fed1efd6f98 | f7 | 31 | 23 | fe | d1 | ef | d6 | f9 | 8 | 11110111 | 110001 | 100011 | 11111110 | 11010001 | 11101111 | 11010110 | 11111001 | 1000 |   | 214 | 249 |  
  | 23.78 | 74.8 | 0 | nw | f73123fdfffec3c70 | f7 | 31 | 23 | fd | ff | fe | c3 | c7 | 0 | 11110111 | 110001 | 100011 | 11111101 | 11111111 | 11111110 | 11000011 | 11000111 | 0 |   | 195 | 199 |  
  | 23.89 | 75 | 0 | n | f73123fdfffffcd88 | f7 | 31 | 23 | fd | ff | ff | fc | d8 | 8 | 11110111 | 110001 | 100011 | 11111101 | 11111111 | 11111111 | 11111100 | 11011000 | 1000 |   | 252 | 216 |  
  | 19.61 | 67.3 | 0 | w | f73123fed47fd6db8 | f7 | 31 | 23 | fe | d4 | 7f | d6 | db | 8 | 11110111 | 110001 | 100011 | 11111110 | 11010100 | 1111111 | 11010110 | 11011011 | 1000 |   | 214 | 219 |  
  | 17.39 | 63.3 | 0 | w | f73123fed5dfd7538 | f7 | 31 | 23 | fe | d5 | df | d7 | 53 | 8 | 11110111 | 110001 | 100011 | 11111110 | 11010101 | 11011111 | 11010111 | 1010011 | 1000 |   | 215 | 83 |  
  | 13.50 | 56.3 | 0 | w | f73123fed84fd6120 | f7 | 31 | 23 | fe | d8 | 4f | d6 | 12 | 0 | 11110111 | 110001 | 100011 | 11111110 | 11011000 | 1001111 | 11010110 | 10010 | 0 |   | 214 | 18 |  
  | 10.78 | 51.4 | 0 | w | f73123fed9ffd5960 | f7 | 31 | 23 | fe | d9 | ff | d5 | 96 | 0 | 11110111 | 110001 | 100011 | 11111110 | 11011001 | 11111111 | 11010101 | 10010110 | 0 |   | 213 | 150 |  
  | 8.61 | 47.5 | 0 | w | f73123fedb5fd5e58 | f7 | 31 | 23 | fe | db | 5f | d5 | e5 | 8 | 11110111 | 110001 | 100011 | 11111110 | 11011011 | 1011111 | 11010101 | 11100101 | 1000 |   | 213 | 229 |  
  | 6.50 | 43.7 | 0 | w | f73123fedcafd51e8 | f7 | 31 | 23 | fe | dc | af | d5 | 1e | 8 | 11110111 | 110001 | 100011 | 11111110 | 11011100 | 10101111 | 11010101 | 11110 | 1000 |   | 213 | 30 |  
  | 4.72 | 40.5 | 0 | w | f73123feddcfd50d8 | f7 | 31 | 23 | fe | dd | cf | d5 | 0d | 8 | 11110111 | 110001 | 100011 | 11111110 | 11011101 | 11001111 | 11010101 | 1101 | 1000 |   | 213 | 13 | 21313
  | 3.11 | 37.6 | 0 | w | f73123fedecfd6948 | f7 | 31 | 23 | fe | de | cf | d6 | 94 | 8 | 11110111 | 110001 | 100011 | 11111110 | 11011110 | 11001111 | 11010110 | 10010100 | 1000 |   | 214 | 148 | 214148
  | 1.72 | 35.1 | 0 | w | f73123fedfafd5d48 | f7 | 31 | 23 | fe | df | af | d5 | d4 | 8 | 11110111 | 110001 | 100011 | 11111110 | 11011111 | 10101111 | 11010101 | 11010100 | 1000 |   | 213 | 212 | 213212
  | 0.39 | 32.7 | 0 | w | f73123fee07fd62f8 | f7 | 31 | 23 | fe | e0 | 7f | d6 | 2f | 8 | 11110111 | 110001 | 100011 | 11111110 | 11100000 | 1111111 | 11010110 | 101111 | 1000 |   | 214 | 47 | 21447
  | 0.39 | 32.7 | 0 | w | f73123fdfffef4958 | f7 | 31 | 23 | fd | ff | fe | f4 | 95 | 8 | 11110111 | 110001 | 100011 | 11111101 | 11111111 | 11111110 | 11110100 | 10010101 | 1000 |   | 244 | 149 | 244149
[... cut ...]
low batt | 1.78 | 35.2 | 0 | wnw | f731237dfffed6808 | f7 | 31 | 23 | 7d | ff | fe | d6 | 80 | 8 | 11110111 | 110001 | 100011 | 1111101 | 11111111 | 11111110 | 11010110 | 10000000 | 1000 |   | 214 | 128 | 214128
low batt | 1.78 | 35.2 | 0 | wnw | f731237dfffed6808 | f7 | 31 | 23 | 7d | ff | fe | d6 | 80 | 8 | 11110111 | 110001 | 100011 | 1111101 | 11111111 | 11111110 | 11010110 | 10000000 | 1000 |   | 214 | 128 | 214128
low batt | 1.78 | 35.2 | 0 | wnw | f731237dfffed6800 | f7 | 31 | 23 | 7d | ff | fe | d6 | 80 | 0 | 11110111 | 110001 | 100011 | 1111101 | 11111111 | 11111110 | 11010110 | 10000000 | 0 |   | 214 | 128 | 214128

`

zuckschwerdt, 2020-01-28:
Nice work! A battery low flag on the top bit of a byte (0x80) is very common, good.
There are too many `1`s compared to `0` (The data feels to "heavy"), it's likely inverted.
After inverting the temperature is easy to spot: 12 bits after the (battery-)flag byte, offset 500, scale 10.
Here is the [BitBench with some decoding](http://triq.net/bitbench?c=f73123fed20fd6458%5B23.50%20S%5D&c=f73123fdffff48a38%5B-17.78%5D&c=f73123fed1ffd6978%5B-17.78%5D&c=f73123fdffff478d8%5B23.61%20s%5D&c=f73123fdffff30e28%5B23.72%20ssw%5D&c=f73123fdffff19ec8%5B23.72%20sw%5D&c=f73123fdfffefe4e8%5B23.72%20wsw%5D&c=f73123fed1efd6f98%5B23.72%20w%5D&c=f73123fdfffec3c70%5B23.78%20nw%5D&c=f73123fdfffffcd88%5B23.89%20n%5D&c=f73123fed47fd6db8%5B19.61%20w%5D&c=f73123fed5dfd7538%5B17.39%20w%5D&c=f73123fed84fd6120%5B13.50%20w%5D&c=f73123fed9ffd5960%5B10.78%20w%5D&c=f73123fedb5fd5e58%5B8.61%20w%5D&c=f73123fedcafd51e8%5B6.50%20w%5D&c=f73123feddcfd50d8%5B4.72%20w%5D&c=f73123fedecfd6948%5B3.11%20w%5D&c=f73123fedfafd5d48%5B1.72%20w%5D&c=f73123fee07fd62f8%5B0.39%20w%5D&c=f73123fdfffef4958%5B0.39%20w%5D&c=f73123fee11fd56f8%5B-0.61%20w%5D&c=f73123fdfffef4958%5B-0.61%20w%5D&c=f731237eeb3fd59c8%5BB%20-16.78%20wnw%5D&c=f731237eeb3fd59c8%5BB%20-16.78%20wnw%5D&c=f731237eeb3fd59c8%5BB%20-16.78%20wnw%5D&c=f731237eeb3fd59c8%5BB%20-16.78%20wnw%5D&c=f731237dfffed6808%5BB%20-16.78%20wnw%5D&c=f731237dfffed6808%5BB%20-16.78%20wnw%5D&c=f731237dfffed6800%5BB%20-16.78%20wnw%5D&c=f731237edf9fd7f58%5BB%201.78%20wnw%5D&c=f731237edf9fd7f58%5BB%201.78%20wnw%5D&c=f731237edf9fd7f58%5BB%201.78%20wnw%5D&c=f731237edf9fd7f58%5BB%201.78%20wnw%5D&c=f731237dfffed6808%5BB%201.78%20wnw%5D&c=f731237dfffed6808%5BB%201.78%20wnw%5D&c=f731237dfffed6800%5BB%201.78%20wnw%5D&f=hh%20ID%3Ahhhh%20BATT%3A4b%20MSG%3Ah%20TEMP%3A12d%204h%208h%20CHK%3F8h%204h&i=true).
Note that there are two message types: 1 and 2 in the flag byte.

weberzach, 2020-01-29:
The two messages have 

ID BATT MESG_TYPE:1  TEMP RH%  CRC?
ID BATT MESG_TYPE:2  SPEED_KHM Heading_DEGREES  CRC?

I'm not 100% sure I'm going to figure out the CRC, but I'll try.  If it's a polynomial equation, is there a popular algorithm used for 433 data like this?  Is it safe to say a simple checksum would be used instead?

I made a sheet to help me understand the goodness.   

I used a freezer for a 0 temp, a boxfan for wind.  

Is there anything else?  What's the next step to incorporate it into the software? 

here's my proofs:
[image]

[BitBench](http://triq.net/bitbench?c=f73123fdffff30e28%20%5B%2023.72%20ssw%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdffff19ec8%20%5B%2023.72%20sw%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdfffefe4e8%20%5B%2023.72%20wsw%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed1efd6f98%20%5B%2023.72%20w%2073.7%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdfffec3c70%20%5B%2023.77%20nw%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdfffffcd88%20%5B%2023.88%20n%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed47fd6db8%20%5B%2019.61%20w%2069.6%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed5dfd7538%20%5B%2017.38%20w%2067.4%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed84fd6120%20%5B%2013.5%20w%2063.5%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed9ffd5960%20%5B%2010.77%20w%2060.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fedb5fd5e58%20%5B%208.61%20w%2058.6%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fedcafd51e8%20%5B%206.5%20w%2056.5%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123feddcfd50d8%20%5B%204.72%20w%2054.7%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fedecfd6948%20%5B%203.11%20w%2053.1%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fedfafd5d48%20%5B%201.72%20w%2051.7%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fee07fd62f8%20%5B%200.38%20w%2050.4%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdfffef4958%20%5B%200.38%20w%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fee11fd56f8%20%5B%20-0.61%20w%2049.4%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdfffef4958%20%5B%20-0.61%20w%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237eeb3fd59c8%20%5B%20-16.77%20wnw%2033.2%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237eeb3fd59c8%20%5B%20-16.77%20wnw%2033.2%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237eeb3fd59c8%20%5B%20-16.77%20wnw%2033.2%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237eeb3fd59c8%20%5B%20-16.77%20wnw%2033.2%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237dfffed6808%20%5B%20-16.77%20wnw%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237dfffed6808%20%5B%20-16.77%20wnw%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237dfffed6800%20%5B%20-16.77%20wnw%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237edf9fd7f58%20%5B%201.77%20wnw%2051.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237edf9fd7f58%20%5B%201.77%20wnw%2051.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237edf9fd7f58%20%5B%201.77%20wnw%2051.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237edf9fd7f58%20%5B%201.77%20wnw%2051.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237dfffed6808%20%5B%201.77%20wnw%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237dfffed6808%20%5B%201.77%20wnw%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f731237dfffed6800%20%5B%201.77%20wnw%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed24fd4ae0%20%5B%2023.1%20s%2073.1%20deg_DIR%20171%25%200%20KMH%20%5D&c=f73123fed27fd52d8%20%5B%2022.8%20n%2072.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed27fd52d8%20%5B%2022.8%20n%2072.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed27fd52d0%20%5B%2022.8%20n%2072.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed27fd52d8%20%5B%2022.8%20n%2072.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdffff549d0%20%5B%2022.8%20n%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdffff549d8%20%5B%2022.8%20n%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdffff549d8%20%5B%2022.8%20n%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed27fd52d8%20%5B%2022.8%20S%2072.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed27fd52d8%20%5B%2022.8%20S%2072.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed27fd52d0%20%5B%2022.8%20S%2072.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed27fd52d8%20%5B%2022.8%20S%2072.8%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdffff549d8%20%5B%2022.8%20S%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdffff549d8%20%5B%2022.8%20S%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fdffff549d0%20%5B%2022.8%20S%200%20deg_DIR%20%25%200%20KMH%20%5D&c=f73123fed2bfd5878%20%5B%2022.4%20N%2072.4%20deg_DIR%200.42%25%200%20KMH%20%5D&c=f73123fed2bfd5878%20%5B%2022.4%20N%2072.4%20deg_DIR%200.42%25%200%20KMH%20%5D&c=f73123fed2bfd5870%20%5B%2022.4%20N%2072.4%20deg_DIR%200.42%25%200%20KMH%20%5D&c=f73123fed2bfd5878%20%5B%2022.4%20N%2072.4%20deg_DIR%200.42%25%200%20KMH%20%5D&c=f73123fed2bfd5878%20%5B%2022.4%20N%2072.4%20deg_DIR%200.42%25%200%20KMH%20%5D&c=f73123fed2bfd5878%20%5B%2022.4%20N%2072.4%20deg_DIR%200.42%25%200%20KMH%20%5D&c=f73123fed2bfd5870%20%5B%2022.4%20N%2072.4%20deg_DIR%200.42%25%200%20KMH%20%5D&c=f73123fed2bfd5878%20%5B%2022.4%20N%2072.4%20deg_DIR%200.42%25%200%20KMH%20%5D&c=f73123fdfffe9da68%20%5B%2022.4%20N%200%20deg_DIR%200.42%25%200%20KMH%20%5D&c=f73123fdfffe9da68%20%5B%2022.4%20N%200%20deg_DIR%200.42%25%200%20KMH%20%5D&c=f73123fdfffe9da60%20%5B%2022.4%20N%200%20deg_DIR%200.42%25%200%20KMH%20%5D&c=f73123fed29fd55b8%20%5B%2022.8%20N%2072.6%20deg_DIR%200.43%25%200%20KMH%20%5D&c=f73123fed29fd55b8%20%5B%2022.8%20N%2072.6%20deg_DIR%200.43%25%200%20KMH%20%5D&c=f73123fed29fd55b0%20%5B%2022.8%20N%2072.6%20deg_DIR%200.43%25%200%20KMH%20%5D&c=f73123fed29fd55b8%20%5B%2022.8%20N%2072.6%20deg_DIR%200.43%25%200%20KMH%20%5D&c=f73123fdfffe9da68%20%5B%2022.8%20N%200%20deg_DIR%200.43%25%200%20KMH% [... cut ...]
  | 21.60 | 70.88 | 10 | 250 | 42 | f73123fdf9df05ab8 | f | 7 | 3 | 1 | 2 | 3 | f | d | f | 9 | d | f | 0 | 5 | a | b | 8 | 0000 | 1000 | 1100 | 1110 | 1101 | 1100 | 0000 | 0010 | 0000 | 0110 | 0010 | 0000 | 1111 | 1010 | 0101 | 0100 | 0111 | 062 | -40.2 | 0FA | 250 | 9.8 | f73123fdf9df05ab8 [ 21.6 250 9.8 deg_DIR 42% 10 KMH ]
  | 21.60 | 70.88 | 10 | 250 | 42 | f73123fdf9df05ab0 | f | 7 | 3 | 1 | 2 | 3 | f | d | f | 9 | d | f | 0 | 5 | a | b | 0 | 0000 | 1000 | 1100 | 1110 | 1101 | 1100 | 0000 | 0010 | 0000 | 0110 | 0010 | 0000 | 1111 | 1010 | 0101 | 0100 | 1111 | 062 | -40.2 | 0FA | 250 | 9.8 | f73123fdf9df05ab0 [ 21.6 250 9.8 deg_DIR 42% 10 KMH ]

zuckschwerdt, 2020-01-29:
I'll take a stab at the CRC, then I can add the decoder. Some testing needed afterwards and then we are done.

zuckschwerdt, 2020-01-29:
Checksum is the well-known CRC-8, poly 0x31, init 0x00 :)

weberzach, 2020-01-29:
Thanks, that was fast.  What do you use for calculating your polynomial?  I was trying to figure how to make a binary algebraic field in a spreadsheet for 25 minutes, you had it immediately.

weberzach, 2020-01-29:
Nevermind, I was using the raw instead of inverse. Learnding is hard.  

Thanks for all your help.

zuckschwerdt, 2020-01-29:
…but ultimatly rewarding! :)

I recommend RevEng, bruteforce-crc is also good. But I used [revdgst](https://github.com/triq-org/revdgst/) which tries other types of sums too.

zuckschwerdt, 2020-01-29:
Done! Support added with ac4cc4f. Thanks for analyzing this!

closed 2020-01-29 by zuckschwerdt

merbanan, 2020-01-29:
@zuckschwerdt mic is missing in the json output. Can you add it?

zuckschwerdt, 2020-01-29:
oops. done.

weberzach, 2020-01-30:
Thanks a ton for the commit.

Will you take a peek at this:
`    if (bitbuffer->bits_per_row[r] >= 64) {
        device = LACROSSE_TX141W;
    }
    else if (bitbuffer->bits_per_row[r] > 40) {
        return DECODE_ABORT_LENGTH;
    }
    else if (bitbuffer->bits_per_row[r] >= 40) {
        device = LACROSSE_TX141TH;
    }
    else if (bitbuffer->bits_per_row[r] >= 37) {
        device = LACROSSE_TX141;
    }
    else {
        device = LACROSSE_TX141BV3;
    }
 `
Im thinking the greater than or equal 40 will always win over the 64. So the type "2" message will never be true.

weberzach, 2020-01-30:
Finally home to check this out.  My tx145wsdth seems to come in as a LaCrosse-TX141W, so the wind-speed direction don't make the cut.   I'll peek around and see if I can find the error.    

Here's the output with 
`
acurite_txr_decode: bitbuffer:: Number of rows: 7 
[00] {65} 08 ce dc 01 2d 10 2c 39 00 
[01] {65} 08 ce dc 01 2d 10 2c 39 00 
[02] {65} 08 ce dc 01 2d 10 2c 39 80 
[03] {65} 08 ce dc 01 2d 10 2c 39 00 
[04] {65} 08 ce dc 02 00 01 34 81 00 
[05] {65} 08 ce dc 02 00 01 34 81 00 
[06] {65} 08 ce dc 02 00 01 34 81 80 
acurite_txr_decode: row 0 bits 65, bytes 9 
acurite_txr_decode: bad checksum: {65} 08 ce dc 01 2d 10 2c 39 00 
acurite_txr_decode: row 1 bits 65, bytes 9 
acurite_txr_decode: bad checksum: {65} 08 ce dc 01 2d 10 2c 39 00 
acurite_txr_decode: row 2 bits 65, bytes 9 
acurite_txr_decode: bad checksum: {65} 08 ce dc 01 2d 10 2c 39 80 
acurite_txr_decode: row 3 bits 65, bytes 9 
acurite_txr_decode: bad checksum: {65} 08 ce dc 01 2d 10 2c 39 00 
acurite_txr_decode: row 4 bits 65, bytes 9 
acurite_txr_decode: bad checksum: {65} 08 ce dc 02 00 01 34 81 00 
acurite_txr_decode: row 5 bits 65, bytes 9 
acurite_txr_decode: bad checksum: {65} 08 ce dc 02 00 01 34 81 00 
acurite_txr_decode: row 6 bits 65, bytes 9 
acurite_txr_decode: bad checksum: {65} 08 ce dc 02 00 01 34 81 80 
acurite_986_decode: row 0 bits 275, bytes 5 
acurite_986_decode: skipping wrong len
acurite_986_decode: row 1 bits 206, bytes 5 
acurite_986_decode: skipping wrong len
radiohead_ask_extract: preamble not found
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-01-29 22:23:11
model     : LaCrosse-TX141W                        Sensor ID : 0cedc
Channel   : 0            Battery level: 1          Temperature: 22.10 C      Humidity  : 44 %          Test?     : 0             Integrity : CRC
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor 
bitbuffer:: Number of rows: 5 
[00] { 1} 80                         : 1
[01] {65} 08 ce dc 01 2d 10 2c 39 00 
[02] {65} 08 ce dc 01 2d 10 2c 39 00 
[03] {65} 08 ce dc 01 2d 10 2c 39 80 
[04] {65} 08 ce dc 01 2d 10 2c 39 00 
pulse_demod_pwm(): LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor 
bitbuffer:: Number of rows: 3 
[00] {65} f7 31 23 fd ff fe cb 7e 80 
[01] {65} f7 31 23 fd ff fe cb 7e 80 
[02] {65} f7 31 23 fd ff fe cb 7e 00 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 2 
[00] {65} 08 ce dc 01 2d 10 2c 39 00 
[01] { 0}                            : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 2 
[00] {65} 08 ce dc 01 2d 10 2c 39 00 
[01] { 0}                            : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 2 
[00] {65} 08 ce dc 01 2d 10 2c 39 80 
[01] { 0}                            : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] {65} 08 ce dc 01 2d 10 2c 39 00 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 2 
[00] {65} 08 ce dc 02 00 01 34 81 00 
[01] { 0}                            : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 2 
[00] {65} 08 ce dc 02 00 01 34 81 00 
[01] { 0}                            : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] { 0}           : 
acurite_00275rm_decode: bitbuffer:: Number of rows: 1 
[00] {65} 08 ce dc 02 00 01 34 81 80 
philips_aj3650_decode: wrong number of bits (483)
Interlogix: Preamble not found, bit_offset: 64
Interlogix: Preamble not found, bit_offset: 64
Interlogix: Preamble not found, bit_offset: 64
Interlogix: Preamble not found, bit_offset: 64
Interlogix: Preamble not found, bit_offset: 64
Interlogix: Preamble not found, bit_offset: 64
radiohead_ask_extract: preamble not found
AmbientWeather-TX8300: wrong size (482 bits)
philips_aj7010_decode: wrong number of bits (0)
gt_tmbbq05_decode: Possible Quigg BBQ: bitbuffer:: Number of rows: 1 

`

zuckschwerdt, 2020-01-30:
Yes, the TX141W and TX145wsdth are nearly the same. Both report the same data also.
In my tests the type-2 messages work? Try
```
rtl_433 -y 'f73123fdf45f06e6/f73123fdf45f06e6/f73123fdf45f06e6/f73123fdf45f06e6'
```
You did list the type-2 messages above, but we don't have a sample file for those. Perhaps you can record 1 or 2 samples with wind?

weberzach, 2020-01-30:
> Yes, the TX141W and TX145wsdth are nearly the same. Both report the same data also.
> In my tests the type-2 messages work? Try
>
> ```
> rtl_433 -y [...]

[image]

I'll pull some with higher wind speeds.  Thank you so much for looking at that.  I really appreciate your time.

zuckschwerdt, 2020-01-30:
Oh, so it works :) If you can grab one with `-S known` that would be great.

weberzach, 2020-01-30:
[radiopulls.zip](https://github.com/merbanan/rtl_433/files/4135485/radiopulls.zip)
I'm note quite sure how to find the codes I was getting before.  
I thought I was just running: rtl_433 -X 'n=TX145wsdth,m=OOK_PWM,s=256,l=500,y=744,r=1872'
and pasting the codes into my spreadsheet.  Perhaps I've lost my luck.

Anyway, I did more pulls today and attached them.  Just like all the previous, they should have wind and wind heading embedded in them, but I can't remember how I got those code.

zuckschwerdt, 2020-01-30:
Ok, thanks. Found it and it's fixed. You should get the wind messages now.

weberzach, 2020-01-30:
[image]

Over VPN, this looks excellent! Thank you for implementation.  I can't see the actual measurements on the display, but it was roughly 10.5 km/h when I put a fan in front of it this morning.

Thank you all so much. If I can help out in anyway, I will. Your donated time is greatly appreciated!

commit ac4cc4f4 (2020-01-29): Add LaCrosse TX145wsdth support (closes #1272)