merbanan/rtl_433 #1275: Requesting support for Ecowitt WH40 or WH5360B rain gauge sensor

Helmut63, 2020-01-27:
I´d appreciate if this sensor could be added to the list. Seems to be the same as Fine Offset WH5360 or Ecowitt WH5360B.
http://www.ecowitt.com/Rain_Gauge/10.html
I ahve already uploaded test files as https://github.com/merbanan/rtl_433_tests/tree/master/tests/EcoWitt-WH40
Some file contain some manual bucket tips to indicate rain.

Please let me know if you need any additional information.
Thanks in advance!

zuckschwerdt, 2020-01-27:
Samples merged with https://github.com/merbanan/rtl_433_tests/pull/327

The data is FSK PCM, roughly matching the Ambient Weather WH31 (`2dd4` sync word).

This works to get the raw data:
```
rtl_433 -X 'n=EcoWitt-WH40,m=FSK_PCM,s=58,l=58,r=2000,preamble=aaaa2dd4'
```

Data in the samples is:
```
g003_433.92M_250k.cu8
{115}4000cd6f10000064f00001de00b00
{116}4000cd6f10000064f00002de00b01

g022_433.92M_250k.cu8
{116}4000cd6f1000296a1f0002f600000

g026_433.92M_250k.cu8
{115}4000cd6f10000064f00002bd00000

g037_433.92M_250k.cu8
{116}4000cd6f10000064f000027b00000
{115}4000cd6f10000064f000027b00000
```
[BitBench](http://triq.net/bitbench?c=%7B115%7D4000cd6f10000064f00001de00b00%5Bg003%5D&c=%7B116%7D4000cd6f10000064f00002de00b01%5Bg003%5D&c=%7B116%7D4000cd6f1000296a1f0002f600000%5Bg022%5D&c=%7B115%7D4000cd6f10000064f00002bd00000%5Bg026%5D&c=%7B116%7D4000cd6f10000064f000027b00000%5Bg037%5D&c=%7B115%7D4000cd6f10000064f000027b00000%5Bg037%5D&f=8h%20) (without preamble and sync)

zuckschwerdt, 2020-01-27:
Can you record the codes received with the command above and between each receive tip the bucket once? That way we can see where in the code the counter is.

Also when you reset the battery the ID should change and we can spot that in the codes as well.

Helmut63, 2020-01-27:
[WH40-samples.zip](https://github.com/merbanan/rtl_433/files/4119111/WH40-samples.zip)

Just did some tests with the requested command! Attached is the output I got on the screen. tried to tip the bucket between every message, but might have missed some as other sensors also sent their reports which I did cut off from this file.
As I reset the battery, this message appeared afterwards: "pulse_FSK_detect(): Maximum number of pulses reached!"

Hope that helps! Otherwise let me know.

zuckschwerdt, 2020-01-27:
Yes, the codes are in the output posted:
```
{116} 4000 cd6f 1000 0155 e200 027b 00000
{116} 4000 cd6f 1000 0155 e200 027b 00000
{115} 4000 cd6f 1000 0206 9400 02f6 00000
{116} 4000 cd6f 1000 0064 f000 01de 00b00
{116} 4000 cd6f 1000 0064 f000 02de 00b00
{116} 4000 cd6f 1000 0206 9400 027b 00000
{116} 4000 cd6f 1000 0206 9400 027b 00000
```
I expected a checksum but don't quite see it. Also I don't really see a counter (number of bucket tips) going up.

This command will skip all other decoders and require `4000` in the code -- that should only show WH40 then:
```
rtl_433 -R 0 -X 'n=EcoWitt-WH40,m=FSK_PCM,s=58,l=58,r=2000,preamble=aaaa2dd4,match=4000'
```
Maybe you can run this for longer so we get perhaps 20-40 different codes. Just post the list of codes like above.

Helmut63, 2020-01-28:
Here is what I got applying above command and filtering out only the code line:
codes     :  {91} 30d0 8288 2e79 b100 0000 000
codes     :  {91} 30d0 8288 2e79 b100 0000 000
codes     : {116} 4000 cd6f 1000 0064 f000 027b 00000
codes     : {116} 4000 cd6f 1000 0064 f000 027b 00000
codes     : {116} 4000 cd6f 1000 0337 c600 02f6 00000
codes     : {116} 4000 cd6f 1000 04a0 3000 02ed 00001
codes     : {116} 4000 cd6f 1000 04a0 3000 02ed 00000
codes     : {116} 4000 cd6f 1000 06c2 5400 02db 00000
codes     :  {90} 30d0 8286 2e14 4a00 0000 000
codes     :  {90} 30d0 8286 2e14 4a00 0000 000
codes     : {116} 4000 cd6f 1000 06c2 5400 026f 00001
codes     : {115} 2431 c662 2028 0000 01d7 0000 00000
codes     :  {90} 30d0 8286 2e14 4a00 0000 000
codes     :  {90} 30d0 8286 2e14 4a00 0000 000
codes     : {116} 4000 cd6f 1000 06c2 5400 02de 00001
codes     : {115} 4000 cd6f 1000 06c2 5400 02de 00000
codes     : {115} 2431 d162 2028 0000 01d7 0000 00000
codes     : {115} 2431 ba62 2028 0000 01d7 0000 00000
codes     : {115} 2431 ba62 2028 0000 01d7 0000 00000
codes     : {116} 4000 cd6f 1000 06c2 5400 02bd 00000
codes     :  {90} 30d0 8286 2e14 4a00 0000 000
codes     :  {90} 30d0 8286 2e14 4a00 0000 000
codes     : {115} 4000 cd6f 1000 09ec 8100 027b 00000
codes     : {115} 4000 cd6f 1000 09ec 8100 027b 00000
codes     : {115} 2431 bf62 2028 0000 01d7 0000 00000
codes     :  {90} 30d0 8286 2e14 4a00 0000 000
codes     :  {90} 30d0 8286 2e14 4a00 0000 000
codes     : {115} 2431 c262 2028 0000 01d7 0000 00000
codes     : {115} 2431 b462 2028 0000 01d7 0000 00000
codes     : {116} 4000 cd6f 1000 09ec 8100 02f6 00000
codes     : {115} 2431 ba62 2028 0000 01d7 0000 00000
codes     : {116} 4000 cd6f 1000 0c19 b100 02ed 00000
codes     : {116} 4000 cd6f 1000 0c19 b100 02ed 00000
codes     : {115} 2431 bb62 2028 0000 01d7 0000 00000
codes     : {116} 4000 cd6f 1000 0c19 b100 02db 00001
codes     : {115} 2431 c162 2028 0000 01d7 0000 00000
codes     : {115} 2431 ca62 2028 0000 01d7 0000 00000

Hope this helps, otherwise I´ll be glad to make more tests.

Thanks and regards,
Helmut

zuckschwerdt, 2020-01-28:
The codes starting `30` would be the WH31, etc. The codes starting `40` are the WH40 ;)
This is a shortened list. In the groups Tolip suspected the last column to be the bucket counter. But it goes up and down -- not what I expect.
```
4000 cd6f 1000 0064 f000 027b
4000 cd6f 1000 0064 f000 027b
4000 cd6f 1000 0337 c600 02f6
4000 cd6f 1000 04a0 3000 02ed
4000 cd6f 1000 04a0 3000 02ed
4000 cd6f 1000 06c2 5400 02db
4000 cd6f 1000 06c2 5400 026f
4000 cd6f 1000 06c2 5400 02de
4000 cd6f 1000 06c2 5400 02de
4000 cd6f 1000 06c2 5400 02bd
4000 cd6f 1000 09ec 8100 027b
4000 cd6f 1000 09ec 8100 027b
4000 cd6f 1000 09ec 8100 02f6
4000 cd6f 1000 0c19 b100 02ed
4000 cd6f 1000 0c19 b100 02ed
4000 cd6f 1000 0c19 b100 02db
```

Helmut63, 2020-01-28:
Hi, here is another attempt, adding "|grep 4000" to the end of the command line, and 1 manual bucket tip in between every message:
{116} 4000 cd6f 1000 0064 f000 027b 00000
{116} 4000 cd6f 1000 0064 f000 027b 00000
{115} 4000 cd6f 1000 0155 e200 02f6 00000
{116} 4000 cd6f 1000 0206 9400 02ed 00001
{116} 4000 cd6f 1000 0206 9400 02ed 00000
{116} 4000 cd6f 1000 0337 c600 02db 00001
{115} 4000 cd6f 1000 04a0 3000 02b7 00000
{116} 4000 cd6f 1000 04a0 3000 02b7 00000
{116} 4000 cd6f 1000 0591 2200 02de 00000
{115} 4000 cd6f 1000 0591 2200 02de 00000
{116} 4000 cd6f 0000 6c25 4000 2bd0 0000
{116} 4000 cd6f 1000 07f3 8600 027b 00000
{116} 4000 cd6f 1000 07f3 8600 027b 00000
{116} 4000 cd6f 1000 08dd 7100 02f6 00001
{115} 4000 cd6f 1000 09ec 8100 02ed 00000
{116} 4000 cd6f 1000 09ec 8100 02ed 00000
{116} 4000 cd6f 1000 0abf 5500 02db 00000

Please let me know if this makes more sense... also took another reading for abut half an hour with no bucket tips at all. I can share those as well upon your request.

BR,
Helmut

zuckschwerdt, 2020-01-28:
Great! That's a better picture. The 4th group seems to be the counter. E.g. the `04` here:
` {115} 4000 cd6f 1000 `**`04`**`a0 3000 02b7 00000`.
(I've reformatted your post above)
It goes from 0 to 10 (a). The one or two byte following would be a checksum and the rest unknown/garbage.
I'll prototype a decoder soon.

zuckschwerdt, 2020-01-28:
Yes, checksums work. It's a CRC-8 and SUM-8.
```
crc8(&b[0], 7, 0x31, 0x00) ^ b[7] == 0; // (100.0%)
add_bytes(&b[0], 8) - b[8] == 0x00; // (100.0%)
```

Helmut63, 2020-01-28:
Great! That´s good news indeed!!! Thks

zuckschwerdt, 2020-01-31:
Support has been added with 12c2298 now. I would be great if you can find out how much rain a single bucket tip accounts for. Values might be 0.254mm, .2mm, ...

closed 2020-01-31 by zuckschwerdt

Helmut63, 2020-02-03:
Already sent an email to Ecowitt Support asking for this information...
Meanwhile, I run some more tests with following outputs:
$sudo rtl_433 -M utc
time      : 2020-02-03 20:30:21
model     : EcoWitt-WH40 id        : 52591
Channel   : 2            Battery   : 0             Rain count: 4
Extra Data: 0002bd0000   Integrity : CRC

$sudo rtl_433 -M utc -F json
{"time" : "2020-02-03 20:32:48", "model" : "EcoWitt-WH40", "id" : 52591, "channel" : 2, "battery_ok" : 0, "rain_raw" : 4, "data" : "0002ed0000", "mic" : "CRC"}

Fields of "Rain count:4" and "rain_raw" : 4 increment with every manual tip of rain bucket.

weewx in debug mode with same command in sdr section reports something similar:
Feb  3 17:13:16 raspberrypi weewx[3069]: sdr: MainThread: parse_json: unknown model EcoWitt-WH40
Feb  3 17:13:16 raspberrypi weewx[3069]: sdr: MainThread: punt unrecognized line '{"time" : "2020-02-03 20:13:12", "model" : "EcoWitt-WH40", "id" : 52591, "channel" : 2, "battery_ok" : 0, "rain_raw" : 3, "data" : "0002ed0000", "mic" : "CRC"}#012'

I'll get back once I get response from Ecowitt on the mm for each bucket tip.

What are the next steps? Is there anything to be done within weewx? I'm running 3.9.2, will there be an updated version with WH40 included? Anything we can do as we wait for the response from Ecowitt?
Thks and best regards!

zuckschwerdt, 2020-02-03:
Can you try a fresh battery and also a known flat battery (or maybe a variable power supply dialed down to just above the cut off). The battery_ok should change, it may be inverted?

Are there channel switches? Can you try different settings and see if it matches?

zuckschwerdt, 2020-02-03:
If there is no feedback from EcoWitt, you could measure the inlet area and slowly pour water in. Amout and area should give the mm value (use 10 tips or so if 1 tip is too little to measure the amount).

Helmut63, 2020-02-04:
Replaying from your comment regarding a battery flag from above:
I've been testing with some low voltage batteries, and found that all batteries still report a "battery OK" from driver output.
So i decided extracting some samples like before with different battery voltages. please have look at previous samples taken with a 1.5V alkaline battery, and compare to samples below, taken with 1.2V and 0.9V alkaline batteries
$sudo rtl_433 -R 0 -X 'n=EcoWitt-WH40,m=FSK_PCM,s=58,l=58,r=2000,preamble=aaaa2dd4,match=4000'
Looking for "battery" bit:
Samples with 1.2V battery (last 2 samples contain 1 manual bucket tip)

codes     : {116} 4000 cd6f 1000 0064 f000 01de 00b00
codes     : {116} 4000 cd6f 1000 0064 f000 02de 00b00
codes     : {116} 4000 cd6f 1000 0064 f000 02bd 00001
codes     : {116} 4000 cd6f 1000 0155 e200 027b 00000
codes     : {115} 4000 cd6f 1000 0155 e200 027b 00000

Samples with 0.9V battery (last 3 samples contain 1 manual bucket tip)

codes     : {115} 4000 cd6f 1000 0064 f000 16de 00000
codes     : {115} 4000 cd6f 1000 0064 f000 02de 00000
codes     : {115} 4000 cd6f 1000 0155 e200 02bd 00000
codes     : {116} 4000 cd6f 1000 0155 e200 027b 00000
codes     : {115} 4000 cd6f 1000 0155 e200 027b 00000
 
Next steps in my testing  will be regarding the mm value...

I´d appreciate you comments! 

Thanks and regards!
Helmut

zuckschwerdt, 2020-02-04:
I don't spot anything like a battery good/bad flag :/
```
Samples with 1.2V battery (last 2 samples contain 1 manual bucket tip)
    TYPE  ID Flag? Tips Checks ; Unknown
    4000 cd6f 10 0000  64 f0 ; 00 01de 00b0
    4000 cd6f 10 0000  64 f0 ; 00 02de 00b0
    4000 cd6f 10 0000  64 f0 ; 00 02bd 0000
    4000 cd6f 10 0001  55 e2 ; 00 027b 0000
    4000 cd6f 10 0001  55 e2 ; 00 027b 0000
Samples with 0.9V battery (last 3 samples contain 1 manual bucket tip)
    4000 cd6f 10 0000  64 f0 ; 00 16de 0000
    4000 cd6f 10 0000  64 f0 ; 00 02de 0000
    4000 cd6f 10 0001  55 e2 ; 00 02bd 0000
    4000 cd6f 10 0001  55 e2 ; 00 027b 0000
    4000 cd6f 10 0001  55 e2 ; 00 027b 0000
```

Helmut63, 2020-02-04:
I understand, it doesn´t make much sense to me either... any other test you might suggest running for the battery status?

Meanwhile (still waiting for an official response from EcoWitt), I played a bit with the tip counter vs mm value and I came up that 1 tip represents 0.1mm, or 10 tips=1mm of rain.
Here´s the math: inner diameter of rain gauge = 17.85cm, so the cylinder base area would be (17.85/2)^2*3.14159=250cm^2, and if I take 1mm (.1cm) height it equals to 250*.1=25cm^3 or 25ml. Pouring 25ml water into the rain gauge gives me exactly 10 bucket tips. (I actually poured 50ml and obtained 20 tips). 
Hope that helps to continue with the conversion and adding WH40 into Weewx sensor map. 

I'd appreciate your comments regarding next steps, or what needs to be done to have the sensor added to weewx, overcoming the odd messages of "unrecongnized line" and "unknown model EcoWitt-WH40" as seen on debug messages of weewx.

Thanks again and best regards!

zuckschwerdt, 2020-02-04:
.1mm per tip sounds right, and ~6000mm in .1 increments neatly fits the 16 bit range (65535). I'll add that.
Strange about that leading `10` -- it should be some kind of flags, battery and channel most likely.

I'm not a weewx user, can't help with that. But there are some users around. Maybe someone other can advice on this.

Helmut63, 2020-02-05:
Just adding official response received from EcoWitt, confirming our conclusion: "A single bucket tip accounts for 0.1mm."

zuckschwerdt, 2020-02-05:
Great! I've now updated the decoding for `rain_mm`.

commit 12c2298d (2020-01-31): Add EcoWitt WH40 support (closes #1275)