merbanan/rtl_433 #1276: Default auto FSK detector not working for Honeywell Doorbell

peteakalad, 2020-01-27:
After updating rtl_433 it wasn't decoding my doorbell using this;
`rtl_433 -d 1 -F mqtt://localhost:1883 -M utc -f 868221000 -R 116`

After some diagnosing seems I need to change the FSK detector to classic.
`rtl_433 -d 1 -F mqtt://localhost:1883 -M utc -f 868221000 -R 116 -Y classic`

Had a look at the code and it may be worth mentioning in docs / --help that the new FSK detector will be used over FSK_PULSE_DETECTOR_LIMIT (800Mhz currently).

Not sure about the purpose of the new detector, but its broken for my device/uses.

Sample, Honeywell ActivLink, Wireless Doorbell (FSK) 
[g005_868.221M_1000k.cu8.zip](https://github.com/merbanan/rtl_433/files/4118953/g005_868.221M_1000k.cu8.zip)

merbanan, 2020-01-27:
@peteakalad the new fsk detector is needed for some other sensors. Ideally we will change the code so both can be run at the same time. But right now there has not been enough time to fix that.

peteakalad, 2020-01-28:
Are there lots of sensors > 800 Mhz not decoding with the classic FSK decoder?

I ask as I can see there numerous other issues raised with people stumbling into this same issue with -f over 800Mhz and having to do -Y classic to get it working again.

If the auto threshold of 800Mhz was decided as more over 800Mhz sensors are broken without the new decoder, fair enough leave as-is.

If its the other way round or we dont know I'd suggest a blanket default of -Y classic until either both decoders can run at the same time or perhaps devices can be tied to the decoder type?  Then the minority of devices which do need the new FSK detector can explicity set -Y minmax.

merbanan, 2020-01-28:
I don't really know the answer to these question. And there is little point in figuring it out. It is better to focus on enabling both demodulators at the same time or making the minmax version handle more signals.

merbanan, 2020-01-28:
Hopefully I will have the time to fix this in the next 2 weeks.

peteakalad, 2020-01-28:
Ok sure understand.

Thanks for the great project btw.

evilpete, 2020-02-05:
I've seen problems with the FSK decoding also, mostly false translations resulting in pulses having drops lasting only one or two samples ( 1ns ?).   (no data Whitening ? )

@merbanan I'll be happy to talk about this offline as I a still trying to verity if the cause is the demodulation or the CC1111 based transmitter before I open a issue/bug report.

merbanan, 2020-02-17:
Command line output has been added to inform users how they can enable the old defaults. It is planned to run both fsk demodulators at the same time so that the user doesn't need to do anything. Closing issue as it is a known limitation.

closed 2020-02-17 by merbanan