merbanan/rtl_433 #1277: option -C si gives inches instead of millimeters

dgrostoto, 2020-01-28:
my try with a PCR800
./rtl_433 -F csv -C si
outputs
time,msg,codes,model,button,id,channel,battery,temperature_C,mic,subtype,rid,humidity,state,status,brand,rain_rate,rain_rate_mm_h,rain_total,rain_mm,gust,average,direction,wind_max_m_s,wind_avg_m_s,wind_dir_deg,pressure_hPa,uv,power_W,energy_kWh,radio_clock,sequence,unit,group_call,command,dim,dim_value,wind_speed,wind_gust,wind_direction,wind_avg_km_h,wind_max_km_h,dipswitch,rbutton,device,battery_ok,setpoint_C,switch,cmd,cmd_id,tristate,direction_deg,speed,rain,msg_type,signal,sensor_code,uv_status,uv_index,lux,wm,seq,rainfall_mm,wind_speed_ms,gust_speed_ms,current,interval,learn,message_type,sensor_id,sequence_num,battery_low,wind_speed_mph,wind_speed_kph,rain_inch,rc,gust_speed_mph,wind_approach,flags,maybetemp,binding_countdown,depth,depth_cm,dev_id,sensor_type,power0,power1,power2,node,ct1,ct2,ct3,ct4,Vrms/batt,batt_Vrms,temp1_C,temp2_C,temp3_C,temp4_C,temp5_C,temp6_C,pulse,address,button1,button2,button3,button4,data,sid,group,transmit,moisture,type,pressure_kPa,battery_mV,pulses,pulsecount,energy,len,to,from,payload,event,alarm,tamper,heartbeat,temperature1_C,temperature2_C,temperature_1_C,temperature_2_C,test,probe,water,humidity_1,ptemperature_C,phumidity,newbattery,heating,heating_temp,uvi,light_lux,pm2_5_ug_m3,pm10_0_ug_m3,counter,code,repeat,maybe_battery,device_type,raw_message,switch1,switch2,switch3,switch4,switch5,extradata,house_id,module_id,sensor_count,alarms,sensor_value,battery_voltage,mode,version,type_string,failed,class,alert,secret_knock,relay,wind_dev_deg,exposure_mins,transmit_s,button_id,button_name,encrypted,misc,current_A,voltage_V,pairing,connected,gap,impulses,triggered,storage
2020-01-28 03:08:11,,,PCR800,,134,0,OK,,,,,,,,OS,0.270,,1.531,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
and
./rtl_433  -C si gives (i don't have .conf file)
time      : 2020-01-28 03:08:58                    brand     : OS
model     : PCR800       House Code: 134
Channel   : 0            Battery   : OK            Rain Rate : 0.3 in/hr     Total Rain: 1.5 in

i believe that inches are not compliant to metric system.
so there might be a conversion to make (conversion function in the code exist inch2mm()  but apparently not executed).
The problem is : there is a 3 colums for rainrate and 3 for total rain (csv format) : one without unit, inchs and the last one in inches. Only the first one (without unit) is filled with data.
Maybe we should fill only the columns with unit and discards (don't output) other ?
(I don't feel confortable about sending a column without specifying the unit)
what is your best bet on this ?

zuckschwerdt, 2020-01-28:
You need `-M newmodel` for the conversion to detect the keys. The displayed unit isn't great, just fixed that with 08ba329.

closed 2020-01-28 by zuckschwerdt

dgrostoto, 2020-01-28:
OK, tested with -M newmodel, conversion is done. Domoticz (which read output from rtl_433 in csv format)  is not compliant yet with newmodel. I'll try a modification if possible on domoticz side. I note that rain was send in inches since the beginning

dgrostoto, 2020-01-28:
Just a tough : doc says in readme that -C si make the conversion. Maybe a word could be added about newmodel ?

zuckschwerdt, 2020-01-28:
The grace period to switch to `newmodel` is now up (it's been around for year!). We'll default to `newmodel` shortly. (You can then still switch to `oldmodel` for maybe half a year before we remove the old stuff.)

dgrostoto, 2020-01-28:
OK, good to know. i try to propose the switch on domoticz side ASAP.