merbanan/rtl_433 #1281: Hackrf segmentation fault

loewal, 2020-01-30:
Hi there,

Having next problem running RTL_433 with a Hackrf (with Furrtek Havoc firmware):
[INFO] Opening HackRF One #0 223c69dc299f894f...
Using device HackRF One: clock source=external part id=a000cb3c0065434d serial=0000000000000000223c69dc299f894f version=local-79baef77
Found 1 antenna(s): TX/RX
Found 3 gain(s): LNA 0 - 116 (step 0) AMP 0 - 116 (step 0) VGA 0 - 116 (step 0)
Found 1 frequencies: RF
Found 1 frequency range(s): 0 - 7250000000 (step 0)
Found 20 sample rate range(s): 1000000 2000000 3000000 4000000 5000000 6000000 7000000 8000000 9000000 10000000 11000000 12000000 13000000 14000000 15000000 16000000 17000000 18000000 19000000 20000000
Found 16 bandwidth range(s): 1750000 - 1750000 (step 0) 2500000 - 2500000 (step 0) 3500000 - 3500000 (step 0) 5000000 - 5000000 (step 0) 5500000 - 5500000 (step 0) 6000000 - 6000000 (step 0) 7000000 - 7000000 (step 0) 8000000 - 8000000 (step 0) 9000000 - 9000000 (step 0) 10000000 - 10000000 (step 0) 12000000 - 12000000 (step 0) 14000000 - 14000000 (step 0) 15000000 - 15000000 (step 0) 20000000 - 20000000 (step 0) 24000000 - 24000000 (step 0) 28000000 - 28000000 (step 0)
Found current bandwidth 0
Found 4 stream format(s): CS8 CS16 CF32 CF64
Found native stream format: CS8 (full scale: 128.0)
zsh: segmentation fault rtl_433 -d driver="hackrf" -g "LNA=40,AMP=14,VGA=30" -M newmodel -vvv

What could be the problem.
My Hackrf is found, but the command

~/ soapysdrutil --probe="hackrf"
######################################################

Soapy SDR -- the SDR abstraction library
######################################################

Probe device hackrf
[INFO] Make connection: ''
Error probing device: Failed to make connection with ''

Gives no Hackrf....
Anyone??

forgot this:
~/ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000223c69dc299f894f
Board ID Number: 2 (HackRF One)
Firmware Version: local-79baef77 (API:1.03)
Part ID Number: 0xa000cb3c 0x0065434d

loewal, 2020-01-30:
and after cmake -DCMAKE_BUILD_TYPE=Debug ..

[INFO] Opening HackRF One #0 223c69dc299f894f...
Using device HackRF One: clock source=external part id=a000cb3c0065434d serial=0000000000000000223c69dc299f894f version=local-79baef77
Found 1 antenna(s): TX/RX
Found 3 gain(s): LNA 0 - 116 (step 0) AMP 0 - 116 (step 0) VGA 0 - 116 (step 0)
Found 1 frequencies: RF
Found 1 frequency range(s): 0 - 7250000000 (step 0)
Found 20 sample rate range(s): 1000000 2000000 3000000 4000000 5000000 6000000 7000000 8000000 9000000 10000000 11000000 12000000 13000000 14000000 15000000 16000000 17000000 18000000 19000000 20000000
Found 16 bandwidth range(s): 1750000 - 1750000 (step 0) 2500000 - 2500000 (step 0) 3500000 - 3500000 (step 0) 5000000 - 5000000 (step 0) 5500000 - 5500000 (step 0) 6000000 - 6000000 (step 0) 7000000 - 7000000 (step 0) 8000000 - 8000000 (step 0) 9000000 - 9000000 (step 0) 10000000 - 10000000 (step 0) 12000000 - 12000000 (step 0) 14000000 - 14000000 (step 0) 15000000 - 15000000 (step 0) 20000000 - 20000000 (step 0) 24000000 - 24000000 (step 0) 28000000 - 28000000 (step 0)
Found current bandwidth 0
Found 4 stream format(s): CS8 CS16 CF32 CF64
Found native stream format: CS8 (full scale: 128.0)
AddressSanitizer:DEADLYSIGNAL
=================================================================
==18757==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7fff66b27c92 bp 0x7ffee3bfa080 sp 0x7ffee3bfa080 T0)
==18757==The signal is caused by a READ memory access.
==18757==Hint: address points to the zero page.
    #0 0x7fff66b27c91 in _platform_strlen (libsystem_platform.dylib:x86_64+0x1c91)
    #1 0x10c660309 in wrap_strlen (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x18309)
    #2 0x10c60c6f3 in SoapySDRDevice_setupStream (libSoapySDR.0.8.0.dylib:x86_64+0x1b6f3)
    #3 0x10c1d678f in sdr_open_soapy sdr.c:692
    #4 0x10c1d2f2e in sdr_open sdr.c:796
    #5 0x10c1b42fa in main rtl_433.c:1476
    #6 0x7fff669317fc in start (libdyld.dylib:x86_64+0x1a7fc)

==18757==Register values:
rax = 0x000000010d2f968c  rbx = 0x0000615000005000  rcx = 0x0000000000000001  rdx = 0x0000000000000001
rdi = 0x0000000000000000  rsi = 0x00006070000007a0  rbp = 0x00007ffee3bfa080  rsp = 0x00007ffee3bfa080
 r8 = 0x0000000000000000   r9 = 0x0000000000000000  r10 = 0x00007fff9075f1e0  r11 = 0x000000010c60c6a0
r12 = 0x000000010d2f9688  r13 = 0x0000000000000001  r14 = 0x0000000000000001  r15 = 0x0000000000000000
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (libsystem_platform.dylib:x86_64+0x1c91) in _platform_strlen
==18757==ABORTING
zsh: abort      rtl_433 -d driver="hackrf" -g "LNA=40,AMP=14,VGA=30" -M newmodel -vvv

zuckschwerdt, 2020-01-30:
I guess this is a problem with the `SoapySDRDevice_setupStream` compat macro (or rather missing macro). You should not be able to compile rtl_433 against recent SoapySDR master, but perhaps some older headers are around?
I'd try SoapySDR 0.7.2. I'll look into wrapping SoapySDRDevice_setupStream for 0.8 compat soon.

loewal, 2020-01-30:
Compiled 0.7.2 and then again rtl_433 and the result:

Registered 119 out of 147 device decoding protocols
**_Failed to open sdr device matching 'driver=hackrf'._**

loewal, 2020-01-30:
forgot..
rtl_433 version 19.08-167-g74c527b branch master at 202001301036 inputs file rtl_tcp RTL-SDR SoapySDR

loewal, 2020-01-30:
Mounting a 820T....

Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
/Users/loewal/Downloads/rtl_433/src/devices/auriol_afw2a1.c:69:9: runtime error: index -1 out of bounds for type 'bitarray_t' (aka 'bitrow_t [25]')
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior /Users/loewal/Downloads/rtl_433/src/devices/auriol_afw2a1.c:69:9 in
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-01-30 12:30:56
model     : GT-WT02      ID Code   : 65
Channel   : 1            Battery   : LOW           Temperature: 9.1 C        Humidity  : 71 %          Button    : 0             Integrity : CHECKSUM

loewal, 2020-01-30:
Thnx in advance @zuckschwerdt !

merbanan, 2020-01-30:
/Users/loewal/Downloads/rtl_433/src/devices/auriol_afw2a1.c:69:9: runtime error: index -1 out of bounds for type 'bitarray_t' (aka 'bitrow_t [25]')

Index check is missing. A bug in the code. Should be fixed.

zuckschwerdt, 2020-01-30:
Oh, good find. The Auriol AFW2A1 is new and there is a missing check… yeah ^^^that.

zuckschwerdt, 2020-01-30:
Auriol AFW2A1 fixed with 8682c50
and SoapySDR 0.8 API with 056804c.

Make sure you don't have older 0.7 API headers around when compiling.

loewal, 2020-01-30:
Hi @zuckschwerdt , Christian
This is my new output:
[INFO] Opening HackRF One #0 223c69dc299f894f...
AddressSanitizer:DEADLYSIGNAL
=================================================================
==23517==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7fff66b27c92 bp 0x7ffeedf67060 sp 0x7ffeedf67060 T0)
==23517==The signal is caused by a READ memory access.
==23517==Hint: address points to the zero page.
    #0 0x7fff66b27c91 in _platform_strlen (libsystem_platform.dylib:x86_64+0x1c91)
    #1 0x1022ec309 in wrap_strlen (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x18309)
    #2 0x1022976f3 in SoapySDRDevice_setupStream (libSoapySDR.0.8.0.dylib:x86_64+0x1b6f3)
    #3 0x101e69966 in sdr_open_soapy sdr.c:701
    #4 0x101e660de in sdr_open sdr.c:807
    #5 0x101e474aa in main rtl_433.c:1476
    #6 0x7fff669317fc in start (libdyld.dylib:x86_64+0x1a7fc)

==23517==Register values:
rax = 0x0000000102f8568c  rbx = 0x0000615000005f00  rcx = 0x0000000000000001  rdx = 0x0000000000000001
rdi = 0x0000000000000000  rsi = 0x00006070000007a0  rbp = 0x00007ffeedf67060  rsp = 0x00007ffeedf67060
 r8 = 0x0000000000000000   r9 = 0x0000000000000000  r10 = 0x00007fff9075f1e0  r11 = 0x00000001022976a0
r12 = 0x0000000102f85688  r13 = 0x0000000000000001  r14 = 0x0000000000000001  r15 = 0x0000000000000000
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (libsystem_platform.dylib:x86_64+0x1c91) in _platform_strlen
==23517==ABORTING
zsh: abort      rtl_433 -d driver=hackrf -g "LNA=40,AMP=14,VGA=30" -M newmodel

loewal, 2020-01-30:
Do I perhaps need https://github.com/pothosware/SoapyHackRF?

loewal, 2020-01-30:
I wonder if my Hackrf-firmware (Furrtek-Havoc) could be the source of the problem.....

zuckschwerdt, 2020-01-30:
I would guess you already have the SoapyHackRF module.
Running `SoapySDRUtil --info` should show something like
```
Lib Version: v0.8 ...
Module found: /usr/local/lib/SoapySDR/modules0.8/libHackRFSupport.so
...
```
But the error isn't in HackRF, it's `SoapySDRDevice_setupStream()` calling `strlen(NULL)`,
which could be the `format` parameter filled in by `SoapySDRDevice_getNativeStreamFormat()`.
It shouldn't be NULL. Perhaps you can drop a few printf() in there to verify that. E.g.
```
    fprintf(stderr, "SoapySDRDevice_setupStream with %p, %p, %p (%s)\n", dev, dev->soapy_dev, (void *)format, format);
```
on https://github.com/merbanan/rtl_433/blob/master/src/sdr.c#L691

loewal, 2020-01-30:
This is the output after compiling:

Opening HackRF One #0 223c69dc299f894f...
SoapySDRDevice_setupStream with 0x607000000790, 0x615000005f80, 0x10ca0bfe0 (CS16)
AddressSanitizer:DEADLYSIGNAL

==24897==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7fff66b27c92 bp 0x7ffee3541040 sp 0x7ffee3541040 T0)
==24897==The signal is caused by a READ memory access.
==24897==Hint: address points to the zero page.
    #0 0x7fff66b27c91 in _platform_strlen (libsystem_platform.dylib:x86_64+0x1c91)
    #1 0x10cd16309 in wrap_strlen (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x18309)
    #2 0x10ccbe6f3 in SoapySDRDevice_setupStream (libSoapySDR.0.8.0.dylib:x86_64+0x1b6f3)
    #3 0x10c88f90d in sdr_open_soapy sdr.c:701
    #4 0x10c88bf3e in sdr_open sdr.c:807
    #5 0x10c86d30a in main rtl_433.c:1476
    #6 0x7fff669317fc in start (libdyld.dylib:x86_64+0x1a7fc)

==24897==Register values:
rax = 0x000000010d9af68c  rbx = 0x0000615000005f00  rcx = 0x0000000000000001  rdx = 0x0000000000000001
rdi = 0x0000000000000000  rsi = 0x00006070000007a0  rbp = 0x00007ffee3541040  rsp = 0x00007ffee3541040
 r8 = 0x0000000000000000   r9 = 0x0000000000000000  r10 = 0x00007fff9075f1e0  r11 = 0x000000010ccbe6a0
r12 = 0x000000010d9af688  r13 = 0x0000000000000001  r14 = 0x0000000000000001  r15 = 0x0000000000000000
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (libsystem_platform.dylib:x86_64+0x1c91) in _platform_strlen
==24897==ABORTING
zsh: abort      rtl_433 -d driver=hackrf -g "LNA=40,AMP=14,VGA=30" -M newmodel

And now for something completely strange (for me):
~/Downloads/rtl_433-master/build/ **soapysdrutil --info**
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.2-unknown
API Version: v0.7.1
ABI Version: v0.7
Install root: /opt/local
Search path:  /opt/local/lib/SoapySDR/modules0.7
Module found: /opt/local/lib/SoapySDR/modules0.7/libLMS7Support.so (19.04.0)
Available factories... lime
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

Version 0.8.0 AND 0.7.2.....what went wrong?

loewal, 2020-01-30:
I uninstalled the port-install version 0.7.2
Now it shows:
~/Downloads/rtl_433-master/build/ soapysdrutil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-unknown
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libHackRFSupport.so (0.3.3-3c514ce)
Available factories... hackrf
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

loewal, 2020-01-30:
Is this a problem?

[ 14%] Building C object src/CMakeFiles/rtl_433.dir/sdr.c.o
/Users/loewal/Downloads/rtl_433-master/src/sdr.c:691:74: warning: format specifies type 'void *' but the argument has type 'sdr_dev_t *' (aka 'struct sdr_dev *') [-Wformat-pedantic]
    fprintf(stderr, "SoapySDRDevice_setupStream with %p, %p, %p (%s)\n", dev, dev->soapy_dev, (void *)format, format);
                                                     ~~                  ^~~
/Users/loewal/Downloads/rtl_433-master/src/sdr.c:691:79: warning: format specifies type 'void *' but the argument has type 'SoapySDRDevice *' (aka 'struct SoapySDRDevice *') [-Wformat-pedantic]
    fprintf(stderr, "SoapySDRDevice_setupStream with %p, %p, %p (%s)\n", dev, dev->soapy_dev, (void *)format, format);
                                                         ~~                   ^~~~~~~~~~~~~~
2 warnings generated.

loewal, 2020-01-30:
Yes yes yes!!!! After recomplilation:

~/Downloads/rtl_433-master/build/ rtl_433 -d driver=hackrf -g "LNA=40,AMP=14,VGA=30" -M newmodel
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/loewal/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 119 out of 147 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 131-147 ]
[INFO] Opening HackRF One #0 223c69dc299f894f...
SoapySDRDevice_setupStream with 0x607000000790, 0x615000005f80, 0x102c48fe0 (CS16)
Sample rate set to 250000 S/s.
Setting gain element LNA: 40.000000 dB
Setting gain element AMP: 14.000000 dB
Setting gain element VGA: 30.000000 dB
Tuned to 433.920MHz.

and waiting!!

loewal, 2020-01-30:
Thank you @zuckschwerdt !!

loewal, 2020-01-30:
Perhaps too early:

~/Downloads/rtl_433-master/build/ soapysdrutil --probe="hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device hackrf
[ERROR] hackrf_exit() failed -- one or more HackRFs still in use
[ERROR] SoapySDR::Device::enumerate(hackrf) stoi: no conversion
[ERROR] hackrf_exit() failed -- one or more HackRFs still in use
Error probing device: no hackrf device matches
█▓▒░loewal@iMac-van-Loe-746█▓▒░ do jan 30 06:07:30P
~/Downloads/rtl_433-master/build/ soapysdrutil --probe="hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device hackrf
[ERROR] hackrf_exit() failed -- one or more HackRFs still in use
[ERROR] SoapySDR::Device::enumerate(hackrf) stoi: no conversion
[ERROR] hackrf_exit() failed -- one or more HackRFs still in use
Error probing device: no hackrf device matches
█▓▒░loewal@iMac-van-Loe-746█▓▒░ do jan 30 06:08:02P
~/Downloads/rtl_433-master/build/ soapysdrutil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-unknown
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libHackRFSupport.so (0.3.3-3c514ce)
Available factories... hackrf
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

~/Downloads/rtl_433-master/build/ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000223c69dc299f894f
Board ID Number: 2 (HackRF One)
Firmware Version: local-79baef77 (API:1.03)
Part ID Number: 0xa000cb3c 0x0065434d

Not busy at all. Message appears even after reset of Hackrf

loewal, 2020-01-30:
Another test:
~/ soapysdrutil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 223c69dc299f894f
  part_id = a000cb3c0065434d
  serial = 0000000000000000223c69dc299f894f
  version = local-79baef77

Why is probing going wrong?

loewal, 2020-01-30:
And even stranger?????

~/ soapysdrutil --probe="rtlsdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device rtlsdr
[INFO] Opening HackRF One #0 223c69dc299f894f...

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=HackRF
  hardware=HackRF One
  clock source=external
  part id=a000cb3c0065434d
  serial=0000000000000000223c69dc299f894f
  version=local-79baef77

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: NO
  Other Settings:
     * Antenna Bias - Antenna port power control.
       [key=bias_tx, default=false, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: NO
  Stream formats: CS8, CS16, CF32, CF64
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Count - Number of buffers per read.
       [key=buffers, units=buffers, default=15, type=int]
  Antennas: TX/RX
  Full gain range: [0, 116] dB
    LNA gain range: [0, 40] dB
    AMP gain range: [0, 14] dB
    VGA gain range: [0, 62] dB
  Full freq range: [0, 7250] MHz
    RF freq range: [0, 7250] MHz
  Sample rates: 1, 2, 3, 4, 5, ..., 16, 17, 18, 19, 20 MSps
  Filter bandwidths: 1.75, 2.5, 3.5, 5, 5.5, ..., 14, 15, 20, 24, 28 MHz

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: NO
  Stream formats: CS8, CS16, CF32, CF64
  Native format: CS8 [full-scale=128]
  Stream args:
     * Buffer Count - Number of buffers per read.
       [key=buffers, units=buffers, default=15, type=int]
  Antennas: TX/RX
  Full gain range: [0, 61] dB
    VGA gain range: [0, 47] dB
    AMP gain range: [0, 14] dB
  Full freq range: [0, 7250] MHz
    RF freq range: [0, 7250] MHz
  Sample rates: 1, 2, 3, 4, 5, ..., 16, 17, 18, 19, 20 MSps
  Filter bandwidths: 1.75, 2.5, 3.5, 5, 5.5, ..., 14, 15, 20, 24, 28 MHz

loewal, 2020-01-30:
I hope you will read next confusing lines....crosslinked rtlsdr and hackrf??

~/ brew uninstall soapyhackrf
Uninstalling /usr/local/Cellar/soapyhackrf/0.3.3... (6 files, 114.4KB)
█▓▒░loewal@iMac-van-Loe-746█▓▒░ do jan 30 06:27:00P
~/ soapysdrutil --find="hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

No devices found! hackrf
█▓▒░loewal@iMac-van-Loe-746█▓▒░ do jan 30 06:27:06P
~/ soapysdrutil --find="rtlsdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

No devices found! rtlsdr
█▓▒░loewal@iMac-van-Loe-746█▓▒░ do jan 30 06:27:28P
~/ brew install soapyhackrf
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).
==> Updated Formulae
axel               convox             dvc                ffuf               gitlab-runner      helm               kops               mavsdk             n                  pre-commit         serverless         ungit
borgmatic          coq                exiftool           flow               go@1.12            igraph             kubeseal           maxima             nim                procs              sn0int             unrar
broot              docutils           exploitdb          fonttools          grafana            jenkins            logtalk            mesa               osquery            protoc-gen-go      stress-ng          ykman
cfn-lint           drone-cli          fastlane           gatsby-cli         hadolint           jenkins-lts        math-comp          mps-youtube        pgformatter        rocksdb            topgrade           yubico-piv-tool

==> Installing soapyhackrf from pothosware/pothos
==> Downloading https://github.com/pothosware/SoapyHackRF/archive/soapy-hackrf-0.3.3.tar.gz
Already downloaded: /Users/loewal/Library/Caches/Homebrew/downloads/b6a15b43b2a75f938add3132f78a40a52bc625e01fc20e195592c86312013aa1--SoapyHackRF-soapy-hackrf-0.3.3.tar.gz
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/soapyhackrf/0.3.3 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev
==> make install
Warning: pothosware/pothos/soapyhackrf dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
🍺  /usr/local/Cellar/soapyhackrf/0.3.3: 6 files, 104.4KB, built in 12 seconds
█▓▒░loewal@iMac-van-Loe-746█▓▒░ do jan 30 06:28:38P
~/ soapysdrutil --find="rtlsdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 223c69dc299f894f
  part_id = a000cb3c0065434d
  serial = 0000000000000000223c69dc299f894f
  version = local-79baef77

█▓▒░loewal@iMac-van-Loe-746█▓▒░ do jan 30 06:28:43P
~/ soapysdrutil --find="hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[ERROR] SoapySDR::Device::enumerate(hackrf) stoi: no conversion
No devices found! hackrf
█▓▒░loewal@iMac-van-Loe-746█▓▒░ do jan 30 06:28:50P
~/ brew uninstall soapyhackrf
Uninstalling /usr/local/Cellar/soapyhackrf/0.3.3... (6 files, 104.4KB)
█▓▒░loewal@iMac-van-Loe-746█▓▒░ do jan 30 06:29:03P
~/ soapysdrutil --find="rtlsdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

No devices found! rtlsdr

loewal, 2020-01-30:
and the last try:

~/ brew install soapyhackrf
==> Installing soapyhackrf from pothosware/pothos
==> Downloading https://github.com/pothosware/SoapyHackRF/archive/soapy-hackrf-0.3.3.tar.gz
Already downloaded: /Users/loewal/Library/Caches/Homebrew/downloads/b6a15b43b2a75f938add3132f78a40a52bc625e01fc20e195592c86312013aa1--SoapyHackRF-soapy-hackrf-0.3.3.tar.gz
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/soapyhackrf/0.3.3 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev
==> make install
Warning: pothosware/pothos/soapyhackrf dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
🍺  /usr/local/Cellar/soapyhackrf/0.3.3: 6 files, 104.4KB, built in 11 seconds
█▓▒░loewal@iMac-van-Loe-746█▓▒░ do jan 30 06:32:45P
~/ soapysdrutil --find="hackrf"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[ERROR] SoapySDR::Device::enumerate(hackrf) stoi: no conversion
No devices found! hackrf
█▓▒░loewal@iMac-van-Loe-746█▓▒░ do jan 30 06:32:48P
~/ soapysdrutil --find="rtlsdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 223c69dc299f894f
  part_id = a000cb3c0065434d
  serial = 0000000000000000223c69dc299f894f
  version = local-79baef77

zuckschwerdt, 2020-01-30:
Yes, it is a known problem, you can't have 0.7 and 0.8 installed at the same time. Headers and libs probably got mixed up, thus the crash.

The RTL-SDR isn't found because you don't have the module around. You need SoapyRTLSDR.
SoapySDRUtil --info needs to show:
Module found: /opt/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so

loewal, 2020-01-30:
Should I uninstall everything related under **port** and **brew**?

zuckschwerdt, 2020-01-30:
Yes, just using Brew or MacPorts is easier on Mac. E.g. with MacPorts get:
- rtl-sdr
- SoapySDR
- SoapyHackRF
- SoapyRTLSDR

Just make sure to uninstall everything from /usr/local first.

closed 2020-02-05 by zuckschwerdt