merbanan/rtl_433 #1282: honda decoder also decodes uk 2013 ford mondeo

mike501, 2020-01-30:
Just wanted to let @zuckschwerdt and @ukshady (honda) and @winterrace (ford) know that the honda protocol is detecting my 2013 uk ford mondeo remote key press. 

Its reporting it as device id 65535 and code unknown for both the lock and unlock key presses.

Some details below - I'm new to this, happy to provide more help but you'll need to talk me through the steps.

First press was unlock, second press was lock.

```
~/rtl_433 $ rtl_433  -A -G 4 
rtl_433 version 19.08-149-g51d47ce branch master at 202001222257 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Use -G for testing only. Enable protocols with -R if you really need them.

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 139 out of 145 device decoding protocols [ 1-4 6-8 10-17 19-26 29-64 67-145 ]
rtl_433: warning: 105 "Wireless M-Bus, Mode S, 32.768kbps (-f 868300000 -s 1000000)" does not support CSV output
rtl_433: warning: 106 "Wireless M-Bus, Mode R, 4.8kbps (-f 868330000)" does not support CSV output
rtl_433: warning: 107 "Wireless M-Bus, Mode F, 2.4kbps" does not support CSV output
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
Detected FSK package	2020-01-30 14:04:52
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-01-30 14:04:52
model     : Honda Remote device id : 65535         code      : unknown
Analyzing pulses...
Total count:  390,  width: 99.32 ms		(24830 S)
Pulse width distribution:
 [ 0] count:    1,  width:   68 us [68;68]	(  17 S)
 [ 1] count:  388,  width:  124 us [120;136]	(  31 S)
 [ 2] count:    1,  width:   12 us [12;12]	(   3 S)
Gap width distribution:
 [ 0] count:    1,  width:   76 us [76;76]	(  19 S)
 [ 1] count:  388,  width:  124 us [112;132]	(  31 S)
Pulse period distribution:
 [ 0] count:    1,  width:  144 us [144;144]	(  36 S)
 [ 1] count:  388,  width:  252 us [240;260]	(  63 S)
Level estimates [high, low]:  15880,    302
RSSI: -0.1 dB SNR: 17.2 dB Noise: -17.3 dB
Frequency offsets [F1, F2]:    6060,  -5633	(+23.1 kHz, -21.5 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 68, long_width: 124, reset_limit: 136, sync_width: 12
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=68,l=124,r=136,g=0,t=0,y=12'
pulse_demod_pwm(): Analyzer Device 
bitbuffer:: Number of rows: 2 
[00] {389} 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
[01] { 0}                                                                                                                                                    : 

Detected FSK package	2020-01-30 14:04:52
Analyzing pulses...
Total count:  121,  width: 21.61 ms		( 5402 S)
Pulse width distribution:
 [ 0] count:   72,  width:   60 us [60;72]	(  15 S)
 [ 1] count:   49,  width:  124 us [120;132]	(  31 S)
Gap width distribution:
 [ 0] count:   72,  width:   64 us [60;72]	(  16 S)
 [ 1] count:   48,  width:  128 us [124;132]	(  32 S)
[... 190 lines cut ...]
Analyzing pulses...
Total count:  117,  width: 21.60 ms		( 5399 S)
Pulse width distribution:
 [ 0] count:   64,  width:   60 us [52;72]	(  15 S)
 [ 1] count:   53,  width:  124 us [116;132]	(  31 S)
Gap width distribution:
 [ 0] count:   64,  width:   64 us [60;76]	(  16 S)
 [ 1] count:   52,  width:  124 us [124;132]	(  31 S)
Pulse period distribution:
 [ 0] count:   49,  width:  128 us [124;148]	(  32 S)
 [ 1] count:   38,  width:  252 us [248;256]	(  63 S)
 [ 2] count:   29,  width:  188 us [188;196]	(  47 S)
Level estimates [high, low]:  15953,    248
RSSI: -0.1 dB SNR: 18.1 dB Noise: -18.2 dB
Frequency offsets [F1, F2]:    7010,  -4897	(+26.7 kHz, -18.7 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 60, long_width: 0, reset_limit: 136, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=60,l=0,r=136'
pulse_demod_manchester_zerobit(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {170} 2a aa aa aa e9 ca 87 13 04 49 d2 59 54 b5 2e fc f6 7c f0 2c 55 c0 

^CSignal caught, exiting!
Reattached kernel driver
```

zuckschwerdt, 2020-01-30:
This looks like the real data payload
```
{170} 2a aa aa aa e9 ca 87 13 04 49 d2 2f 22 c3 58 82 8a f6 78 00 2d 40 
```

To get good data use a command of
```
rtl_433 -R 0 -X 'n=Mondeo-Remote,m=OOK_MC_ZEROBIT,s=64,r=136,preamble=aaae'
```

mike501, 2020-01-30:
I've only been using rtl_433 for a couple of days - so i'm likely to be missing understanding. I've ran the command and got nothing outputted to the screen. Running it again. writing the output to a file (via -W) and pressing open 3 times then close 3 times i've got a 4mb file . Trying various options against the file (using -r) and nothing makes sense. 

Bear in mind i'm in a noisy 433mhz environment - i've got the neighbours tpms and wireless alarms kicking out lots of messages.

[mondeo-remote.zip](https://github.com/merbanan/rtl_433/files/4136123/mondeo-remote.zip)

zuckschwerdt, 2020-01-30:
The file is good, I'll analyze it soon. If you drop it on https://triq.net/iqs you can see there are only two other signals, one is weak (the streak over nearly the whole recording) and the other is very local. Neither obsure anything :) The 6 presses are visible as a long preamble and then two data packets each.

zuckschwerdt, 2020-01-30:
My bad, it's obviously FSK:
```
rtl_433 -R 0 -X 'n=Mondeo-Remote,m=FSK_MC_ZEROBIT,s=64,r=136,preamble=aaae'
```

zuckschwerdt, 2020-01-30:
Btw. the data shows the expected clear header, then encrypted data, then perhaps a 16-bit checksum. It's strange though that so much of the encrypted data collides. It should be random -- it isn't. Even in those 6 presses you managed to get a pair of nearly identical codes, marked below.
```
9ca8 7130 449d 20 78a6 bf03 2225 a938 eb48
9ca8 7130 449d 25 d50b 12af 8daf 883c da7c (*)

9ca8 7130 449d 20 70ae b70b 2a2d a144 7380
9ca8 7130 449d 25 d50b 12af 8da7 8842 ec84 (*)

9ca8 7130 449d 25 c51b 02be 9dbf 994d 620c
9ca8 7130 449d 20 78a6 bf02 2225 b84c 4f4c

9ca8 7130 449d 25 c51b 02bf 1db7 98d5 8d44

9ca8 7130 449d 20 5886 9f22 8205 c8de a078
9ca8 7130 449d 25 954b 52ef cdcf c85d 0e84

9ca8 7130 449d 25 954b 52ef 4dc7 c8e7 7c8c
9ca8 7130 449d 20 508e 972a 0a0d c066 7c44
```
Perhaps you can grab a bunch more codes. If there are enough codes differing by only a few bits then we can deduce the checksum.

mike501, 2020-01-30:
I wonder if the repeated codes is because I pressed the buttons so close together?

I'll try and grab more tomorrow. I'll work out a sequence of repeated presses with a short delay, then long delay, then more realistic lock / unlock sequences.

mike501, 2020-01-31:
I've done a new capture, the sequence should be as follows,
quickly press unlock 3 times
quickly press lock 3 times
wait 2s then press press unlock
wait 2s then press lock
wait 2s then press press unlock
wait 2s then press lock
wait 2s then press press unlock
wait 2s then press lock
press unlock boot two times

Where can i find a guide to do the analysis you're doing? I'd like to do the same to my wifes Nissan. I'll hopefully be ok looking for patterns in binary data (i used to be a developer many years ago so am used to binary) but taking a file like i've attached and converting it to binary is new to me.

[mondeo-remote2.zip](https://github.com/merbanan/rtl_433/files/4139889/mondeo-remote2.zip)

zuckschwerdt, 2020-01-31:
Just use
```
rtl_433 -R 0 -X 'n=Mondeo-Remote,m=FSK_MC_ZEROBIT,s=64,r=136,preamble=aaae'
```
eihter live or on the captured files.

mike501, 2020-01-31:
Its complained that i'm missing the length, after running the previous command I could see a length of 25391 listed a few times so i added l=25391. I've now got the same output as you for the first file. I'll have a play with the second file later and see what i find & if it matches what you see.

thanks for the help.

zuckschwerdt, 2020-01-31:
The data in the second sample is
```
 1.556328s  9ca8 7130 449d 20 c816 0fb3 92a2 9251 1c0c
 1.955552s  9ca8 7130 449d 22 62bc a519 383a aa5b 8e6c
 2.254972s  9ca8 7130 449d 20 9846 5fe3 c2d2 c263 edc4
 2.554380s  9ca8 7130 449d 20 9846 5fe2 c2d2 c161 3284
 3.781460s  9ca8 7130 449d 20 9846 5fe2 42da c1e8 46bc
 4.180692s  9ca8 7130 449d 20 8856 4ff2 52c2 d1f6 5344
 4.480096s  9ca8 7130 449d 22 429c 8538 981a c9fb 8b38
 4.779528s  9ca8 7130 449d 20 8856 4ff2 d2ca d17f 277c
 6.537500s  9ca8 7130 449d 22 6ab4 ad11 3032 f282 fd54
 6.936712s  9ca8 7130 449d 22 6ab4 ad10 3032 f180 2214
 8.100160s  9ca8 7130 449d 22 62bc a518 b83a fa08 9d2c
 8.499376s  9ca8 7130 449d 20 9846 5fe2 c2fa c188 71e4
 9.715763s  9ca8 7130 449d 20 8856 4ff3 d2e2 d294 bb58
10.114984s  9ca8 7130 449d 20 8856 4ff2 d2e2 d196 6418
11.200936s  9ca8 7130 449d 20 8856 4ff2 52ea d21c 7514
11.600164s  9ca8 7130 449d 22 62bc a518 383a e999 85ac
12.698332s  9ca8 7130 449d 20 d806 1fa3 8212 82a4 518c
13.097556s  9ca8 7130 449d 20 d806 1fa2 8212 81a6 8ecc
14.125748s  9ca8 7130 449d 20 d806 1fa2 021a 822c 9fc4
14.524984s  9ca8 7130 449d 22 c21c 05b8 989a 19ab 5f3c
16.020329s  9ca8 7130 449d 20 c816 0fb0 9202 93b0 917c
16.419556s  9ca8 7130 449d 22 ca14 0db0 9092 01b2 f884
16.912588s  9ca8 7130 449d 22 c21c 05ba 989a 0bb8 5cfc
17.311829s  9ca8 7130 449d 20 c816 0fb2 920a 91b8 9b34
```

zuckschwerdt, 2020-01-31:
The "missing long length" was fixed recently. just set it to the short length, i.e. `,l=64`

commit 99ee9174 (2022-06-19): Add Mondeo remote decoder conf (closes #1282)
closed 2022-06-19 by zuckschwerdt