merbanan/rtl_433 #1290: Floating point exception with invalid sample rate

evilpete, 2020-02-04:
```
/usr/bin/rtl_433 -s 1000 -f 915.00M
Found Rafael Micro R820T tuner
Invalid sample rate: 1000 Hz
WARNING: Failed to set sample rate.
Tuner gain set to Auto.
Tuned to 915.000MHz.
Floating point exception (core dumped)
```

it's happening somewhere in `pulse_demod_pcm`

given the cause is most likely the Invalid sample rate I didn't feel that motivated to track it any further then the frame. :-)

zuckschwerdt, 2020-02-04:
Yes, we currently need at least 100k sample rate. Lower rates will round widhts in some decoders to zero and then div-by-zero.
I'll add an error and abort in the protocol registration. Each active protocol will then check for the some minimum required sample rate.

closed 2020-02-04 by zuckschwerdt

evilpete, 2020-02-04:
yea garbage in garbage out, I figured you should be alerted to the Segfault resulting form a command line typo

zuckschwerdt, 2020-02-04:
Sure, thanks! It's fixed now :)

commit c19650c3 (2020-02-04): Fix failure on low sample rates (closes #1290)