merbanan/rtl_433 #1292: Release

Mindavi, 2020-02-05:
Hi, I was wondering if a release is planned or if it possibly should be planned. I'm not sure about the distribution of users, but I assume there might be some users that use the version distributed by their package manager, e.g. ubuntu (https://packages.ubuntu.com/eoan/rtl-433). Because their repositories only contain releases, those users might miss out on some recently added features and decoders.

Would it be good to make a release and if so, are there any issues that need to be solved before doing so? Or is releasing something that's considered non-important/not needed?

Honestly, I really can't tell if there are users that just use the version provided by the package repositories, because this is a niche tool mostly used by enthousiasts, I'd assume.

merbanan, 2020-02-05:
My idea was to make a release before February 10. @zuckschwerdt any comments on that?

zuckschwerdt, 2020-02-05:
I don't have any short term feature PRs, just "someday" stuff. (And decoders will always be evolving.) I'd say we are good for a release tag. However:

I would be really interested about the user base, what platforms and if it's precompiled or from git. I've been playing with static builds and CI for armhf/64, i686/amd64 to help users just download a binary. But I'm not too sure if that's a thing for SDR hackers ;)

Maybe we could output a prominent link with some encoded but visible stats? Allthough I'm not one for automatic usage stats in background, I'd click a simple link. E.g.

> Click this link to let the developers know what system is most used:
> https:// foo /stats?platfom=foo&ver=foo

merbanan, 2020-02-06:
Things before release:
Fix Thermopro 11 and 12 integrity check.

merbanan, 2020-02-09:
I'd like to get the release out soon. Do we have any nice to have stuff that we want in the release ? Some of the long gone PR's would be nice.

Mindavi, 2020-02-09:
Maybe we could do a sweep through all decoders to fill in any missing return code for the diagnostic mode. I'm not sure what else is important, but according to the bug reports something about the minmax vs classic decoder should definitely be mentioned indeed.

merbanan, 2020-02-09:
@Mindavi I'm like 25% through all decoder regarding that.

alexmyczko, 2020-02-10:
@mindavi https://github.com/merbanan/rtl_433/issues/1095 

https://qa.debian.org/popcon.php?package=rtl-433

Mindavi, 2020-02-10:
Shall I pick up a part of the error code implementation? If so, maybe we should track progress somewhere?

merbanan, 2020-02-10:
@Mindavi I did it alphabetically, last file was holman_ws5029.c in feat-error-codes2 branch. I mixed some other stuff in there so it can't be applied yet. Anyway if you continue and only do the error codes we can merge those quickly. Anyway that would be very helpful.

merbanan, 2020-02-10:
@Mindavi just push the return code fixes directly. Simple fixes can be pushed directly without a PR. More complex changes can go as a PR.

Mindavi, 2020-02-10:
Most devices have correct(ish) return codes now. #1314 lists some missing ones, which use the 'event pattern'.

Notably, I've skipped `m_bus.c`, which needs a good look.

You can find all non-implemented decoders using `grep -R "return 0" -l | sort | less`.

This gives the following list, of which most should be mentioned in #1314.

```
acurite.c
brennenstuhl_rcs_2044.c
current_cost.c
digitech_xc0324.c
dsc.c
flex.c
gt_wt_02.c
hondaremote.c
ht680.c
ikea_sparsnas.c
m_bus.c
new_template.c
oil_standard.c
oregon_scientific.c
oregon_scientific_sl109h.c
radiohead_ask.c
tpms_ford.c
tpms_pmv107j.c
tpms_renault.c
tpms_toyota.c
ttx201.c
```

merbanan, 2020-02-10:
I'd say it's enough return codes for the release. The ones left needs a more in-depth look.

merbanan, 2020-02-16:
I have added the output information. I have no further things I think is vital before a release. @zuckschwerdt do you have something you want to add? If not can you create a release tomorrow?

zuckschwerdt, 2020-02-16:
Ok, I'll compile the changelog and tag a release tomorrow.

closed 2020-02-17 by zuckschwerdt

alexmyczko, 2020-02-17:
how am i supposed to build the docs? ./build-docs.sh ? and the tarball release spits out
```
$ rtl_433 -V
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
```

zuckschwerdt, 2020-02-17:
The Doxygen stuff is just for the maintainers. It's really not used but a style check for now.
Use `cmake -DBUILD_DOCUMENTATION:BOOL=OFF ..` if you accidentally turned it on.

Yes, we need a solution to insert a version number when not building from Git with CMake. Not sure if we really want to put a version number (and keep it up to date) somewhere.

alexmyczko, 2020-02-17:
> The Doxygen stuff is just for the maintainers. It's really not used but a style check for now.
> Use `cmake -DBUILD_DOCUMENTATION:BOOL=OFF ..` if you accidentally turned it on.

I maintain the debian package, and fail to build the docs?

zuckschwerdt, 2020-02-17:
As for distrib docs, that's just the plain .md files.
Those docs (not doxygen source code docs) can be found at https://triq.org/rtl_433/

The doxygen docs are not used. I guess you can build them, but they are not distributed or available anywhere. There is no useful index or intro page.

I'm not aware of an error in the troff man page. What checker is that?

alexmyczko, 2020-02-17:
something from https://salsa.debian.org/lintian/lintian

zuckschwerdt, 2020-02-17:
Probably https://salsa.debian.org/lintian/lintian/blob/master/checks/documentation/man.pm then. Let's see if we can spot what's wrong.

alexmyczko, 2020-02-17:
I fixed the -doc thing by dropping the package for now.

merbanan, 2020-02-17:
@alexmyczko can you sponsor a patch to the debian rtl-sdr package for me?

merbanan, 2020-02-17:
https://github.com/osmocom/rtl-sdr/pull/9

alexmyczko, 2020-02-17:
guess so well i can put that in the source package and a DD might sponsor it, i can not ;)

zuckschwerdt, 2020-02-18:
For the troff checking the script runs `man --warnings -E UTF-8 -l -Tutf8 -Z rtl_433.1`.
I don't get the warning with man 2.7.6.1 here. But it looks like an overlong line warning.