merbanan/rtl_433 #1294: THGR222 temperature and humidity sensor

pcruchet, 2020-02-05:
Hello, has anyone tried to decode the frames of a THGR222 it seems that it is oregon scientific V3 frames. Here's what I get:
Badly formatted OS v2.1 message encountered.
Raw Data: {311} 00 00 00 be 28 51 cf 04 48 0d 83 e4 3f ff ff fe 
be 28 51 cf 04 48 0d 83 e4 3f ff ff fe be 28 51 cf 04 48 0d 83 e4 3e 

Possible Oregon Scientific v3 message, but sync nibble wasn't found
Raw Data: {311} 00 00 00 be 28 51 cf 04 48 0d 83 e4 3f ff ff fe 
be 28 51 cf 04 48 0d 83 e4 3f ff ff fe be 28 51 cf 04 48 0d 83 e4 3e 

Thanks if you have a solution
Philippe

hackery, 2020-02-07:
Do you mean THGR222 - or THGR221? I don't see a THGR222 listed anywhere - and I've also just bought a THGR221 which I want to add decoding for.

zuckschwerdt, 2020-02-07:
It's best to grab some samples (-S all) find a good one (https://triq.net/iqs) and post that here with a summary on the sensor (what does it measure) and expected readings in that sample (what values did it probably measure). A screenshot from Pulseview can also help to quickly discuss the coding, and also the timing output of the analyzer (-A).

pcruchet, 2020-02-07:
It is indeed **THGR221
[433.92M_250K.cu8.zip](https://github.com/merbanan/rtl_433/files/4171073/433.92M_250K.cu8.zip)

**.
This sensor measures Temperature and Humidity. It looks like the THGR810

**I was able to get this information once :**
Detected OOK package	2020-02-06 19:36:47
time      : 2020-02-06 19:36:47                    brand     : OS
model     : THGR810      House Code: 188
Channel   : 1            Battery   : OK            Celsius   : 21.80 C       Humidity  : 39 %
Analyzing pulses...
Total count:  255,  width: 305.26 ms		(76316 S)
Pulse width distribution:
 [ 0] count:    1,  width:   44 us [44;44]	(  11 S)
 [ 1] count:  197,  width:  504 us [500;520]	( 126 S)
 [ 2] count:   57,  width:  992 us [992;1000]	( 248 S)
Gap width distribution:
 [ 0] count:    1,  width: 1948 us [1948;1948]	( 487 S)
 [ 1] count:  196,  width:  468 us [460;476]	( 117 S)
 [ 2] count:   57,  width:  956 us [952;964]	( 239 S)
Pulse period distribution:
 [ 0] count:   37,  width: 1952 us [1952;1992]	( 488 S)
 [ 1] count:  176,  width:  976 us [972;988]	( 244 S)
 [ 2] count:   41,  width: 1464 us [1460;1472]	( 366 S)
Level estimates [high, low]:  15960,    334
RSSI: -0.1 dB SNR: 16.8 dB Noise: -16.9 dB
Frequency offsets [F1, F2]:   17274,      0	(+65.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 504, long_width: 992, reset_limit: 1952, sync_width: 44
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=504,l=992,r=1952,g=0,t=0,y=44'
pulse_demod_pwm(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {254} ff ff fe 7e 73 dd b7 be bf 58 ef ff ff f3 f3 9e ed bd f5 fa c7 7f ff ff 9f 9c f7 6d ef af d6 38 ``

**most often I get with the -A option**
Detected OOK package	2020-02-07 14:34:37
Analyzing pulses...
Total count:  251,  width: 303.26 ms		(75816 S)
Pulse width distribution:
 [ 0] count:  191,  width:  504 us [500;512]	( 126 S)
 [ 1] count:   60,  width:  992 us [992;1000]	( 248 S)
Gap width distribution:
 [ 0] count:  190,  width:  468 us [464;476]	( 117 S)
 [ 1] count:   60,  width:  956 us [956;964]	( 239 S)
Pulse period distribution:
 [ 0] count:  175,  width:  976 us [972;984]	( 244 S)
 [ 1] count:   30,  width: 1464 us [1460;1468]	( 366 S)
 [ 2] count:   45,  width: 1952 us [1948;1956]	( 488 S)
Level estimates [high, low]:  15991,    380
RSSI: -0.1 dB SNR: 16.2 dB Noise: -16.3 dB
Frequency offsets [F1, F2]:   18703,      0	(+71.3 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 504, long_width: 0, reset_limit: 968, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=504,l=0,r=968'
pulse_demod_manchester_zerobit(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {311} 00 00 00 be 28 51 2b 11 48 01 8b c4 17 ff ff fe be 28 51 2b 11 48 01 8b c4 17 ff ff fe be 28 51 2b 11 48 01 8b c4 16 

Thanks for your help

pcruchet, 2020-02-07:
Sensor photography
[image]

hackery, 2020-02-07:
I'm trying to get some additional IQ files, but my environment has a fair few devices in range giving  sample files from different unknown devices - how best to isolate a new device under test (short of building a Faraday cage...) ?

zuckschwerdt, 2020-02-07:
It's FSK PCM with a bit width of 50 µs.
[image]

```
rtl_433 -R 0 -X 'n=THGR221,m=FSK_PCM,s=50,l=50,r=1000'
```
There is an identical payload in all samples of:
```
...555555ff401965bdfe6adaa7194f481fc7fd487880
```
This doesn't look much like the OS protocol at fist glance.

If you can grab more (different) codes lines with the flex decoder above that would be great!

zuckschwerdt, 2020-02-07:
Remove the antenna and place the device at 10cm to the receiver, that mostly isolates the transmissions.

hackery, 2020-02-07:
I'm now seeing this with no other devices showing:
```
time      : 2020-02-07 14:42:20                    brand     : OS
model     : THGR810      House Code: 213
Channel   : 1            Battery   : OK            Celsius   : 22.30 C       Humidity  : 40 %
```
`Channel` changes between 1/2/3 matching the channel slider on the rear of the unit.

pcruchet, 2020-02-07:
I think OREGON Scientific THGR221 is a new device and the protocol is a little different from the previous ones.

### **With no antenna**

pcruchet@b107PCT:~$ **rtl_433 -R 0 -X 'n=THGR221,m=OOK_PWM,s=504,l=992,r=1952,g=0,t=0,y=44'**
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/USERS/PROFS/pcruchet/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 121 device decoding protocols [ ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-02-07 16:10:29
model     : THGR221      count     : 1             num_rows  : 1             rows      : 
len       : 245          data      : fffffe7e734375fafb8affffff9f9cd0dd7ebee2bfffffe7e734375fafb8a8
codes     : {245}fffffe7e734375fafb8affffff9f9cd0dd7ebee2bfffffe7e734375fafb8a8

With this command : **rtl_433 -R 0 -X 'n=THGR221,m=FSK_PCM,s=50,l=50,r=1000'** nothing

[g009_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/4171483/g009_433.92M_250k.cu8.zip)

and the last one 

pcruchet@b107PCT:~$ rtl_433 -R 0 -A
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/USERS/PROFS/pcruchet/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 0 out of 121 device decoding protocols [ ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Detected OOK package	2020-02-07 16:18:00
Analyzing pulses...
Total count:  249,  width: 303.76 ms		(75940 S)
Pulse width distribution:
 [ 0] count:  187,  width:  504 us [500;520]	( 126 S)
 [ 1] count:   62,  width:  992 us [988;1000]	( 248 S)
Gap width distribution:
 [ 0] count:  185,  width:  468 us [464;476]	( 117 S)
 [ 1] count:   63,  width:  956 us [956;964]	( 239 S)
Pulse period distribution:
 [ 0] count:  168,  width:  976 us [972;992]	( 244 S)
 [ 1] count:   35,  width: 1464 us [1460;1468]	( 366 S)
 [ 2] count:   45,  width: 1952 us [1948;1960]	( 488 S)
Level estimates [high, low]:  15964,     15
RSSI: -0.1 dB SNR: 30.0 dB Noise: -30.1 dB
Frequency offsets [F1, F2]:   20012,      0	(+76.3 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 504, long_width: 0, reset_limit: 968, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=504,l=0,r=968'
pulse_demod_manchester_zerobit(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {311} 00 00 00 be 28 51 2a 82 48 19 83 e4 65 ff ff fe be 28 51 2a 82 48 19 83 e4 65 ff ff fe be 28 51 2a 82 48 19 83 e4 64

zuckschwerdt, 2020-02-07:
Very strange. The THGR810 (and all other OS) are OOK Manchester at 440 µs half-bit width.
The samples posted don't match that at all. Maybe those are from something else?

zuckschwerdt, 2020-02-07:
I'll take a look at the new samples.

zuckschwerdt, 2020-02-07:
Yes, that g009 sample is OOK Manchester, of the OS style. What's that you captured in first zip file?

pcruchet, 2020-02-07:
surely noise, with no antenna it's better

zuckschwerdt, 2020-02-07:
not noise ;) It's a big chunk of real data. Likely a neighbours weather station.
It's not a TPMS -- the transmission and data is too stable. And it's not just temperature/humidity -- it's a sophisticated coding with lots of content.

pcruchet, 2020-02-07:
If I could only decode the THGR221 frame it would be fine. It's hard to find old sensors. It is already the second that I buy and that did not recognize by the software.

hackery, 2020-02-07:
This is behaving fine for me so far, treating it as a THGR800 with no code changes from master branch ("Office" graph in image):
[image]

zuckschwerdt, 2020-02-07:
@pcruchet you can always analyze and add support for new sensors. Here is an overview:

rtl_433 processes radio data in multiple stages.

First a radio data packet is found and framed.
Use CubicSDR or Gqrx to verify you receice a signal.
Note the frequency, pick a frequency a little off, e.g 50k above or below.
Then grab the signal with rtl_433, e.g. `rtl_433 -f 433.92M -S all`
Visually verify the samples in https://triq.net/iqs

Then next stage is demodulation of OOK or FSK data.
A run of pulse/gap (OOK) or mark/space (FSK) timings is generated by the demod.
Run `rtl_433 -A SAMPLE.cu8` to get an overview of the timings,
or `rtl_433 -w OOK:- SAMPLE.cu8` to see the raw data.

Now build a flex decoder to slice the data into bits.
Use the suggestion or Sigrok's Pulseview with `rtl_433 -W out.sr SAMPLE.cu8` to make a guess on the coding.

The last stage is the protocol decoding from the bit data.
Build a table of codes and the expected sensor values to identify where the bytes are and what is contained.
Preferably put the codes and annotations in a BitBench.

pcruchet, 2020-02-10:
Hi, I cloned the latest version, and it works. THGR221 is recognized as THGR810
Thanks for your help.

closed 2020-02-10 by pcruchet

zuckschwerdt, 2020-02-10:
Is the THGR221 similar to the THGR810? Is it perhaps an alternative product or successor?
If they are supposed to be (nearly) the same and report the same, then I'll just make a comment noting: "also THGR221" :)

hackery, 2020-02-11:
I don't understand their naming conventions in general or lineage between devices. Their 2018 [compatibility matrix](http://global.oregonscientific.com/2018/Oregon_Scientific_IoT_Sensor_Compatibility/index.html) doesn't even mention the THGR810. Between similar sensors, all it indicates is which base stations they came/come with.
[FCC filings](https://fccid.io/NMT) show THGR810 as being from 2006, and the THGR211 from 2018, successor seems likely, but it's not clearly so.