merbanan/rtl_433 #1298: switch fallthrough warning in rtl_433/src/devices/m_bus.c

jcwren, 2020-02-06:
Got this warning compiling the latest pull from the master branch. Looks like there should be a `break` at line 226? Or perhaps a `__attribute__ ((fallthrough))`, if that's really what is intended.

This is with gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)

```
/home/zzz/rtl_433/src/devices/m_bus.c: In function ‘m_bus_decode_records’:
/home/zzz/rtl_433/src/devices/m_bus.c:219:13: warning: this statement may fall through [-Wimplicit-fallthrough=]
             switch(vif_uam>>1) {
             ^~~~~~
/home/zzz/rtl_433/src/devices/m_bus.c:226:9: note: here
         case 0x7D:
         ^~~~
```

merbanan, 2020-02-06:
Should be fixed now.

closed 2020-02-06 by merbanan

jcwren, 2020-02-06:
The `break` needs to go after the closing brace for the `case 0x7B:`. You added a default for the switch statement that's inside the `case 0x7B`.

jcwren, 2020-02-06:
That got it :)  Thanks!

merbanan, 2020-02-06:
Send a PR next time you need something to be done properly ;)