merbanan/rtl_433 #1301: rtl_433 detects protocol from device but not from saved file

jcwren, 2020-02-07:
This is probably a user error, but I'm not sure what I'm doing wrong :)  Trying to decode my Ford Edge keyfob that's on 902.37MHz. Running the analysis in real-time recommends using the flex decoder option, but running it on a saved file reports 'Not a clue'.

Running `rtl_433 -f 902370000 -A -S all` produces the following result:
```
$ rtl_433 -f 902370000 -A -S all
rtl_433 version 19.08-198-g68c267c branch master at 202002071208 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 121 out of 148 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-148 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 902.370MHz.
Allocating 15 zero-copy buffers
Detected FSK package    2020-02-07 12:36:21
Analyzing pulses...
Total count: 1200,  width: 9.67 ms              ( 9666 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]        (   0 S)
 [ 1] count:  621,  width:    2 us [2;2]        (   2 S)
 [ 2] count:  515,  width:    1 us [1;1]        (   1 S)
 [ 3] count:   47,  width:    3 us [3;3]        (   3 S)
 [ 4] count:   11,  width:    4 us [4;4]        (   4 S)
 [ 5] count:    3,  width:    5 us [5;5]        (   5 S)
 [ 6] count:    1,  width:    8 us [8;8]        (   8 S)
 [ 7] count:    1,  width:   10 us [10;10]      (  10 S)
Gap width distribution:
 [ 0] count:   77,  width:    3 us [3;3]        (   3 S)
 [ 1] count:  260,  width:    1 us [1;1]        (   1 S)
 [ 2] count:  495,  width:    2 us [2;2]        (   2 S)
 [ 3] count:  300,  width:    4 us [4;4]        (   4 S)
 [ 4] count:   29,  width:    7 us [6;8]        (   7 S)
 [ 5] count:    4,  width:    5 us [5;5]        (   5 S)
 [ 6] count:   17,  width:  239 us [210;244]    ( 239 S)
 [ 7] count:    5,  width:   27 us [24;32]      (  27 S)
 [ 8] count:    4,  width:   78 us [70;89]      (  78 S)
 [ 9] count:    1,  width:  111 us [111;111]    ( 111 S)
 [10] count:    2,  width:   41 us [41;42]      (  41 S)
 [11] count:    3,  width:   11 us [11;12]      (  11 S)
 [12] count:    2,  width:   17 us [16;19]      (  17 S)
Pulse period distribution:
 [ 0] count:  454,  width:    3 us [3;3]        (   3 S)
 [ 1] count:   56,  width:    2 us [2;2]        (   2 S)
 [ 2] count:  202,  width:    4 us [4;4]        (   4 S)
 [ 3] count:  420,  width:    5 us [5;6]        (   5 S)
 [ 4] count:   32,  width:    8 us [7;9]        (   8 S)
 [ 5] count:   17,  width:  240 us [212;246]    ( 240 S)
 [ 6] count:    5,  width:   28 us [25;33]      (  28 S)
 [ 7] count:    4,  width:   79 us [71;90]      (  79 S)
 [ 8] count:    1,  width:  112 us [112;112]    ( 112 S)
 [ 9] count:    2,  width:   43 us [43;43]      (  43 S)
 [10] count:    4,  width:   12 us [11;13]      (  12 S)
 [11] count:    2,  width:   18 us [17;20]      (  18 S)
Level estimates [high, low]:  15929,    257
RSSI: -0.1 dB SNR: 17.9 dB Noise: -18.0 dB
Frequency offsets [F1, F2]:       0,      0     (+0.0 kHz, +0.0 kHz)
Guessing modulation: Pulse Code Modulation (Not Return to Zero)
Attempting demodulation... short_width: 1, long_width: 1, reset_limit: 1024, sync_width: 0
Use a flex decoder with -X 'n=name,m=FSK_PCM,s=1,l=1,r=1024'

Detected FSK package    2020-02-07 12:36:21
Analyzing pulses...
Total count: 1200,  width: 9.79 ms              ( 9794 S)
Pulse width distribution:
[... 243 lines cut ...]
 [ 5] count:    9,  width:  220 us [200;248]    (  55 S)
 [ 6] count:    1,  width: 4128 us [4128;4128]  (1032 S)
 [ 7] count:    2,  width:   52 us [52;52]      (  13 S)
 [ 8] count:    5,  width:  136 us [120;160]    (  34 S)
 [ 9] count:    2,  width:   72 us [72;72]      (  18 S)
 [10] count:    1,  width:  292 us [292;292]    (  73 S)
 [11] count:    1,  width:  512 us [512;512]    ( 128 S)
Pulse period distribution:
 [ 0] count:   26,  width: 1620 us [1260;1964]  ( 405 S)
 [ 1] count:   28,  width: 2064 us [1776;2456]  ( 516 S)
 [ 2] count:   13,  width:  616 us [524;760]    ( 154 S)
 [ 3] count:    6,  width: 1096 us [948;1276]   ( 274 S)
 [ 4] count:   19,  width: 2768 us [2480;3304]  ( 692 S)
 [ 5] count:    6,  width: 3592 us [3208;3852]  ( 898 S)
 [ 6] count:    2,  width:  828 us [800;856]    ( 207 S)
 [ 7] count:    8,  width:  392 us [336;460]    (  98 S)
 [ 8] count:    1,  width: 5828 us [5828;5828]  (1457 S)
 [ 9] count:    1,  width:  112 us [112;112]    (  28 S)
 [10] count:    1,  width:  188 us [188;188]    (  47 S)
Level estimates [high, low]:   1523,    304
RSSI: -10.3 dB SNR: 7.0 dB Noise: -17.3 dB
Frequency offsets [F1, F2]:   -4027, -24818     (-15.4 kHz, -94.7 kHz)
Guessing modulation: No clue...
$ 
```

merbanan, 2020-02-07:
rtl_433 -f 902370000 -Y min_max -r gfile*.cu8 should make it work to replay

merbanan, 2020-02-07:
It is related to the new fsk code and defaults

merbanan, 2020-02-07:
So yes, user error because of no documentation by developers.

jcwren, 2020-02-07:
> So yes, user error because of no documentation by developers.

LOL. OK, I'll give that a try.

jcwren, 2020-02-07:
OK. `min_max` is actually `minmax`, but that now produces the same results as running it for-reals.

zuckschwerdt, 2020-02-07:
rtl_433 processes radio data in multiple stages.

First a radio data packet is found and framed.
Use CubicSDR or Gqrx to verify you receice a signal.
Note the frequency, pick a frequency a little off, e.g 50k above or below.
Then grab the signal with rtl_433, e.g. `rtl_433 -f 433.92M -S all`
Visually verify the samples in https://triq.net/iqs

To get a clean signal remove the receiver antenna and place the device at 10cm to the receiver, that mostly isolates the transmissions.

Then next stage is demodulation of OOK or FSK data.
A run of pulse/gap (OOK) or mark/space (FSK) timings is generated by the demod.
Run `rtl_433 -A SAMPLE.cu8` to get an overview of the timings,
or `rtl_433 -w OOK:- SAMPLE.cu8` to see the raw data.

Now build a flex decoder to slice the data into bits.
Use the suggestion or Sigrok's Pulseview with `rtl_433 -W out.sr SAMPLE.cu8` to make a guess on the coding.

The last stage is the protocol decoding from the bit data.
Build a table of codes and the expected sensor values to identify where the bytes are and what is contained.
Preferably put the codes and annotations in a BitBench.

merbanan, 2020-07-20:
Closing issue.

closed 2020-07-20 by merbanan