merbanan/rtl_433 #1308: MinGW strtod() issue

gvanem, 2020-02-09:
With a small patch to `src/optparse.c`:
```diff
a/src/optparse.c 2019-12-13 21:53:35
b/src/optparse.c 2020-02-09 00:19:25
@@ -307,6 +311,7 @@
     ASSERT_EQUALS(atouint32_metric("1.0", ""), 1);
     ASSERT_EQUALS(atouint32_metric("1.024k", ""), 1024);
     ASSERT_EQUALS(atouint32_metric("433.92MHz", ""), 433920000);
+    ASSERT_EQUALS(atouint32_metric("433.94M", ""), 433940000);
     ASSERT_EQUALS(atouint32_metric(" +1 G ", ""), 1000000000);
```

Running the test-program gives:
```
...
decimal fraction (1.000000) did you forget k, M, or G suffix?
FAIL: 433939999 <> 433940000
optparse:: atoi_time
optparse:: test (34/35) passed, (1) failed.
```

I see no failed test with MSVC or clang-cl. Only this:
```
optparse:: atoi_time
decimal fraction (-4294967355.000000) did you forget m, or h suffix?
optparse:: test (35/35) passed, (0) failed.
```

Using TDM-MinGW, gcc v. 5.1.0.

closed 2020-02-09 by zuckschwerdt

zuckschwerdt, 2020-02-09:
The check there wasn't too clever. Is it fixed now?

gvanem, 2020-02-09:
>  Is it fixed now?

No. For MinGW it's the same odd rounding-error. Okay for MSVC + clang-cl though.
Speaking of clang-cl, there are lots of these warnings in `optparse.c`:
```
src/optparse.c(308,48): warning: empty expression statement has no effect; remove unnecessary ';' to silence this warning
      [-Wextra-semi-stmt]
    ASSERT_EQUALS(atouint32_metric("0", ""), 0);
                                               ^
```

Fix:
```diff
--- a/src/optparse.c 2020-02-09 10:30:11
+++ b/src/optparse.c 2020-02-09 10:58:04
@@ -294,7 +294,16 @@

 // Unit testing
 #ifdef _TEST
-#define ASSERT_EQUALS(a,b) if ((a) == (b)) { ++passed; } else { ++failed; fprintf(stderr, "FAIL: %d <> %d\n", (a), (b)); }
+
+#define ASSERT_EQUALS(a,b) do {              \
+                             if ((a) == (b)) \
+                                ++passed;    \
+                              else {         \
+                               ++failed;     \
+                               fprintf(stderr, "FAIL: %d <> %d\n", (a), (b)); \
+                             } \
+                           } while (0)
+
 int main(int argc, char **argv)
 {
     unsigned passed = 0;
```

zuckschwerdt, 2020-02-10:
I was afraid of that. The check wasn't great to start, so at least that's fixed.
Now for the rounding. Can you test add `val += 1e-6;` before the test [`if (val - (uint32_t)val > 1e-6) {`](https://github.com/merbanan/rtl_433/blob/master/src/optparse.c#L140) and change to condition to `> 2e-6` or `> 1e-5` maybe?
The idea is that we want to ensure that after scaling there are "all" zeros after the decimal. So we need to trigger a rounding up ( `x.999999`->`x.000000` and `x.000000`->`x.000001`) then test with a slightly bigger value.
But I'm not sure where that rounding can happen for double as we likely already have 9-10 digits in the mantissa left of the decimal point.

Good find with the macros, the definition shouldn't have a closing semicolon, that should be in the line where it's used. Looks saner.

zuckschwerdt, 2020-02-10:
Ok, with 1dfa26b6 it should work now?

gvanem, 2020-02-10:
Works fine now. Thanks!

commit f6be17a9 (2020-02-09): Fix optparse strtod rounding (closes #1308)