merbanan/rtl_433 #1309: Undefined reference to write_sigrok when compiling on Linux using Autotools

colerd24, 2020-02-09:
Hello

It looks like a recent commit broke compiling on Linux using autotools (probably a forgotten include in a makefile or something). I believe write_sigrok.c needs to be added based on the error gotten when running make:
```
/usr/bin/ld: r_api.o: in function `close_dumpers':
/home/rc/RTL/rtl_433/rtl_433/src/r_api.c:906: undefined reference to `write_sigrok'
/usr/bin/ld: /home/rc/RTL/rtl_433/rtl_433/src/r_api.c:909: undefined reference to `open_pulseview'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:992: rtl_433] Error 1
make[2]: Leaving directory '/home/rc/RTL/rtl_433/rtl_433/src'
make[1]: *** [Makefile:496: all-recursive] Error 1
make[1]: Leaving directory '/home/rc/RTL/rtl_433/rtl_433'
make: *** [Makefile:382: all] Error 2

```
This is probably from the [commit yesterday](https://github.com/merbanan/rtl_433/commit/44d2af5e2793dd9eec45a40e90bdbdd461d73026) that adds the new sigrok writer. Looks like it had to be added to the Visual Studio project so I'm guessing it needs to be similarly added to the Linux build stuff.

I can confirm that it was a recent break because I was able to build from commit [232de06](https://github.com/merbanan/rtl_433/commit/232de06f71dcba0a4fe696e29d03b457a0cebd01) from Feb 5, 2020 just fine.

zuckschwerdt, 2020-02-09:
Thanks! I'll fix this shortly.
Could you give some feedback on the autotools build? Do you prefer it to CMake? Do you have CMake as alternative? Does it work? If not what platform are you on?

closed 2020-02-09 by zuckschwerdt

colerd24, 2020-02-09:
Honestly I've never heard of autotools until I tried building this lol. To me it seems similar to CMake in terms of end-user usage. I don't have CMake on my machine but did have autotools so I just went with that. I ran autotools on the latest commit and it worked without issue. I also installed CMake and it build just fine that way too. 

Running Manjaro with kernel 5.4 and everything seems like it is working great. Able to pick up my neighbor's weather stations! Next up is sniffing some TPMS sensors...

Thanks for the hard work!

commit b60af3b9 (2020-02-09): Fix Autotools build (closes #1309)