merbanan/rtl_433 #1310: endless loop in m_bus

Mindavi, 2020-02-09:
Hey, I've managed to trigger an endless loop in m_bus.
For this checksums/CRCs must be disabled, like I've done in https://github.com/Mindavi/rtl_433/tree/fuzzing-no-checksum.

The input `{163} 2a aa aa aa aa 1e ea a6 68 4a 39 69 64 21 49 bb 89 a0 62 f1 30` triggers the bug in that build.

It seems to take place during record decoding.
https://github.com/merbanan/rtl_433/blob/cbe1fb01979723551463760bf70e991fca066275/src/devices/m_bus.c#L203

the `m_bus_decode_records` function seems to return 0 indefinitely, causing a while loop to keep running.

https://github.com/merbanan/rtl_433/blob/cbe1fb01979723551463760bf70e991fca066275/src/devices/m_bus.c#L273-L319

Probably some extra sanity checking might help avoiding this issue, but I'm not sure what exactly is happening in that code.

merbanan, 2020-02-09:
Not sure how this could have triggered. I changed one parsing point that could have consumed 0 bytes. It's not really correct but we don't support any sensor that uses a dif_coding value of 0.

merbanan, 2020-02-09:
@Mindavi can you check and see if that fixes it?

Mindavi, 2020-02-10:
Wasn't sure what you were referring to before, but I see the new commits now.

The patch seems to work well, good stuff 👍 

```
rtl_433-fuzzing-no-checksum/build$ ./src/rtl_433 -y '{163} 2a aa aa aa aa 1e ea a6 68 4a 39 69 64 21 49 bb 89 a0 62 f1 30' -F json
rtl_433 version unknown inputs file rtl_tcp
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/ricsch/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 121 out of 148 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-148 ]
{"time" : "2020-02-10 10:00:27", "model" : "Wireless-MBus", "mode" : "T", "M" : "___", "id" : 166645865, "version" : 255, "type" : 0, "type_string" : "Other", "C" : 255, "data_length" : 256, "data" : "fffffffffff7fdffff00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000", "mic" : "CRC"}
```

closed 2020-02-10 by Mindavi