merbanan/rtl_433 #1312: Acurite TXR, 5n1, 6045 message integrity, add parity check.

rct, 2020-02-10:
Acurite TXR, 5n1, 6045 and possibly others like the new Atlas weather station use parity in addition to checksums for integrity checking. 

Currently the parity isn't verified, it's only looked at in verbose mode.

I've noticed I seem to get messages for non-existent devices from time to time. I suspect the message is mangled but the checksum still passes.  The parity check in addition to the checksum might reduce these.

Only the data bytes.  in the Acurite message have parity checks.  The ID bytes, and checksum byte are 8 bits of data with no parity.

(Note: I'm opening this as an issue to let others know. I will eventually fix this, but it will probably be a few weeks before I get some time and am near my Acurite gear for testing.)

ChrisM510, 2020-02-13:
Let me know if I can help... I might need detailed instructions on how to gather the info, but I have RTL_433 + SDR + WeeWX running on my Pi4.

jcwren, 2020-02-18:
If you need any data from an Acurite 6045M, I just acquired one. Just tell me what you need.

jcwren, 2020-02-18:
I see that the 6045M decoder has two unknown bits. One of these might indicate that the remote has gone into silent mode for 2 hours after 10 strike detections. Manual doesn't say anything about a time window on those 10 strikes. I'd hope they'd have to occur within a 5 or 10 minute period.

rct, 2020-02-18:
I haven't touched this in a while, but I need to get back to it.

I think I was down to 1 unknown bit - `ussb1`, and I believe I finally figured out is it really the LSB of the strike count bit, which changes that to 8 bits - wraps at 255 strikes, and matches what people were occasionally reporting about their consoles. 

I haven't seen a thunderstorm in months.

The 10 strikes count is just for stopping the unit from beeping.  I haven't seen anything that would indicate it needs to send that in the message.  There really isn't a reason to send that as any receiver could make that determination itself.

I've noticed via the `active` bit, that my sensor seems to cycle into active mode very regularly about once an hour.

jcwren, 2020-02-18:
Hrm, not sure about USSB1. I don't have any hits, but it's set. I do see the active bit flipping periodically. Here's the log to date where active has flipped (obviously I deleted all the lines where it was the same for some duration).

```
{"time":"2020-02-18 11:11:58","model":"Acurite-6045M","id":96,"channel":"A","battery_ok":1,"temperature_F":70.1,"humidity":42,"strike_count":0,"storm_dist":31,"active":0,"rfi":0,"ussb1":1,"exception":0,"raw_msg":"c0606faa1199005f42","mod":"ASK","freq":433.95091,"rssi":-0.122585,"snr":28.2195,"noise":-28.3421}
{"time":"2020-02-18 12:13:02","model":"Acurite-6045M","id":96,"channel":"A","battery_ok":1,"temperature_F":72.8,"humidity":39,"strike_count":0,"storm_dist":31,"active":1,"rfi":0,"ussb1":1,"exception":0,"raw_msg":"c0606f27d1b4005f9a","mod":"ASK","freq":433.95226,"rssi":-0.102726,"snr":30.58019,"noise":-30.6829}
{"time":"2020-02-18 12:42:54","model":"Acurite-6045M","id":96,"channel":"A","battery_ok":1,"temperature_F":72.3,"humidity":39,"strike_count":0,"storm_dist":31,"active":0,"rfi":0,"ussb1":1,"exception":0,"raw_msg":"c0606f2711af005fd5","mod":"ASK","freq":433.95312,"rssi":-0.129955,"snr":31.60032,"noise":-31.7303}
{"time":"2020-02-18 13:13:02","model":"Acurite-6045M","id":96,"channel":"A","battery_ok":1,"temperature_F":72.5,"humidity":39,"strike_count":0,"storm_dist":31,"active":1,"rfi":0,"ussb1":1,"exception":0,"raw_msg":"c0606f27d1b1005f97","mod":"ASK","freq":433.948,"rssi":-0.125042,"snr":27.70552,"noise":-27.8306}
{"time":"2020-02-18 13:42:53","model":"Acurite-6045M","id":96,"channel":"A","battery_ok":1,"temperature_F":72.5,"humidity":39,"strike_count":0,"storm_dist":31,"active":0,"rfi":0,"ussb1":1,"exception":0,"raw_msg":"c0606f2711b1005fd7","mod":"ASK","freq":433.95331,"rssi":-0.104084,"snr":25.41254,"noise":-25.5166}
{"time":"2020-02-18 14:13:02","model":"Acurite-6045M","id":96,"channel":"A","battery_ok":1,"temperature_F":72.7,"humidity":39,"strike_count":0,"storm_dist":31,"active":1,"rfi":0,"ussb1":1,"exception":0,"raw_msg":"c0606f27d133005f19","mod":"ASK","freq":433.95085,"rssi":-0.1049,"snr":29.029,"noise":-29.1339}
{"time":"2020-02-18 14:42:54","model":"Acurite-6045M","id":96,"channel":"A","battery_ok":1,"temperature_F":72.8,"humidity":39,"strike_count":0,"storm_dist":31,"active":0,"rfi":0,"ussb1":1,"exception":0,"raw_msg":"c0606f2711b4005fda","mod":"ASK","freq":433.94986,"rssi":-0.135693,"snr":30.86907,"noise":-31.0048}
```

rct, 2020-02-19:
@jcwren - ok that seems to match what I'm seeing, and supports the idea that the sensor shifts into active mode periodically even if nothing seems to be going on storm-wise.  

Since the sensor transmits so frequently when in active mode, it is responsible for these regular swings in my number of rtl_433 events processed.  IIRC, I've got rtl_433 set to dump stats every 10 minutes so those are 10 minute buckets.

[image]

Stormtracker017, 2020-06-04:
I have a node set up in Node Red that displays several 06045M. I noticed the decoded value for the temperature is about 1.8 F to low compared to reading on my console. The humidity matches the console. As for distances of lightning strikes, I don't have enough data to compare.  The console I'm using is a 06047M.

rct, 2020-06-04:
@Stormtracker017 I just checked my Acurite Access log and I see a difference of 1.7 F.  I'm very surprised. Please open a separate issue for that.  

Re: storm distance, I've captured many of the correlations to miles from the Acurite Access. I need to get a few more on the far end of the range.
 
Oh and the issue mentioned about the strike counter and ussb1 were fixed in #1348

Stormtracker017, 2020-06-04:
**@rct**  I'm using the latest that wraps around @ 255.  I will try to capture data for the far end distances, when storms come through.

rct, 2023-03-18:
This was addressed by #2162 a while back.  Only the 3n1 doesn't have parity checking because it wasn't clear whether that device used it based on the available information.

closed 2023-03-18 by rct