merbanan/rtl_433 #1314: Return codes for decoders using events patterns

Mindavi, 2020-02-10:
There are some decoders which try to decode every row they receive. How should these handle the return codes meant for showing what went wrong?

Might it be a good idea to only remember the last seen error code (if applicable), otherwise return amount of decoded signals?

- [ ] acurite
- [ ] brennenstuhl_rcs_2044
- [ ] gt_wt_02
- [ ] ht680
- [ ] oil_standard
- [ ] oregon_scientific
- [ ] tpms_ford
- [ ] tpms_pmv107j
- [ ] ttx201

merbanan, 2020-02-10:
We should convert them to bitbuffer_find_repeated_row().

merbanan, 2020-02-10:
But time/motivation etc. And then we have sensors that just take the first row by default. Those should be converted also (proove.c). Lots of janitor work to do, but maybe the current code already works good enough? So I'd rather spend my time on writing new decoders and better dsp tools. And changes might risk breaking things that work. And one has to remember that all these checks are done at every message reception. Some day there will not be enough energy in the universe to try everything.

zuckschwerdt, 2020-02-11:
For every data_make there should be one count in the return, even if processing multiple rows successfully the return should only account the data_make calls.
I guess for the looping decoders a DECODE_ABORT_LENGTH or DECODE_ABORT_EARLY should be good enough, expect if you known they all failed on DECODE_FAIL_MIC.
The TPMS are a good example where we possibly need a better approach so we can remove the common looping and just declare that we need a fixed preamble or something like that. I haven't found a nice solution so far though.

zuckschwerdt, 2020-02-11:
Also if the decoder function delegates to a subfunction it should have a doc comment of
```
/** @sa brennenstuhl_rcs_2044_process_row() */
```
(see [brennenstuhl_rcs_2044.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/brennenstuhl_rcs_2044.c#L100)).
That way we can find a way from the r_device to the actual doc comment automatically.

zuckschwerdt, 2020-02-11:
The obvious broken return code loops using events are fixed now.

zuckschwerdt, 2020-02-11:
Not all `return events;` are accounting wrongly, but's that's the line to audit for.

Mindavi, 2020-02-11:
Yeah, that's the solution I was thinking of. Looking good.

Mindavi, 2020-02-11:
The return codes for those calls need to be fixed still, but this at least solved the higher-up problem. Let's close this.

closed 2020-02-11 by Mindavi