merbanan/rtl_433 #1315: Bresser 5-in-1 wind issue (not displayed correctly)

mercurystorm, 2020-02-11:
I have been running the Bresser 5-in-1 for over a year

I have noticed that the wind speed displayed on the console is always higher than what is recorded in rtl_433

For example, the wind speed on console would be 7.2km/h but my rtl_433 stream would show 1.6ms (5.76km/h)

Any ideas where to even start?

mercurystorm, 2020-02-11:
`{"time" : "2020-02-11 01:15:41", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 278, "data" : "2aaaaaaaaa8b753a5d9ff5dff5bbf1bfdf9b7bbfc5a2600a200a440e40206484400000"}], "codes" : ["{278}2aaaaaaaaa8b753a5d9ff5dff5bbf1bfdf9b7bbfc5a2600a200a440e40206484400000"]}
{"time" : "2020-02-11 01:15:41", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 3.900, "humidity" : 81, "wind_max_m_s" : 2.800, "wind_avg_m_s" : 2.900, "wind_dir_deg" : 180.000, "rain_mm" : 19.200, "mic" : "CHECKSUM"}
{"time" : "2020-02-11 01:15:53", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 279, "data" : "955555555545ba9d2ecffd6dfb7df8dfefcdbddfe2d130029204820720103242200000"}], "codes" : ["{279}955555555545ba9d2ecffd6dfb7df8dfefcdbddfe2d130029204820720103242200000"]}
{"time" : "2020-02-11 01:15:53", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 3.900, "humidity" : 81, "wind_max_m_s" : 1.400, "wind_avg_m_s" : 2.400, "wind_dir_deg" : 202.500, "rain_mm" : 19.200, "mic" : "CHECKSUM"}
{"time" : "2020-02-11 01:16:05", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 278, "data" : "2aaaaaaaaa8b7539dd9fec8ff37bf1ffdf977bbfc6226013700c840e00206484400000"}], "codes" : ["{278}2aaaaaaaaa8b7539dd9fec8ff37bf1ffdf977bbfc6226013700c840e00206484400000"]}
{"time" : "2020-02-11 01:16:17", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 277, "data" : "d55555555516ea733b3fd91ff6f7e3dfff34f77f8cc4c026e019081c0040cb08800000"}], "codes" : ["{277}d55555555516ea733b3fd91ff6f7e3dfff34f77f8cc4c026e019081c0040cb08800000"]}
{"time" : "2020-02-11 01:16:29", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 276, "data" : "aaaaaaaaaa2dd4e7767f9c4fbfefc7ff7e69eeff18898063b04010380081961100000"}], "codes" : ["{276}aaaaaaaaaa2dd4e7767f9c4fbfefc7ff7e69eeff18898063b04010380081961100000"]}
{"time" : "2020-02-11 01:16:29", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 3.800, "humidity" : 81, "wind_max_m_s" : 6.300, "wind_avg_m_s" : 4.000, "wind_dir_deg" : 247.500, "rain_mm" : 19.600, "mic" : "CHECKSUM"}
{"time" : "2020-02-11 01:16:41", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 276, "data" : "aaaaaaaaaa2dd4e6767f7b2faaefc7ff7e69eeff19898084d05510380081961100000"}], "codes" : ["{276}aaaaaaaaaa2dd4e6767f7b2faaefc7ff7e69eeff19898084d05510380081961100000"]}
{"time" : "2020-02-11 01:16:41", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 3.800, "humidity" : 81, "wind_max_m_s" : 8.400, "wind_avg_m_s" : 5.500, "wind_dir_deg" : 292.500, "rain_mm" : 19.600, "mic" : "CHECKSUM"}
{"time" : "2020-02-11 01:16:53", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 278, "data" : "6aaaaaaaaa8b753a1d9ff36be9fbf1ffdfda7bbfc5e2600c9416040e00202584400000"}], "codes" : ["{278}6aaaaaaaaa8b753a1d9ff36be9fbf1ffdfda7bbfc5e2600c9416040e00202584400000"]}
{"time" : "2020-02-11 01:17:05", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 278, "data" : "2aaaaaaaaa8b753a1d9ff36be9f7f1ffdfda7bbfc5e2600c9416040e00202584400000"}], "codes" : ["{278}2aaaaaaaaa8b753a1d9ff36be9f7f1ffdfda7bbfc5e2600c9416040e00202584400000"]}
{"time" : "2020-02-11 01:17:17", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 277, "data" : "555555555516ea73bb3fdec7d677e3ffbfb4f77f8c44c0213829881c00404b08800000"}], "codes" : ["{277}555555555516ea73bb3fdec7d677e3ffbfb4f77f8c44c0213829881c00404b08800000"]}
{"time" : "2020-02-11 01:17:17", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 3.800, "humidity" : 80, "wind_max_m_s" : 4.200, "wind_avg_m_s" : 5.300, "wind_dir_deg" : 157.500, "rain_mm" : 19.600, "mic" : "CHECKSUM"}`

mercurystorm, 2020-02-11:
[Bresser5in1.zip](https://github.com/merbanan/rtl_433/files/4183792/Bresser5in1.zip)

Capture files using: rtl_433 -S all -f 868.3M

zuckschwerdt, 2020-02-11:
Does the console show gust and average? Perhaps it relates to both? Can you check that?
The actual values are transmitted as plain BCD, so we didn't apply any conversion. E.g. in one of your examples above:
```
aaaaaaaaaa2dd4 e7767f9c4fbfefc7ff7e69eeff 18898 *063 b *040* 10380081961100000
Wind Gust : 6.3 m/s       Wind Speed: 4.0 m/s
```
see also: https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_5in1.c#L21

mercurystorm, 2020-02-11:
The console shows both gust and average - I watched the console for a few packets and recorded the updates

{"time" : "2020-02-11 09:26:53", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 276, "data" : "d55555555516ea733b3fd91fec77ec7fbd7df6ff8cc4c026e01388138042820900000"}], "codes" : ["{276}d55555555516ea733b3fd91fec77ec7fbd7df6ff8cc4c026e01388138042820900000"]}
{"time" : "2020-02-11 09:26:53", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 2.700, "humidity" : 85, "wind_max_m_s" : 5.300, "wind_avg_m_s" : 2.700, "wind_dir_deg" : 270.000, "rain_mm" : 20.400, "mic" : "CHECKSUM"}

Console: Wind Average - 9.7km/h
Console: Wind Gust - 27.7km/h

{"time" : "2020-02-11 09:27:05", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 276, "data" : "555555555516ea74bb3fe6efeb77ec7fbd7df6ff8b44c019101488138042820900000"}], "codes" : ["{276}555555555516ea74bb3fe6efeb77ec7fbd7df6ff8b44c019101488138042820900000"]}
{"time" : "2020-02-11 09:27:05", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 2.700, "humidity" : 85, "wind_max_m_s" : 3.200, "wind_avg_m_s" : 2.900, "wind_dir_deg" : 45.000, "rain_mm" : 20.400, "mic" : "CHECKSUM"}

Console: Wind Average - 10.4km/h
Console: Wind Gust -  18km/h

{"time" : "2020-02-11 09:27:17", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 276, "data" : "555555555516ea743b3ff0f7ecf7ec7fbcfdf6ff8bc4c00f081308138043020900000"}], "codes" : ["{276}555555555516ea743b3ff0f7ecf7ec7fbcfdf6ff8bc4c00f081308138043020900000"]}
{"time" : "2020-02-11 09:27:17", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 2.700, "humidity" : 86, "wind_max_m_s" : 2.400, "wind_avg_m_s" : 2.600, "wind_dir_deg" : 22.500, "rain_mm" : 20.400, "mic" : "CHECKSUM"}

Console: Wind Average - 9.4km/h
Console: Wind Gust - 10.8km/h

{"time" : "2020-02-11 09:27:29", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 276, "data" : "555555555516ea743b3ff0f7ecf7ec7fbcfdf6ff8bc4c00f081308138043020900000"}], "codes" : ["{276}555555555516ea743b3ff0f7ecf7ec7fbcfdf6ff8bc4c00f081308138043020900000"]}
{"time" : "2020-02-11 09:27:29", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 2.700, "humidity" : 86, "wind_max_m_s" : 2.400, "wind_avg_m_s" : 2.600, "wind_dir_deg" : 22.500, "rain_mm" : 20.400, "mic" : "CHECKSUM"}

Console: Wind Average - 9.4km/h
Console: Wind Gust - 10.8km/h

{"time" : "2020-02-11 09:27:41", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 278, "data" : "955555555545ba9d2ecffc3bfb7dfb1fef3f7dbfe2d13003c4048204e010c082400000"}], "codes" : ["{278}955555555545ba9d2ecffc3bfb7dfb1fef3f7dbfe2d13003c4048204e010c082400000"]}
{"time" : "2020-02-11 09:27:41", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 2.700, "humidity" : 86, "wind_max_m_s" : 2.400, "wind_avg_m_s" : 2.400, "wind_dir_deg" : 45.000, "rain_mm" : 20.400, "mic" : "CHECKSUM"}

Console: Wind Average - 8.6km/h
Console: Wind Gust - 10.8km/h

{"time" : "2020-02-11 09:27:53", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 277, "data" : "2aaaaaaaaa8b753a5d9ff35bf77bf63fde7efb7fc5a2600ca4088409c0218104800000"}], "codes" : ["{277}2aaaaaaaaa8b753a5d9ff35bf77bf63fde7efb7fc5a2600ca4088409c0218104800000"]}
{"time" : "2020-02-11 09:27:53", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 2.700, "humidity" : 86, "wind_max_m_s" : 3.200, "wind_avg_m_s" : 2.200, "wind_dir_deg" : 202.500, "rain_mm" : 20.400, "mic" : "CHECKSUM"}

Console: Wind Average - 7.9km/h
Console: Wind Gust -  18.0km/h

mercurystorm, 2020-02-11:
I changed the console to report in meters per second to stop conversations needing to be done.. it seems the average is always correct but the gusts are incorrect

{"time" : "2020-02-11 09:55:05", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 276, "data" : "d55555555516ea73bb3ff0afec77ee7fbd7bf6ff8c44c00f501388118042840900000"}], "codes" : ["{276}d55555555516ea73bb3ff0afec77ee7fbd7bf6ff8c44c00f501388118042840900000"]}
{"time" : "2020-02-11 09:55:05", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 2.300, "humidity" : 85, "wind_max_m_s" : 2.400, "wind_avg_m_s" : 2.700, "wind_dir_deg" : 225.000, "rain_mm" : 20.800, "mic" : "CHECKSUM"}
Console: Wind Average - 2.7m/s
Console: Wind Gust - 3.0m/s

{"time" : "2020-02-11 09:55:17", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 163, "data" : "aaaaaaaaaa2dd4e7767fe15fd8efdcff7af7edff0"}], "codes" : ["{163}aaaaaaaaaa2dd4e7767fe15fd8efdcff7af7edff0"]}
Console: Wind Average - 2.7m/s
Console: Wind Gust - 3.0m/s

{"time" : "2020-02-11 09:55:29", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 276, "data" : "555555555516ea753b3ff5a7ef77ee7fbd7bf6ff8ac4c00a581088118042840900000"}], "codes" : ["{276}555555555516ea753b3ff5a7ef77ee7fbd7bf6ff8ac4c00a581088118042840900000"]}
{"time" : "2020-02-11 09:55:29", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 2.300, "humidity" : 85, "wind_max_m_s" : 1.400, "wind_avg_m_s" : 2.100, "wind_dir_deg" : 247.500, "rain_mm" : 20.800, "mic" : "CHECKSUM"}
Console: Wind Average - 2.1m/s
Console: Wind Gust -  2.0m/s

{"time" : "2020-02-11 09:55:41", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 276, "data" : "555555555516ba743b3ff097f377eeffbd7bf6ff8bc4c00f680c88110042840900000"}], "codes" : ["{276}555555555516ba743b3ff097f377eeffbd7bf6ff8bc4c00f680c88110042840900000"]}
Console: Wind Average - 1.9m/s
Console: Wind Gust - 3.0m/s

{"time" : "2020-02-11 09:55:53", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 15, "data" : "5554"}], "codes" : ["{15}5554"]}
Console: Wind Average - 1.9m/s
Console: Wind Gust - 3.0m/s

{"time" : "2020-02-11 09:56:05", "model" : "Bresser-5in1", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 276, "data" : "555555555516ea75bb3ff5cff3f7ee7fbd7bf6ff8a44c00a300c08118042840900000"}], "codes" : ["{276}555555555516ea75bb3ff5cff3f7ee7fbd7bf6ff8a44c00a300c08118042840900000"]}
{"time" : "2020-02-11 09:56:05", "model" : "Bresser-5in1", "id" : 137, "temperature_C" : 2.300, "humidity" : 85, "wind_max_m_s" : 1.400, "wind_avg_m_s" : 1.800, "wind_dir_deg" : 135.000, "rain_mm" : 20.800, "mic" : "CHECKSUM"}
Console: Wind Average - 1.8m/s
Console: Wind Gust - 2.0m/s

zuckschwerdt, 2020-02-11:
I noticed the same from your data :)
I remember some other user reporting that the display unit collects the gust values for a certain time and reports something like "running maximum for the last hour". Not sure which issue or sensor that was, sorry.
If you have your values stored (InfluxDB, CSV table, …) you can try a few aggregations and see what matches?

mercurystorm, 2020-02-11:
I have all the data that is received by rtl_433 stored in a MySQL database (as part of WeeWX) but I do not have the values from the console as this does not have any usb or anything to access the data

The console gust speed is definitely updating in the same frequency (every 12 seconds) as the message so I don't believe its an form of running maximum but is in fact a real time update.. just wish I understood what field it was reading in the message as its certainly different

mercurystorm, 2020-02-11:
The Bresser Manual indicates: 
• AVERAGE: average of all wind speed numbers recorded in the previous 30 seconds
• GUST: highest wind speed (gust) recorded from last reading

merbanan, 2020-02-11:
"wind_max_m_s" : 2.400 vs Wind Gust - 3.0m/s
"wind_max_m_s" : 1.400 vs Wind Gust - 2.0m/s

Is there a bias of 0.6 ?

mercurystorm, 2020-02-11:
> "wind_max_m_s" : 2.400 vs Wind Gust - 3.0m/s
> "wind_max_m_s" : 1.400 vs Wind Gust - 2.0m/s
> 
> Is there a bias of 0.6 ?

It would seem to be a bias of 0.6.. but I would need to take more captures to confirm this

Also would have no idea where to address that! :)

merbanan, 2020-02-11:
Give us conclusive evidence and we'll fix it. It might be linear around 1.4-2.4 but not elsewhere.

mercurystorm, 2020-02-11:
Ok so its a pretty windy today (it always is in the West of Ireland and this is what I am seeing after monitoring the console and the rtl messages and this is what I found

the lower the wind the smaller the gap (0.8m/s reflects on rtl and console)
The higher the gust the larger the cap (7.4m/s vs 11.0m/s)

"wind_max_m_s" : 2.8m/s vs Wind Gust - 4.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 2.8m/s vs Wind Gust - 4.0m/s
"wind_max_m_s" : 2.8m/s vs Wind Gust - 4.0m/s
"wind_max_m_s" : 4.2m/s vs Wind Gust - 6.6m/s
"wind_max_m_s" : 2.8m/s vs Wind Gust - 4.0m/s
"wind_max_m_s" : 7.4m/s vs Wind Gust - 11.0m/s
"wind_max_m_s" : 2.4m/s vs Wind Gust - 3.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 2.8m/s vs Wind Gust - 4.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 5.3m/s vs Wind Gust - 7.7m/s
"wind_max_m_s" : 2.4m/s vs Wind Gust - 3.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 2.4m/s vs Wind Gust - 3.0m/s
"wind_max_m_s" : 4.2m/s vs Wind Gust - 6.6m/s
"wind_max_m_s" : 5.3m/s vs Wind Gust - 7.7m/s
"wind_max_m_s" : 2.4m/s vs Wind Gust - 3.0m/s
"wind_max_m_s" : 5.3m/s vs Wind Gust - 7.7m/s
"wind_max_m_s" : 2.4m/s vs Wind Gust - 3.0m/s
"wind_max_m_s" : 2.4m/s vs Wind Gust - 3.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 2.4m/s vs Wind Gust - 3.0m/s
"wind_max_m_s" : 2.4m/s vs Wind Gust - 3.0m/s
"wind_max_m_s" : 5.3m/s vs Wind Gust - 7.7m/s
"wind_max_m_s" : 2.8m/s vs Wind Gust - 4.0m/s
"wind_max_m_s" : 5.3m/s vs Wind Gust - 7.7m/s
"wind_max_m_s" : 0.8m/s vs Wind Gust - 0.8m/s
"wind_max_m_s" : 4.2m/s vs Wind Gust - 6.6m/s
"wind_max_m_s" : 4.2m/s vs Wind Gust - 6.6m/s
"wind_max_m_s" : 4.2m/s vs Wind Gust - 6.6m/s
"wind_max_m_s" : 2.8m/s vs Wind Gust - 4.0m/s
"wind_max_m_s" : 3.2m/s vs Wind Gust - 5.0m/s
"wind_max_m_s" : 3.2m/s vs Wind Gust - 5.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 2.4m/s vs Wind Gust - 3.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 2.4m/s vs Wind Gust - 3.0m/s
"wind_max_m_s" : 3.2m/s vs Wind Gust - 5.0m/s
"wind_max_m_s" : 3.2m/s vs Wind Gust - 5.0m/s
"wind_max_m_s" : 3.2m/s vs Wind Gust - 5.0m/s
"wind_max_m_s" : 4.2m/s vs Wind Gust - 6.6m/

zuckschwerdt, 2020-02-11:
Are they maybe adding a fraction of the average?

merbanan, 2020-02-11:
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s
"wind_max_m_s" : 1.4m/s vs Wind Gust - 2.0m/s

@zuckschwerdt looks like a 1-1 mapping

zuckschwerdt, 2020-02-11:
Yes, looks linear.
[image]

merbanan, 2020-02-11:
https://docs.google.com/spreadsheets/d/1tZZ7-lvNSfy3xSUim5qb4nz2n5juvkOVl1SGgV-dpWo/edit?usp=sharing

Looks like factor + offset with rounding errors because of fixed point math.

merbanan, 2020-02-11:
@zuckschwerdt few seconds faster :)

zuckschwerdt, 2020-02-11:
:) Doing it right takes time. I just copied the picture from the first "online linear regression" thing I found.

merbanan, 2020-02-11:
And the values are somewhat linear. Can't see a formula for it. The slope is around 1.4-2, can't really get this to a rounding/quantization error. Can we get some more

mercurystorm, 2020-02-11:
Sure, ill sit with the console and record shortly 👍

merbanan, 2020-02-11:
Anyway, to support this I think we need a table. And when you try to lookup values not present we should do linear interpolation between the values we do have.

mercurystorm, 2020-02-11:
Ok so after 30 minutes of staring at the console these are the unique values that came up.. the rest are duplicates

1.4 | 2.0
2.4 | 3.0
2.8 | 4.0
3.2 | 5.0
4.2 | 6.6
5.3 | 7.7
5.8 | 8.8
6.3 | 9.9
7.4 | 11.0

mercurystorm, 2020-02-11:
And looking at my earlier post they same to be the same as previously captured... dope!

merbanan, 2020-02-11:
2 new values. Anyway, there actually looks to be a pattern. But I think we will just go with a table and linear interpolation.

zuckschwerdt, 2020-02-11:
I wonder if this is correction for a hardware flaw, "burst" factor, or some kind of conversion to "wind experience".
If you rotated the cups steadily, would both avg and max be the same?

merbanan, 2020-02-11:
Could be that they used same hardware transmitter but different wind catcher.

merbanan, 2020-02-11:
"Lets just fix that in the receiver"

mercurystorm, 2020-02-11:
On a less windy and snowy day I can definitely try that

mercurystorm, 2020-02-13:
So looking at the data today (less windy) it seems that gusts are only reported in specific speeds.. for example 0.0, 0.8, 1.4 and 2.4.. seems to have nothing inbetween those values

While wind average seems to include all the values

Column3 | Column5 | Column6
-- | -- | --
'windDir':   337.5 | 'windSpeed': 0.0 | 'windGust': 0.0
'windDir': 337.5 | 'windSpeed': 0.0 | 'windGust': 0.0
'windDir':   337.5 | 'windSpeed': 0.2 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.2 | 'windGust': 0.0
'windDir':   337.5 | 'windSpeed': 0.2 | 'windGust': 0.0
'windDir': 337.5 | 'windSpeed': 0.4 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.4 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.4 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.7 | 'windGust': 1.4
'windDir': 337.5 | 'windSpeed': 0.7 | 'windGust': 1.4
'windDir':   337.5 | 'windSpeed': 0.9 | 'windGust': 1.4
'windDir': 337.5 | 'windSpeed': 0.9 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.9 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 1.2 | 'windGust': 2.4
'windDir':   337.5 | 'windSpeed': 1.2 | 'windGust': 1.4
'windDir': 337.5 | 'windSpeed': 1.2 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 1.2 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 1.2 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.9 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 1.4
'windDir': 337.5 | 'windSpeed': 0.8 | 'windGust': 1.4
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 1.4
'windDir': 337.5 | 'windSpeed': 1.1 | 'windGust': 1.4
'windDir':   337.5 | 'windSpeed': 1.1 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 1.1 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 1.0 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 1.0 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 1.4
'windDir': 337.5 | 'windSpeed': 0.8 | 'windGust': 1.4
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.9 | 'windGust': 1.4
'windDir':   337.5 | 'windSpeed': 0.9 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.9 | 'windGust': 1.4
'windDir':   337.5 | 'windSpeed': 0.9 | 'windGust': 1.4
'windDir': 337.5 | 'windSpeed': 0.9 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.8 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.6 | 'windGust': 0.0
'windDir':   337.5 | 'windSpeed': 0.6 | 'windGust': 0.0
'windDir': 337.5 | 'windSpeed': 0.7 | 'windGust': 1.4
'windDir':   337.5 | 'windSpeed': 0.7 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.7 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.7 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 1.1 | 'windGust': 1.4
'windDir':   337.5 | 'windSpeed': 1.2 | 'windGust': 1.4
'windDir': 337.5 | 'windSpeed': 1.2 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 1.2 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 1.0 | 'windGust': 0.0
'windDir':   337.5 | 'windSpeed': 0.7 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.4 | 'windGust': 0.0
'windDir':   337.5 | 'windSpeed': 0.4 | 'windGust': 0.8
'windDir': 337.5 | 'windSpeed': 0.6 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.4 | 'windGust': 0.0
'windDir': 337.5 | 'windSpeed': 0.4 | 'windGust': 0.0
'windDir':   337.5 | 'windSpeed': 0.2 | 'windGust': 0.0
'windDir': 337.5 | 'windSpeed': 0.2 | 'windGust': 0.8
'windDir':   337.5 | 'windSpeed': 0.2 | 'windGust': 0.0
'windDir': 337.5 | 'windSpeed': 0.4 | 'windGust': 0.8

peteakalad, 2020-04-15:
Have just purchased same weather station and have noticed the same.

Analysed some of the figures and noticed the difference is always a factor of 0.6.  The formula to work out the number of 0.6s to add I havent been able to work out.

[image]

peteakalad, 2020-04-15:
Think I've worked it out, certainly for lower values, need to generate some higher value packets to verify.

So break the gust section up into 3 bit sections, then its multiply places by 8 / 16 / 32 / 64.

e.g. for bresser readout of 3.0m/s gust, msg[15] =0x80 and msg[16]=0x32

Split up into 3 bits thats  1(ignore?) 000 000 000 110 010 

Assuming set of bits represent multiples of 8 i.e. 
```
*64  *32  *16   *8   *1
000 000 000 110 010 
```
Works out as 6*8 + 2*1 = 50 

Then just multiply by 0.1 to get m/s -> 5.0m/s

This is working for all of my captured lower gust value packets.

Will try altering the bresser_5in1.c in line with this and do some more testing with higher values.

mercurystorm, 2020-04-15:
> Think I've worked it out, certainly for lower values, need to generate some higher value packets to verify.
> 
> So break the gust section up into 3 bit sections, then its multiply places by 8 / 16 / 32 / 64.
> 
> e.g. for bresser readout of 5.0m/s gust, msg[15] =0x80 and msg[16]=0x32
> 
> Split up into 3 bits thats 1(ignore?) 000 000 000 110 010
> 
> Assuming set of bits represent multiples of 8 i.e.
> 
> ```
> *64  *32  *16   *8   *1
> 000 000 000 110 010 
> ```
> 
> Works out as (6 x 8) + (2 x 1) = 50
> 
> Then just multiply by 0.1 to get m/s -> 5.0m/s
> 
> This is working for all of my captured lower gust value packets.
> 
> Will try altering the bresser_5in1.c in line with this and do some more testing with higher values.

That is a great breakthrough! I am happy to test on my system when you ready :) (its not a very windy day today!)

zuckschwerdt, 2020-04-15:
> So break the gust section up into 3 bit sections, then its multiply places by 8 / 16 / 32 / 64.

That's just binary, right? 0x000000000110010 = 50 anyway.

peteakalad, 2020-04-15:
Yes seeing that now, seems I over complicated it.  Doing some more testing and will PR a fork.  Need to find a fan :)

peteakalad, 2020-04-15:
Tested 0 to 11m/s with exact same readings as my bresser reciever.

Created fork with fix and raised PR.

@mercurystorm feel free to test my fork, fix for this is the only commit.
https://github.com/peteakalad/rtl_433

zuckschwerdt, 2020-04-15:
Thanks for the PR. Are we sure this applies to all 5-in-1, not just some clone?
It seems strange that Wind and Temp are BCD but Gust ist not?

peteakalad, 2020-04-15:
It was this Bresser 5 in 1 station I purchased this week, states "Sold by Bresser UK and Fulfilled by Amazon." so shouldnt be a clone.
https://www.amazon.co.uk/Bresser-Weather-Station-Control-German/dp/B0170A47DI

Agree it does seem strange.  At lower speeds i.e. values that fit in 8bits both gust and average are correct using the different calculation methods in the PR.  

I've since done some further testing with a leaf blower for higher speeds (my wife thinks I'm mad) and it looks like the bits roll over > 255 to elsewhere in the packet, trying to implement a fix for that now, will update with a new PR if I fix it.

mercurystorm, 2020-04-15:
> It was this Bresser 5 in 1 station I purchased this week, states "Sold by Bresser UK and Fulfilled by Amazon." so shouldnt be a clone. [...]

Its working great on my side with your current PR... the weather station and my weewx is showing the same data now! THANK YOU!!

peteakalad, 2020-04-15:
@mercurystorm Await futher update, for large wind values its broken and always has been for both average and gust wind speeds.  The wrong most significant bit was being taken, have prepared/tested a fix for this, just raising the PR now.

@zuckschwerdt I think since this is fixing both my new station and mercurystorms > 1 year old station that this will apply to all Bresser 5in1s.

zuckschwerdt, 2020-04-15:
It should be fairly easy to spot which of the fields are BCD and which use binary. Log a good amount of packets and inspect them to see if there are (nibble = 4 bit) values bigger than `9`. I.e. BCD goes 0x01 to 0x09 then 0x10. We'd never see a-f in any nibble.

peteakalad, 2020-04-15:
Updated https://github.com/peteakalad/rtl_433 with the fixes for average and gust wind speeds.
New PR raised.

Fixes;
- Wind Gust Fix #1315 Change to use normal binary conversion for gust (for some reason inconsistent with average wind speed).
- Fix for both Gust and Average wind speeds for large values. Wrong MSB was taken for both. MSB taken from bits after LSB like other metrics - confirmed working.
- Update initial comment to correct understanding of bits

closed 2020-05-14 by zuckschwerdt