merbanan/rtl_433 #1317: Battery level not received

scarpy71, 2020-02-14:
Hi!
I have a Bresser 7002510 Weather Station, working with this command:
/usr/local/bin/rtl_433 -f 868.3M -R 119 -M newmodel -F "mqtt://localhost:1883,retain=0,devices=sensors/rtl_433/"

Everything is fine except battery level, never received.
The base station has a Remote battery status icon, so it should be sent but nothing..

Thanks for any help,
Alessandro

merbanan, 2020-02-14:
Hi, the Bresser 5in1 code doesn't have that parameter. If you are willing to help out we can try to add it.

To help out you need to find a flat battery that triggers the low battery indicator and record signals with and without. If you can then reopen the ticket and I'll help out with the needed commands.

closed 2020-02-14 by merbanan

scarpy71, 2020-02-14:
Thank you, Benjamin!
I'd like very much to help!
My weather station is on the roof...  I'm going to go up there in a few
days..
In the meanwhile, I'll try to record data sent by wether station with
charged batteries..

Thanks again..

merbanan, 2020-02-14:
To get good samples signals use the command rtl_433 -S known. Make sure you correlate the correct file with the correct reading. You can use rtl_433 -r "file name" to play back the recording. When you have the files, archive them together and upload them to the ticket. Try to get the same value readings with flat and non flat battery.

scarpy71, 2020-02-14:
> Try to get the same value readings with flat and non flat battery.

For this I'll take the station back at home, Otherwise wind direction and speed may change....

zuckschwerdt, 2020-02-14:
That would be great! There are a few bits/bytes unknown in this protocol:
```
CC CC CC CC CC CC CC CC CC CC CC CC CC uu II ?G GG DW WW ?? TT ?T HH RR ?R ?t

C = Check, inverted data of 13 byte further
u = unknown (data changes from packet to packet, but meaning is still unknown)
? = unknown (mostly 0's)
I = station ID (maybe)
G = wind gust in 1/10 m/s, BCD coded, GGG = 123 => 12.3 m/s
D = wind direction 0..F = N..NNE..E..S..W..NNW
W = wind speed in 1/10 m/s, BCD coded, WWW = 123 => 12.3 m/s
T = temperature in 1/10 °C, BCD coded, TTxT = 1203 => 31.2 °C
t = temperature sign, minus if unequal 0
H = humidity in percent, BCD coded, HH = 23 => 23 %
R = rain in mm, BCD coded, RRxR = 1203 => 31.2 mm
```

peteakalad, 2020-04-16:
Battery level added.  See https://github.com/merbanan/rtl_433/issues/1323