merbanan/rtl_433 #1318: Classify DSC Security Devices?

jcwren, 2020-02-15:
I have 6 DSC WS4939 key fobs and 2 DSC DW4917 door/window sensors. I notice that the ESNs for the  WS4939 start with `0x68` and the DW4917's start with `0x21`. Is there enough information out there where the decoder could classify the device type based on the ESN? 

Doing so would allow the 4 buttons on the key fobs to be identified as `button_1`, `button_2`, `button_3` and `button_4` instead of `tamper`, `xtamper1`, `xtamper2`, and `exception`. For the DW4917, which only has the single internal reed switch, it could be identified as `contact` or some-such.

If there's concern about breakage, maybe just the device name could be added to the JSON, or the new names added in addition to the existing ones.

zuckschwerdt, 2020-02-16:
Yes, Type and ESN are together displayed as `id`. The fist nibble (character) is the type, the remaining five the ESN.
Could you start to build a table what the types are? E.g.

Type | Device | kind
--- | --- | ---
6 |  WS4939 | key fob
2 | DW4917 | door/window sensor

Can other DSC Security owners perhaps join in and list type and device info?

zuckschwerdt, 2020-02-16:
I've added some preliminary type output. Please add more types and other findings here.

rct, 2020-02-16:
I see you haven't changed ESN which is good.  

I just wanted to be on record saying, please don't change the definition of the ESN and/or ID. The DSC system uses a 6 digit(*) ESN as the device ID. That's what's printed on the devices, that's how they get enrolled in the system. etc. So the type is an inherent part of the ESN/ID.

Here's what I have for types.  The type is more of a class, then a map to a specific device:

| Type | Description |
| ---- | ------------ |
| 0      | unknown |
| 1    | unknown |
| 2    | Door and window contacts, also Flood sensor, and possibly others |
| 3    | Motion sensor, Glass break detector, and possibly other burglary sensors |
| 4   | Smoke detector, and possibly other fire sensors |
| 5   | unknown |
| 6   | keyfob, possibly only 1-way keyfobs |
| 7   | unknown |
| 8   | Carbon Dioxide detector |
| 9   | unknown |

A few things to note:

1. I didn't think anyone was using the keyfob but me. I felt it was sufficient to let the downstream consumer apply the meanings. From what I've seen so far the 1-way keyfob is the only thing that doesn't use the same open/closed bit field as the other sensors.  The type nybble is useful for distinguishing the keyfob. I don't think type is really useful for much else, See below:
2. AFAICT, the alarm panel itself doesn't seem to rely on the type for anything. Instead when a sensor is programmed, there is also a type that gets programmed: entry with delay, immediate alarm, fire, water. etc.
4. The WS4945 (one of the type 2's) is also used as a generic Normally Closed (NC) sensor.  It has screw terminals inside so it can be used for anything like heat/rate of rise non-smoke based fire sensors. It's also used with a thermostat for temperature threshold alarms.
5.  The WS4985 flood sensor is a Normally Open (NO) version of the WS4945.  **It is still a type 2, so type can't be used to distinguish it as an environmental sensor**.  It also has screw terminals so can be used as a generic NO sensor.
1. DSC also uses 8 digit ESNs for two-way devices like the wireless keypads, and the two-way keyfobs. These use a different coding, probably with encryption, that rtl_433 doesn't handle yet. (while those are still 433mhz I think higher sample rate like 1024k is needed to reliably recover the signal.)

rct, 2020-02-16:
Here's what I have for the WS4939 button mappings:

| Status | Description |
| ---- | ------ |
| 0x81 | Button 1 - Arm Stay, top left
| 0x87 | Button 2 - Arm Away, top right, walking away icon
| 0x82 | Button 3 - Disarm (unlock icon), lower left
| 0x80 | Button 4 - Panic (red bell icon), lower right, needs to be pressed for 3 seconds

zuckschwerdt, 2020-02-16:
I've updated the Subtype list with proposed strings above. If the output is ambiguous or just a (poor) duplication of info that should be kept some where else (your monitoring panel) we'll probably remove it again. For now consider it just a debugging help.

rct, 2020-02-16:
What would probably have the most value, would be if sub_type == 6, to parse the status byte differently, so you get the button presses instead of interpreting it as a sensor.

Do you want that in rtl_433?

zuckschwerdt, 2020-02-16:
Sure, good idea. The `subtype` key is used in other decoders in a similar way. E.g. to show different or optional functions of the device.

jcwren, 2020-02-18:
> Here's what I have for the WS4939 button mappings:
> 
> Status	Description
> 0x81	Button 1 - Arm Stay, top left
> 0x87	Button 2 - Arm Away, top right, walking away icon
> 0x82	Button 3 - Disarm (unlock icon), lower left
> 0x80	Button 4 - Panic (red bell icon), lower right, needs to be pressed for 3 seconds

Just as an FYI for anyone else that comes across this issue (who reads closed issues, right?) -- Button 4 (panic button) can be programmed for "normal" press and release instead of 3 second hold with this procedure:
* Press and hold all 4 buttons simultaneously until the LED begins to flash.
* Press the Away key (upper right).
* These are toggle features, to return to default value, repeat the steps above.

(The above is from the WS4939 users manual)

jcwren, 2020-02-19:
This is from a DSC WS4904P PIR motion detector. Given @rct's comment about sub-type 3 covering motion, glass break, maybe others, I don't know what would make sense as a sub-type text name. Maybe `Motion/Glass/Etc`?

```
{"time":"2020-02-19 15:26:01","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":1,"event":0,"tamper":1,"battery_ok":1,"xactivity":0,"xtamper1":1,"xtamper2":0,"exception":0,"esn":"3cddaf","status":194,"status_hex":"c2","mic":"CRC","mod":"ASK","freq":433.92554,"rssi":-0.102459,"snr":33.01084,"noise":-33.1133}
{"time":"2020-02-19 15:26:01","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":1,"event":0,"tamper":1,"battery_ok":1,"xactivity":0,"xtamper1":1,"xtamper2":0,"exception":0,"esn":"3cddaf","status":194,"status_hex":"c2","mic":"CRC","mod":"ASK","freq":433.92803,"rssi":-0.111965,"snr":31.61831,"noise":-31.7303}
{"time":"2020-02-19 15:26:14","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":1,"event":1,"tamper":1,"battery_ok":1,"xactivity":0,"xtamper1":0,"xtamper2":1,"exception":0,"esn":"3cddaf","status":147,"status_hex":"93","mic":"CRC","mod":"ASK","freq":433.92678,"rssi":-0.102459,"snr":31.24993,"noise":-31.3524}
{"time":"2020-02-19 15:26:14","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":1,"event":1,"tamper":1,"battery_ok":1,"xactivity":0,"xtamper1":0,"xtamper2":1,"exception":0,"esn":"3cddaf","status":147,"status_hex":"93","mic":"CRC","mod":"ASK","freq":433.92512,"rssi":-0.111965,"snr":31.24042,"noise":-31.3524}
{"time":"2020-02-19 15:26:15","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":1,"event":1,"tamper":1,"battery_ok":1,"xactivity":0,"xtamper1":0,"xtamper2":1,"exception":0,"esn":"3cddaf","status":147,"status_hex":"93","mic":"CRC","mod":"ASK","freq":433.92685,"rssi":-0.137882,"snr":30.86688,"noise":-31.0048}
{"time":"2020-02-19 15:26:15","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":1,"event":1,"tamper":1,"battery_ok":1,"xactivity":0,"xtamper1":0,"xtamper2":1,"exception":0,"esn":"3cddaf","status":147,"status_hex":"93","mic":"CRC","mod":"ASK","freq":433.92614,"rssi":-0.110607,"snr":31.61967,"noise":-31.7303}
{"time":"2020-02-19 15:27:28","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":0,"event":1,"tamper":0,"battery_ok":1,"xactivity":1,"xtamper1":0,"xtamper2":0,"exception":0,"esn":"3cddaf","status":161,"status_hex":"a1","mic":"CRC","mod":"ASK","freq":433.92874,"rssi":-0.116322,"snr":29.47515,"noise":-29.5915}
{"time":"2020-02-19 15:27:28","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":0,"event":1,"tamper":0,"battery_ok":1,"xactivity":1,"xtamper1":0,"xtamper2":0,"exception":0,"esn":"3cddaf","status":161,"status_hex":"a1","mic":"CRC","mod":"ASK","freq":433.92378,"rssi":-0.106804,"snr":29.0271,"noise":-29.1339}
{"time":"2020-02-19 15:27:28","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":0,"event":1,"tamper":0,"battery_ok":1,"xactivity":1,"xtamper1":0,"xtamper2":0,"exception":0,"esn":"3cddaf","status":161,"status_hex":"a1","mic":"CRC","mod":"ASK","freq":433.9264,"rssi":-0.105171,"snr":31.6251,"noise":-31.7303}
{"time":"2020-02-19 15:27:28","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":0,"event":1,"tamper":0,"battery_ok":1,"xactivity":1,"xtamper1":0,"xtamper2":0,"exception":0,"esn":"3cddaf","status":161,"status_hex":"a1","mic":"CRC","mod":"ASK","freq":433.92496,"rssi":-0.116322,"snr":30.5666,"noise":-30.6829}
{"time":"2020-02-19 15:28:18","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":0,"event":1,"tamper":0,"battery_ok":1,"xactivity":1,"xtamper1":0,"xtamper2":0,"exception":0,"esn":"3cddaf","status":161,"status_hex":"a1","mic":"CRC","mod":"ASK","freq":433.92352,"rssi":-0.106533,"snr":30.89824,"noise":-31.0048}
{"time":"2020-02-19 15:28:18","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":0,"event":1,"tamper":0,"battery_ok":1,"xactivity":1,"xtamper1":0,"xtamper2":0,"exception":0,"esn":"3cddaf","status":161,"status_hex":"a1","mic":"CRC","mod":"ASK","freq":433.92624,"rssi":-0.12859,"snr":31.2238,"noise":-31.3524}
{"time":"2020-02-19 15:28:18","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":0,"event":1,"tamper":0,"battery_ok":1,"xactivity":1,"xtamper1":0,"xtamper2":0,"exception":0,"esn":"3cddaf","status":161,"status_hex":"a1","mic":"CRC","mod":"ASK","freq":433.92691,"rssi":-0.111965,"snr":29.47951,"noise":-29.5915}
{"time":"2020-02-19 15:28:19","model":"DSC-Security","subtype":"unknown","id":3988911,"closed":0,"event":1,"tamper":0,"battery_ok":1,"xactivity":1,"xtamper1":0,"xtamper2":0,"exception":0,"esn":"3cddaf","status":161,"status_hex":"a1","mic":"CRC","mod":"ASK","freq":433.92483,"rssi":-0.110607,"snr":30.27268,"noise":-30.3833}
```

zuckschwerdt, 2020-02-19:
Maybe the subtype is a dead end. All classes are some kind of event trigger input that's best  handled in some external logic. You likely want to distinguish Motion and Glass break. Smoke could be a dedicated class but you likely want to handle it like Flood or CO2.
Not much gained by the subtype then. I agree that handling the keyfob different is a sound approach, but otherwise we should drop the subtype if it's not a solid information to group by or act upon.

jcwren, 2020-02-19:
I don't like when companies don't think ahead. I'd prefer that the ESN determine contain the type, or there was a byte for model number, or SOMETHING. Sure, the panel doesn't really care as it's just a contact closure in a zone and the zone type determines the action. It would just be so much more forward thinking for that to be included in what it sends.

But, unfortunately, I think you are right. Just yet another table somewhere to be able to register a particular ESN as a particular device type. No self-learning :(

It might be nice to keep the sub-type as a number. I know it's a duplication of the first 4 bits of the ESN, but it means not parsing the ESN and the next person that comes along doesn't have to be as well educated in knowing that it is the first 4 bits (and not, say, the first 5 bits) of the ESN that make up the sub-type.

rct, 2020-02-19:
The conclusion I came to a while back, other than the keyfob the subtype really isn't useful. I did think about adding it as a field when I decoded the other bits and decided against it.

You need to know where the device is and how the device is being used to do anything useful with it.

zuckschwerdt, 2020-02-19:
Does on of you like to PR the handling of the keyfob?

merbanan, 2023-02-09:
No update for a long time. Closing issue.

closed 2023-02-09 by merbanan