merbanan/rtl_433 #1321: Inconsistent uppercase vs lowercase hex values in JSON output

jcwren, 2020-02-18:
I see in some of the JSON output that hex values are printed with `%X` (**yuck!**) instead of `%x` (yay!). 

Should these all be uniformly made lowercase? If so, I can create a PR with the corrections.

zuckschwerdt, 2020-02-19:
Four devices do that right? I guess there is no one with strong opinions. (Otherwise they are wrong, obviously ;)
Go ahead an PR the change.

gdt, 2023-09-25:
@jcwren We await your PR!

zuckschwerdt, 2023-09-26:
There are now 13 devices printing uppercase hex (regex for `DATA_FORMAT.*%\d*X`). I'm going to bulk change them now.

zuckschwerdt, 2023-09-26:
I'd really like to see a uniform output of just ints for IDs in all decoders.
But there is not much to gain here by breaking the existing format (of hex strings, sometime with uppercase, instead of IDs). Closing.

closed 2023-09-26 by zuckschwerdt

gdt, 2023-09-26:
I prefer hex for ids, but I see your point about ints....