merbanan/rtl_433 #1323: please review battery_low flag for bresser_5in1

devZer0, 2020-02-19:
i have a temperature sensore very similar to bresser_5in1 (has lower features) and found that there is a low-battery bit at bit7 of byte 25.

someone with bresser5in1 device may check if this also applies to that device, i guess it has been overlooked (as it must be tested specifically

also see
https://groups.google.com/forum/#!topic/rtl_433/G0dou0HomGI

peteakalad, 2020-04-15:
Tried some old batteries in my bresser 5in1 station (7002510) but couldnt get it to trip the low battery bit.  Does support it, the manual mentions it.  

Will have to wait till my batteries get low, or if you have some byte captures (output when you add -vv) when yours has a low battery I could try and add it.

peteakalad, 2020-04-16:
Managed to find some old batteries which trigger low battery.

Few packet captures of permutations of scenarios tested;

{286} 00 aa aa aa aa aa 2d d4 e3 fd 7f 89 7e 8a ed 68 fe af 9b fd ff 1c 02 80 76 81 75 12 97 01 50 64 02 00 00 00 [GOOD BATT +TEMP]
{286} 00 aa aa aa aa aa 2d d4 ec 91 ff ff 1f fb ef e7 fe ad ed ff f7 13 6e 00 00 e0 04 10 18 01 52 12 00 08 00 00 [GOOD BATT -TEMP]
{286} 00 aa aa aa aa aa 2d d4 ef a1 ff ff cf ff ef fc fd b6 fb ff 7f 10 5e 00 00 30 00 10 03 02 49 04 00 80 00 00 [LOW BATT +TEMP]
{286} 00 aa aa aa aa aa 2d d4 ed a1 ff ff 1f ff ef 8f ff d6 df ff 77 12 5e 00 00 e0 00 10 70 00 29 20 00 88 00 00 [LOW BATT -TEMP -7.0C]

Key byte is 25 (0 based, counted after the 2d d4 sync);
- MSB is battery low byte (8 when low)
- LSB is the temperature sign (8 when -ve, already implemented)

00 = good batt, +temp
08 = good batt, -temp
80 = low batt, +temp
88 = low batt, -temp

Change tested and commited to https://github.com/peteakalad/rtl_433

closed 2020-05-14 by zuckschwerdt

kroeckx, 2021-03-14:
@peteakalad: I'm trying to understand what's currently in the value currently marked as "station ID (maybe)". You have values "02", "6e" and "5e" there. Is that all from the same device, or are they different devices?

peteakalad, 2021-03-14:
All from same device - these values would have been captured while I was switching batteries (good/bad) to test and those values change when you do that.  Its common for devices like this to have the station/device id changing following battery change/reset.  You may find the same with the rain gauge - reset button on bottom likely to do the same.   But after that expect it to remain constant / unchanged until another reset/battery change.  Will help to confirm if there are bytes which represent the model of the sensor rather than the device/station id.