merbanan/rtl_433 #1324: Generate fake TPM?

tranzmatt, 2020-02-20:
Is there an easy way to generate a synthetic signal?  I'm having a situation where I can run rtl_433 and pick up TPM's around me, both on the host OS and in a Docker, but when try to run kismet, it gives me nothing, even though it says it's connected to rtl433.  Alas, signals around me are sporadic, so I was wanting to generate a fake one to pick up as a test.  The tests I see don't seem to have a way to drive a signal.

merbanan, 2020-02-20:
Look at that, a kismet plugin for rtl_433. Never heard of it before. Anyway run rtl_433 from the command line to see of there are any periodic signals. If there are then the plugin uses the wrong configuration parameters for rtl_433. And for those issues I have to refer you to the plugin author for help.

tranzmatt, 2020-02-20:
Well, the plugin works when I run kismet native on my laptop and I can run the command line ok on a Jetson nano in a docker.  It's kismet on the Jetson docker that seems to be having issues.  I was hoping to fake a transmitter so I could tickle things and see where the disconnect it.

zuckschwerdt, 2020-02-20:
Do you want rtl_433 to emit fake events or do you want to transmit a radio signal?

tranzmatt, 2020-02-20:
Preferably transmit a signal so I can have rtl_433 running in one docker instance and kismet running in another instance (each with it's own SDR of the same/config) and pretend as if a TPM or even a key fob just activated.  It's mostly because devices pass by sporadically and I need to control the signal.

I suspect it's kismet as I'm literally doing this now (same docker instance, running rtl_433 and kismet on different SDR's.  rtl_433 just picked up a passing Toyota that kismet didn't).

Mindavi, 2020-02-20:
You'd need a transmitter/tranceiver, like the hackrf one or something like that. You could also use an OOK transceiver, if the signals are OOK. However, this would require a big effort, while it seems you're looking for some simple testing.

Possibly you could record a signal and read it back using rtl_433 (-s known, -r <filename>) which might simulate the behaviour you want in a 'good enough' fashion.

Lastly, maybe you have friends which have 433mhz sensors that you could borrow for some testing.

tranzmatt, 2020-02-20:
Thanks.  I'll give it a shot.

closed 2020-02-20 by tranzmatt